realsenseai/realsense-ros #3227: My system is ubuntu 18.04lts, camera is realsense d405, RealSense ROS v2.3.2, LibRealSense v2.50.0and when i use the "roslaunch realsense2_camera rs_camera.launch" I got:[ERROR] [1728617727.524661741]: hwmon command 0x75( 0 1df 0 34f ) failed (response -6= Invalid parameter)   this question has trouble me long  days

konwnohing, 2024-10-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }       D405                   | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..        ubuntu18.04                   |
| Librealsense SDK Version       |  { 2.<?>.<?> }           2.50.0                                         | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
[ERROR] [1728617727.524661741]: hwmon command 0x75( 0 1df 0 34f ) failed (response -6= Invalid parameter)

MartyG-RealSense, 2024-10-11:
Hi @konwnohing  Official support in the librealsense SDK for the D405 camera model was not introduced until version **2.51.1**, the version after the 2.50.0 version that you are using.  It is fine to use 2.51.1 with the 2.3.2 ROS1 wrapper.

The recommended firmware driver version for use with 2.50.0 and 2.51.1 is **5.13.0.50**.  Older firmware versions than 5.13.0.50 should not be used with the D405.

konwnohing, 2024-10-13:
ok i will try to use 2.51.1 with the 2.3.2 ros1 wrapper

closed 2024-10-28 by MartyG-RealSense