realsenseai/realsense-ros #3230: Cannot detect D455 by using RK3588S under ubuntu 24.04 and ros2 jazzy

YujieSuCWE, 2024-10-20:
I met the problem same as #3183 of being not able to find my device. My system is ubuntu 24.04, my librealsense version is v2.56.2, my ros distro is jazzy, and my realsense-ros version is 4.56.1. What is different is that I can open realsense-viewer, but still cannot find the device on both rs_launch.py and the viewer. I am pretty sure I used RSUSB backend method when compiling librealsense.

_Originally posted by @YujieSuCWE in https://github.com/IntelRealSense/realsense-ros/issues/3183#issuecomment-2425291390_

YujieSuCWE, 2024-10-21:
I changed the librealsense version to v2.56.1, right now the realsense-ros failed to compile.

YujieSuCWE, 2024-10-21:
I just checked the API change, and I find out that get_calibration_config is a new function in librealsense 2.56.1 which is not yet being supported by the latest realsense-ros version. So I guess that's where the problem is. Is there any way to solve this under my current ubuntu and ros2 version?

MartyG-RealSense, 2024-10-21:
Hi @YujieSuCWE  librealsense 2.56.1 and ROS2 wrapper 4.56.1 are the correct version match, so there should not be an issue of compatibility between them.  I note that you installed librealsense 2.56.2 before 2.56.1 though.  

Each time that the librealsense version is changed, the ROS wrapper must be built again.  Can you confirm whether you rebuilt the 4.56.1 wrapper after changing to librealsense 2.56.1 please?

Can you also confirm if your camera has firmware driver version 5.16.0.1 installed?

YujieSuCWE, 2024-10-22:
Hello @MartyG-RealSense the error of not having get_calibration_config() is during the rebuilding of the 4.56.1 wrapper. I've checked the versions, and my camera firmware driver version was 5.13.x and I updated it. But right now I don't know if this will work since I cannot successfully build the wrapper.

MartyG-RealSense, 2024-10-22:
Are you building from the **ros2-master** branch of the ROS wrapper or **ros2-development** please?  You should be building from **ros2-master**.

YujieSuCWE, 2024-10-22:
I built from the latest release by just downloading its zip.

MartyG-RealSense, 2024-10-23:
The default kernel version of Ubuntu 24.04 is kernel 6.8.  Support for patching kernel 6.8 was added in the current latest librealsense version 2.56.2.

So if libuvc or RSUSB are not working with Ubuntu 24 (there have been past reported cases of problems with this) then an alternative approach may be to build librealsense from source code without using libuvc or RSUSB, and apply the kernel patch script **patch-realsense-ubuntu-lts-hwe.sh** to the 6.8 kernel.  In librealsense 2.56.2, patch-realsense-ubuntu-lts-hwe.sh was updated to support patching kernel 6.8.

YujieSuCWE, 2024-10-23:
I think my linux kernel version is 6.10. Does this also need to add the patch?

MartyG-RealSense, 2024-10-23:
Usually, if support is stated for specific kernel versions then it means that only those versions are supported by patching and not newer ones yet.  In that situation, the choice is either (a) to use libuvc / RSUSB or (b) not to use them and instead install librealsense from source code without applying a kernel patch.

YujieSuCWE, 2024-11-03:
I re-installed ubuntu and redid all the processes. Now it successfully works and is able to detect the device. Thank you for your help!

MartyG-RealSense, 2024-11-04:
You are very welcome.  It's great to hear that you succeeded - thanks very much for the update!  As you have achieved a solution, I will close this case.  Thanks again!

closed 2024-11-04 by MartyG-RealSense