realsenseai/realsense-ros #3231: Unclear Parameter Settings for Hardware Synchronization in ROS2 Humble Driver

Liu-Jinxin, 2024-10-21:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i / D455             | 
| Firmware Version                   | 5.15.1                  | 
| Operating System & Version |  Ubuntu22.04 | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC                     |
| Librealsense SDK Version       |  {2.55.1 }                                                   | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  Humble                               | 
| RealSense ROS Wrapper Version   |  4.55.1                                         | 

### Issue Description
I am working on a project involving hardware synchronization for multiple cameras using the ROS2 driver. Despite setting the following ROS2 driver parameters intended to manage synchronization, the cameras are not synchronizing as they should:
```
 configurable_parameters = [  {'name': 'inter_cam_sync_mode',          'default': '1', 'description': 'Intel camera sync mode'},
{'name': 'output_trigger_enabled',       'default': 'true', 'description': 'Enable output trigger'},]
```
I am unclear on whether these settings are appropriate or even applicable for the ROS2 Humble driver. Could someone clarify if these parameters are correct for achieving hardware synchronization in this environment, or guide me on how to configure the driver properly for this purpose?

Any help or pointers to relevant documentation would be greatly appreciated!

Thank you!

Liu-Jinxin, 2024-10-22:
Hi @MartyG-RealSense 

Thanks for your help! In ROS 1, I was able to configure the hardware synchronization by setting the following parameters directly in the launch file:
```
/camera/stereo_module/inter_cam_sync_mode: 1
/camera/stereo_module/output_trigger_enabled: true
```

However, it seems that these parameters in the launch file are not being correctly picked up in the ROS 2 launch configuration, which might be causing the inability to trigger synchronization as expected. 
```
 {'name': 'stereo_module.intel_cam_sync_mode',   'default': '1', 'description': 'Intel camera sync mode'},
 {'name': 'stereo_module.output trigger enabled','default': 'true', 'description': 'Enable output trigger'},
```
Could you please advise on the correct way to specify these settings in a ROS 2 environment, or if there are different parameters I should be using to achieve this?

Liu-Jinxin, 2024-10-22:
I've managed to resolve the issue. It turns out the correct parameters for ROS 2 should indeed specify the module name. Here are the corrected settings that worked for achieving hardware synchronization:
```
{'name': 'depth_module.inter_cam_sync_mode', 'default': '1', 'description': 'Intel camera sync mode'},
{'name': 'depth_module.output_trigger_enabled', 'default': 'true', 'description': 'Enable output trigger'},
```

Thanks  for your help.

closed 2024-10-22 by Liu-Jinxin

MartyG-RealSense, 2024-10-22:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update and sharing the details of your solution!