realsenseai/realsense-ros #3232: Issue with Unstable Topic Frequency in ROS2 Driver Compared to ROS1

Liu-Jinxin, 2024-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                      | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |    Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC                       |
| Librealsense SDK Version       |  { 2.55.1 }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  Humble}                                           | 
| RealSense ROS Wrapper Version   |  4.55.1                                                  | 

### Issue Description
### Description
I have conducted tests on the same Intel RealSense D435i camera, with the same computer, comparing ROS Noetic (ROS1) and ROS2 Humble. While testing the `compressed image` topic, I noticed that the frequency stability in ROS1 is quite consistent and satisfactory. However, in ROS2 Humble, the frequency is significantly less stable.

### Questions
1. Is there a known issue with topic frequency stability in ROS2 drivers, particularly for image data?
2. Are there specific settings or configurations in ROS2 that I should adjust to improve this stability?

Any advice or suggestions would be greatly appreciated as I continue to integrate ROS2 into my projects.

Thanks for your support and looking forward to your insights!
Noetic:
[image]
Humble:
[image]

MartyG-RealSense, 2024-10-22:
Hi @Liu-Jinxin  There was a case at https://github.com/IntelRealSense/realsense-ros/issues/3105 a couple of months ago about low performance from a compressed depth topic when using the ROS2 wrapper.

After a lot of discussion and investigation, one of my Intel RealSense colleagues published a recommendation at https://github.com/IntelRealSense/realsense-ros/issues/3105#issuecomment-2392436864

MartyG-RealSense, 2024-10-23:
Have you installed the image transport plugin designed for Humble using the instruction below, please?

`sudo apt install ros-humble-image-transport`

Liu-Jinxin, 2024-10-24:
Hi @MartyG-RealSense,  yes I have,

```
➜  ros2_ws dpkg -l | grep ros-humble-image-transport   
ii  ros-humble-image-transport                             3.1.9-1jammy.20240820.161125            amd64        image_transport should always be used to subscribe to and publish images.
ii  ros-humble-image-transport-plugins                     2.5.2-1jammy.20240820.163724            amd64        A set of plugins for publishing and subscribing to sensor_msgs/Image topics in representations other than raw pixel data.
```

MartyG-RealSense, 2024-10-27:
Hi again @Liu-Jinxin  Intel have an existing internal development ticket open related to problems with ROS2 compressed topics and they have added your case to the ticket.

closed 2025-07-01 by Nir-Az