realsenseai/realsense-ros #3233: Unable to compile realsense-ros on Ubuntu 22.04 running Docker

kushal2507, 2024-10-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                       | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Ubuntu 22 | 
| Kernel Version (Linux Only)    |  6.8.0-47-generic                                                                    | 
| Platform                                 | PC                         |
| Librealsense SDK Version       |  2.56.2                           | 
| ROS Distro			         |  Humble                                        | 
| RealSense ROS Wrapper Version   |  4.56.1                                               | 

### Issue Description
I am running docker with Ubuntu 22.04 having kernel 6.8.0.47-generic. The host PC on which the docker is running has the same properties. I followed the instructions mentioned [here](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692) to clone and compile librealsense. In my ros2_ws I have cloned realsense-ros with tag 4.56.1. Upon running colcon build I get the following error and I am unable to compile it.

[image]

I am running my docker container with the following runArgs : 

"runArgs": [
        "--network=host",
        "--cap-add=SYS_PTRACE",
        "--security-opt=seccomp:unconfined",
        "--security-opt=apparmor:unconfined",
        "--ipc=host",
        "--pid=host",
        "--gpus=all",
        "--env=NVIDIA_DRIVER_CAPABILITIES=all",
        "--env=DISPLAY",
        "--privileged",
        "--volume=/dev:/dev",
        "--device-cgroup-rule=c 189:* rmw",
        "--volume=/tmp/.X11-unix:/tmp/.X11-unix",
    ]

MartyG-RealSense, 2024-10-25:
Hi @kushal2507  There was a recent case of the `Can't create temporary cache file /etc/ld.so.cache~: Permission denied` error at https://github.com/IntelRealSense/realsense-ros/issues/3197#issuecomment-2330933365 and the apparent cause of the problem in that particular case was that the ROS wrapper was not the only data in the catkin workspace folder.  The RealSense user in that case had installed librealsense into the catkin folder too.

Can you confirm if the ROS wrapper is the only data installed in your `ros2_ws/src` folder please?

closed 2024-10-25 by kushal2507

MartyG-RealSense, 2024-10-25:
You are very welcome.  I'm pleased to hear that moving librealsense's location worked for you.  :)