realsenseai/realsense-ros #3234: Problems running 2 d435i cameras on Ubuntu 22 with Docker

kushal2507, 2024-10-25:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                           | 
| Firmware Version                   | 5.16.0 .1                     | 
| Operating System & Version |   Ubuntu 22 | 
| Kernel Version (Linux Only)    |  6.8.0-47-generic                                                                       | 
| Platform                                 | PC                        |
| Librealsense SDK Version       |  2.56.2                                            | 
| ROS Distro			         |  Humble                                      | 
| RealSense ROS Wrapper Version   |  v4.55.1                                                | 

### Issue Description
I am truing to run 2 intel d435i cameras on my ros2 humble ubuntu 22 docker container. Attached below is the log after I launch the two cameras : 

```
ros2 launch bgl_hardware_launcher multi_realsense.launch.py 
[INFO] [launch]: All log files can be found below /home/bgldev/.ros/log/2024-10-25-12-43-27-226480-bglnlamrproto-481138
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [481140]
[INFO] [realsense2_camera_node-2]: process started with pid [481142]
[realsense2_camera_node-1] [INFO] [1729853007.345913787] [camera_front]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1729853007.345961512] [camera_front]: Built with LibRealSense v2.56.2
[realsense2_camera_node-1] [INFO] [1729853007.345968348] [camera_front]: Running with LibRealSense v2.56.2
[realsense2_camera_node-2] [INFO] [1729853007.346032214] [camera_rear]: RealSense ROS v4.55.1
[realsense2_camera_node-2] [INFO] [1729853007.346074619] [camera_rear]: Built with LibRealSense v2.56.2
[realsense2_camera_node-2] [INFO] [1729853007.346081609] [camera_rear]: Running with LibRealSense v2.56.2
[realsense2_camera_node-2]  25/10 12:43:27,354 WARNING [134281101837888] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-2]  25/10 12:43:27,364 WARNING [134281101837888] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-2]  25/10 12:43:27,374 ERROR [134281101837888] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[realsense2_camera_node-2]  25/10 12:43:27,374 ERROR [134281133295168] (uvc-sensor.cpp:427) acquire_power failed: failed to set power state
[realsense2_camera_node-2]  25/10 12:43:27,374 ERROR [134281133295168] (rs.cpp:255) [rs2_create_device( info_list:0x7a20a4004cc0, index:0 ) UNKNOWN] failed to set power state
[realsense2_camera_node-2]  25/10 12:43:27,374 ERROR [134281133295168] (rs.cpp:255) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-2]  25/10 12:43:27,374 WARNING [134281133295168] (rs.cpp:391) null pointer passed for argument "device"
[realsense2_camera_node-2] [WARN] [1729853007.374737117] [camera_rear]: Device 1/2 failed with exception: failed to set power state
[realsense2_camera_node-1] [INFO] [1729853007.466937984] [camera_front]: Device with serial number 031522070524 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1729853007.466991343] [camera_front]: Device with physical ID 2-2-33 was found.
[realsense2_camera_node-1] [INFO] [1729853007.466997061] [camera_front]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1729853007.467118276] [camera_front]: Device with port number 2-2 was found.
[realsense2_camera_node-2]  25/10 12:43:27,472 WARNING [134281112323648] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  25/10 12:43:27,485 WARNING [137682535581248] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  25/10 12:43:27,495 WARNING [137682535581248] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  25/10 12:43:27,505 ERROR [137682535581248] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[realsense2_camera_node-1]  25/10 12:43:27,505 ERROR [137683045189184] (uvc-sensor.cpp:427) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  25/10 12:43:27,506 ERROR [137683045189184] (rs.cpp:255) [rs2_create_device( info_list:0x7d38b0004cc0, index:1 ) UNKNOWN] failed to set power state
[realsense2_camera_node-1]  25/10 12:43:27,506 ERROR [137683045189184] (rs.cpp:255) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  25/10 12:43:27,506 WARNING [137683045189184] (rs.cpp:391) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1729853007.506226044] [camera_front]: Device 2/2 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1729853007.506279817] [camera_front]: The requested device with serial number 025222070751 is NOT found. Will Try again.
[realsense2_camera_node-2] [INFO] [1729853007.511756842] [camera_rear]: Device with serial number 025222070751 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1729853007.511808267] [camera_rear]: Device with physical ID 2-4-32 was found.
[realsense2_camera_node-2] [INFO] [1729853007.511824194] [camera_rear]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-2] [INFO] [1729853007.512005283] [camera_rear]: Device with port number 2-4 was found.
[realsense2_camera_node-2] [ERROR] [1729853007.512026405] [camera_rear]: The requested device with serial number 031522070524 is NOT found. Will Try again.
[realsense2_camera_node-2]  25/10 12:43:33,529 WARNING [134280741127744] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  25/10 12:43:33,543 WARNING [137682715936320] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-2]  25/10 12:43:33,543 WARNING [134280741127744] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
[realsense2_camera_node-1]  25/10 12:43:33,553 WARNING [137682715936320] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  25/10 12:43:33,563 ERROR [137682715936320] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[realsense2_camera_node-1]  25/10 12:43:33,563 ERROR [137683045189184] (uvc-sensor.cpp:427) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  25/10 12:43:33,564 ERROR [137683045189184] (rs.cpp:255) [rs2_create_device( info_list:0x7d38b0022470, index:0 ) UNKNOWN] failed to set power state
[realsense2_camera_node-1]  25/10 12:43:33,564 ERROR [137683045189184] (rs.cpp:255) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  25/10 12:43:33,564 WARNING [137683045189184] (rs.cpp:391) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1729853013.564381487] [camera_front]: Device 1/2 failed with exception: failed to set power state
[realsense2_camera_node-2] [INFO] [1729853013.612388814] [camera_rear]: Device with serial number 031522070524 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1729853013.612447756] [camera_rear]: Device with physical ID 2-2-33 was found.
[... 172 lines cut ...]
    {"name": "clip_distance", "default": "-2.", "description": "''"},
    {"name": "angular_velocity_cov", "default": "0.01", "description": "''"},
    {"name": "linear_accel_cov", "default": "0.01", "description": "''"},
    {"name": "diagnostics_period", "default": "0.0", "description": "Rate of publishing diagnostics. 0=Disabled"},
    {"name": "publish_tf", "default": "true", "description": "[bool] enable/disable publishing static & dynamic TF"},
    {"name": "tf_publish_rate", "default": "0.0", "description": "[double] rate in Hz for publishing dynamic TF"},
[... cut ...]
    {"name": "hdr_merge.enable", "default": "false", "description": "hdr_merge filter enablement flag"},
    {"name": "wait_for_device_timeout", "default": "-1.", "description": "Timeout for waiting for device to connect (Seconds)"},
    {"name": "reconnect_timeout", "default": "6.", "description": "Timeout(seconds) between consequtive reconnection attempts"},
]

```
After receiving this error `WARNING [137682959205952] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1729853021018.471680` the images are no longer published for that camera.

MartyG-RealSense, 2024-10-25:
Hi @kushal2507  If you are launching with 2 cameras then I would recommend using the **rs_multi_camera_launch.py** launch file instead of rs_launch.py, or launching each camera with an rs_launch.py instruction for each individual camera in two separate terminal windows.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/launch/rs_multi_camera_launch.py

If your preference is to launch with a single rs_launch.py instruction then you can define a separate **camera_name** for each camera to ensure that the launch can correctly tell the two cameras apart.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-master?tab=readme-ov-file#camera-name-and-camera-namespace

kushal2507, 2024-10-25:
hi @MartyG-RealSense  I have my custom launch file, it is laucnhing the two cameras with different names, 

## Main launch file 
```
# Copyright 2023 Intel Corporation. All Rights Reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# DESCRIPTION #
# ----------- #
# Use this launch file to launch a device and get the params from a YAML file.
# The Parameters available for definition in the command line for the camera are described in rs_launch.configurable_parameters
# command line example:
# ros2 launch realsense2_camera rs_launch_get_params_from_yaml.py

"""Launch realsense2_camera node."""
import os
import copy
from launch import LaunchDescription
from launch.actions import OpaqueFunction
from launch.substitutions import LaunchConfiguration
import sys
import pathlib
from ament_index_python.packages import get_package_share_directory

sys.path.append(str(pathlib.Path(__file__).parent.absolute()))
import realsense_launch  # noqa: E402

sys.path.append(os.path.join(get_package_share_directory("bgl_hardware_launcher"), "launch"))

params_file1 = os.path.join(
    get_package_share_directory("bgl_hardware_launcher"),
    "config",
    "realsense_front.yaml",
)
params_file2 = os.path.join(
    get_package_share_directory("bgl_hardware_launcher"),
    "config",
    "realsense_rear.yaml",
)

local_parameters = [
    {
        "name": "camera_name1",
        "default": "camera_front",
        "description": "camera unique name",
    },
    {
        "name": "camera_name2",
        "default": "camera_rear",
        "description": "camera unique name",
    },
    {
        "name": "config_file1",
        "default": [params_file1],
        "description": "yaml config file",
    },
    {
        "name": "config_file2",
        "default": [params_file2],
[... 215 lines cut ...]
                    "/camera_rear/left/camera_info",
                ),
                (
                    "/camera_rear/infra1/image_rect_raw",
                    "/camera_rear/left/image_rect_raw",
                ),
                (
                    "/camera_rear/infra2/camera_info",
                    "/camera_rear/right/camera_info",
                ),
                (
                    "/camera_rear/infra2/image_rect_raw",
                    "/camera_rear/right/image_rect_raw",
                ),
            ],
        )
    ]

def generate_launch_description():
    return LaunchDescription(
        declare_configurable_parameters(configurable_parameters)
        + [OpaqueFunction(function=launch_setup, kwargs={"params": set_configurable_parameters(configurable_parameters)})]
    )
```
The two yaml files that you see in the launch.py file, adds unique serial numbers to each camera

kushal2507, 2024-11-07:
I was able to solve this by uninstalling the librealsense which I had installed from source and instead used `sudo apt install ros-humble-librealsense2-*` and `sudo apt install ros-humble-realsense2*`

closed 2024-11-07 by kushal2507

MartyG-RealSense, 2024-11-07:
It's great to hear that you achieved a solution, @kushal2507 - thanks very much for the update!