realsenseai/realsense-ros #3235: [SLAM with D435i] TF errors (TF_NAN_INPUT, TF_DENORMALIZED_QUATERNION) and incorrect 'camera_link' frame data

sooyoungmoon, 2024-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | **D435i**                       | 
| Firmware Version                   | **5.14.0**                          | 
| Operating System & Version | **Ubuntu 22.04.5 LTS** | 
| Kernel Version (Linux Only)    |  **6.8.0-47-generic**                                                                    | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  **2.55.1**                                                     | 
| Language                               |  {C++ }         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  **Humble**                                           | 
| RealSense ROS Wrapper Version   |  **4.55.1**                                               | 

### Issue Description

Hello, I'm following [SLAM-with-D435i-wiki](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i) to test SLAM using Intel Realsense2 D435i on ROS2 (Humble) environment. 

After modifying a launch file and a few parameters, I was able to execute all the necessary nodes.
I can see the 'camera_link' frame moves as I change the position and direction of my D435i camera. 
However, soon after I move my camera, **I can see on rviz the 'camera_link' frame 'floats away' from the real pose of the camera and the following error messages are printed on the screen.**

[image]

In addition to that, the /rtabmap/odom topic includes abnormal values (covariance matrix has very big (i.e. 9999) diagonal values) as soon as the errors occur .

I'm curious about the reason why those error occur simultaneously at multiple ROS2 nodes including:
  /rtabmap/rgbd_odometry, /rtabmap/rtabmap, /ukf_filter_node, and /rviz2.

Could you give me some hint to solve the issue? Thank you!!

For your reference, I attach the parameters I changed and the launch file I used (opensource_tracking_ros2.launch.py)

[image]

-------------------- opensource_tracking_ros2.launch.py ------------------------------

import os

from ament_index_python import get_package_share_directory

from launch import LaunchContext, LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription
from launch.actions import SetLaunchConfiguration
from launch.actions import GroupAction

from launch.conditions import IfCondition, UnlessCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.substitutions import TextSubstitution
from launch.substitutions import ThisLaunchFileDir
from launch_ros.actions import Node
from launch_ros.actions import PushRosNamespace

def generate_launch_description():

    remappings=[
        ('/camera/camera/aligned_depth_to_color/image_raw', '/camera/aligned_depth_to_color/image_raw'),
        ('/camera/camera/aligned_depth_to_color/camera_info', '/camera/aligned_depth_to_color/camera_info'),
        ('/camera/camera/aligned_depth_to_color/image_raw/compressed', '/camera/aligned_depth_to_color/image_raw/compressed'),
[... 931 lines cut ...]
        rgb_topic_relay_arg,
        depth_topic_relay_arg,
        left_image_topic_relay_arg,
        right_image_topic_relay_arg,
        rgbd_topic_relay_arg,
        
        frequency_arg,
        base_link_frame_arg,
        odom0_arg,
        odom0_config_arg,
        odom0_relative_arg,
        odom0_pose_rejection_threshold_arg,
        odom0_twist_rejection_threshold_arg,
        imu0_arg,
        imu0_config_arg,
        imu0_differential_arg,
        imu0_relative_arg,
        use_control_arg,
        rs2_camera_launch_include,
        imu_filter_madgwick_node,
        #rtabmap_lite_launch_include,
        rtabmap_ros_launch_include,
        robot_localization_launch_include
    ])
------------------------------------------------ end ----------------------------------------------

MartyG-RealSense, 2024-10-27:
Hi @sooyoungmoon  Converting the SLAM guide for use with ROS2 is not usually done by RealSense users and so there is not a previous example available to compare your own work to in order to diagnose problems, unfortunately.  

RealSense ROS2 users have typically used a combination of **slam_toolbox** and _depthimage_to_laserscan** to implement a SLAM system, as described at https://github.com/IntelRealSense/realsense-ros/issues/3046#issuecomment-1999092540

**slam_toolbox:** https://github.com/SteveMacenski/slam_toolbox

**depthimage_to_laserscan:** https://github.com/ros-perception/depthimage_to_laserscan/tree/ros2

MartyG-RealSense, 2024-10-29:
You are very welcome, @sooyoungmoon - I look forward to your next report.  Good luck!

sooyoungmoon, 2024-10-30:
@MartyG-RealSense 

Hello, for other researchers & developers, I hope you had a good day.

I'm struggling with this 'SLAM with D435i' issue to get any map data shown on rviz2.

Instead of trying [SLAM-with-D435i-wiki](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), 
I installed and tested slam_toolbox and depthimage_to_laserscan.

In short, I found that the rgb->laserscan data conversion worked well, but the map data was not published by slam_toolbox node. I will check similar issues in this and related repos, but I would be appreciated if you give me any hint to

The test procedure is as follows:

1) (Executed the realsense2 camera node) $ ros2 launch realsense2_camera rs_launch.py (with remapping config)

2) (Executed the package to convert rgb to scan) $ ros2 launch depthimage_to_laserscan depthimage_to_laserscan-launch.py
	
3) $ ros2 launch slam_toolbox online_launch.py
	
4) $ rviz2
	
5) (On rviz2) I added '/map' topic -> "No map received" message occurred

6) $ ros2 run tf2_tools view_frames -> there was no 'map' frame on the tf2 tree diagram.

7) I executed a static_tranform_publisher as below:
   $ ros2 run tf2_ros static_transform_publisher --x 0 --y 0 --z 0 --qx 0 --qy 0 --qz 0 --qw 1 --frame-id map --child-frame-id camera_link

8) $ ros2 run tf2_tools view_frames -> 'map' frame was added on the tf2 tree diagram
[tf2_tree(slam_toolbox and depthimage_to_laserscan).pdf](https://github.com/user-attachments/files/17574911/tf2_tree.slam_toolbox.and.depthimage_to_laserscan.pdf)

9) When I added '/scan' , I could see the laser scan image on rviz2.
[image]

Still, I couldn't see any map data on rviz2. 
     $ ros2 topic echo /map -> returned nothing

10) It seemed that the scan data was being published well
    $ ros2 topic echo /scan -> scan data messages were printed on the screen

MartyG-RealSense, 2024-10-31:
Hi @sooyoungmoon  Do you experience the message `[INFO] [1655749660.201251796] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame'` like in the slam_toolbox case at https://github.com/IntelRealSense/realsense-ros/issues/2387 where a RealSense user was not receiving map data?

At the end of that case, another RealSense user suggested at https://github.com/IntelRealSense/realsense-ros/issues/2387#issuecomment-1667319135 to publish the odometry value to the `/odom` topic

sooyoungmoon, 2024-11-04:
> Hi @sooyoungmoon Do you experience the message `[INFO] [1655749660.201251796] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame'` like in the slam_toolbox case at #2387 where a RealSense user was not receiving map data?
> 
> At the end of that case, another RealSense user suggested at [#2387 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2387#issuecomment-1667319135) to publish the odometry value to the `/odom` topic

@MartyG-RealSense 
You are right! 
That's the same message except that in my case the reason is 'discarding-message-because-the-queue-is-full'. Based on the user's suggestion you mentioned and other resources such as a slam_toolbox repo, I believe that I figured it out how to solve this problem. (To create a new node to publish transform data (odom_frame -> base_frame) given imu data.
I think I can solve this issue in a few days. Thank you!

MartyG-RealSense, 2024-11-04:
You are very welcome, @sooyoungmoon - I'm pleased that I could be of help.  Good luck!

sooyoungmoon, 2024-11-05:
@MartyG-RealSense 
> You are very welcome, @sooyoungmoon - I'm pleased that I could be of help. Good luck!

I created a new ROS2 node which calculates the camera pose from imu data and then publishes transform data (frame_id: map, child_frame_id: odom). 
After that, the map topic data was published by /slam_toolbox node. But there is a new issue.

**Q. Is there a better way to compute camera pose compared to my approach (to compute camera pose by using imu data only)?**

At first, I tried to use the orientation within the Imu message to estimate camera pose. But it was too unstable, so I skipped this approach after a few trials.

After that, I am trying to compute the position and orientation of my camera using the linear acceleration and angular velocity values. 

Until now, it is not quite successful because the camera position experiences drift, and the estimated position become farther and farther away from the real position. 

Especially, the absolute value for linear acceleration (y axis) is strangely high (please refer to the topic data shown below)

$ ros2 topic echo /imu/data
...
header:
  stamp:
    sec: 1730819791
    nanosec: 632761600
  frame_id: camera_imu_optical_frame
orientation:
  x: -0.49896361969468156
  y: 0.48194959538813315
  z: -0.5038171864430749
  w: 0.5147117021896622
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.010471975430846214
  y: -0.006981316953897476
  z: 0.003490658476948738
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.127486452460289
  y: -9.551676750183105
  z: 0.4314925968647003
linear_acceleration_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
---
header:
  stamp:
    sec: 1730819791
    nanosec: 637771520
  frame_id: camera_imu_optical_frame
orientation:
  x: -0.4990473420685022
  y: 0.4815993639143508
  z: -0.5041232176980946
  w: 0.5146587067459129
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.001745329238474369
  y: -0.012217304669320583
  z: 0.003490658476948738
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.127486452460289
  y: -9.551676750183105
  z: 0.4314925968647003
linear_acceleration_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
---
header:
  stamp:
    sec: 1730819791
    nanosec: 642781952
  frame_id: camera_imu_optical_frame
orientation:
  x: -0.49911623571819935
  y: 0.48125715574947686
  z: -0.5044428763567813
  w: 0.5145987930177406
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: 0.0
  y: -0.003490658476948738
  z: 0.005235987715423107
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.127486452460289
  y: -9.551676750183105
  z: 0.4314925968647003
linear_acceleration_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
---
header:
  stamp:
    sec: 1730819791
    nanosec: 647792384
  frame_id: camera_imu_optical_frame
orientation:
  x: -0.49915365280729845
  y: 0.4815220580607172
  z: -0.50418338333173
  w: 0.5145689987380432
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.003490658476948738
  y: -0.001745329238474369
  z: 0.0
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.0
  y: -9.532063484191895
  z: 0.3040061295032501
linear_acceleration_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
---
header:
  stamp:
    sec: 1730819791
    nanosec: 652804608
  frame_id: camera_imu_optical_frame
orientation:
  x: -0.49919502532603227
  y: 0.4817768005011726
  z: -0.5039204720266867
  w: 0.5145479560361664
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.001745329238474369
  y: -0.006981316953897476
  z: 0.0
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.0
  y: -9.532063484191895
  z: 0.3040061295032501
linear_acceleration_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
---

MartyG-RealSense, 2024-11-05:
RealSense 400 Series cameras do not have the in-built ability to calculate the relative position of the camera device.  The only approach that I know of (other than the ROS2 one that you created yourself) is a ROS1 user at the link below who used rtabmap to study a recorded pointcloud map file and obtain the relative position from that map.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

MartyG-RealSense, 2024-11-15:
Hi @sooyoungmoon  Do you require further assistance with this case, please?  Thanks!

sooyoungmoon, 2024-11-16:
@MartyG-RealSense 
Hello, I managed to draw a map using slam_toolbox and depthimage_to_laserscan. 
This was possible by providing transform data (odom to camera_link).
I have one more question. When I move my D435i camera to generate maps, the realsense2_camera node often disconnects the connection to the camera and tries to re-connect to it. Could you tell me under which condition does it happen? Thanks.
[image]

MartyG-RealSense, 2024-11-16:
I'm pleased to hear that you succeeded in drawing a map!

In regard to your new question, Intel's D435i SLAM guide for ROS1 provides the following guidance 

*******

The built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track.  It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position.  The odds for recovery get very slim, very quickly

MartyG-RealSense, 2024-11-23:
Hi @sooyoungmoon  Do you have any further questions about the camera disconnecting, please?  Thanks!

sooyoungmoon, 2024-11-26:
> Hi @sooyoungmoon Do you have any further questions about the camera disconnecting, please? Thanks!

@MartyG-RealSense 
Hello, I'm working on to solve the camera connection issue, but I think we can close this issue now.
Thank you for your nice support!

sooyoungmoon, 2024-11-26:
I close this issue now.

closed 2024-11-26 by sooyoungmoon

MartyG-RealSense, 2024-11-26:
You are very welcome.  I'm pleased that I could help!