realsenseai/realsense-ros #3236: problems in receiving IMU data

dyb678, 2024-10-28:
Hi ! I use the D435i and want to get the imu data. But I find I have the topic"/camera/imu" in my rostopic list. But I cann't receive the data .My launch file is this:

<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
 
  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>
 
  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>
 
  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
 
  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>
 
  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="200"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
 
  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
 
  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>
 
  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>
 
  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="emitter_enable"   		default="false"/>
 

<rosparam>
  /camera/stereo_module/emitter_enabled: 0
</rosparam>
 
<rosparam if="$(arg emitter_enable)">
  /camera/stereo_module/emitter_enabled: 1
</rosparam>
 
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
[... 215 lines cut ...]
[ INFO] [1730102773.822249294]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1730102773.823609477]: gyro stream is enabled - fps: 200
[ INFO] [1730102773.823632515]: accel stream is enabled - fps: 200
[ INFO] [1730102773.823650244]: setupPublishers...
[ INFO] [1730102773.825180507]: Expected frequency for depth = 30.00000
[ INFO] [1730102773.843424917]: Expected frequency for infra1 = 30.00000
[ INFO] [1730102773.854231627]: Expected frequency for infra2 = 30.00000
[ INFO] [1730102773.864462827]: Expected frequency for color = 30.00000
[ INFO] [1730102773.875196168]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1730102773.885408199]: Start publisher IMU
[ INFO] [1730102773.886349606]: setupStreams...
 28/10 16:06:14,019 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 28/10 16:06:14,197 WARNING [140154711009024] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 28/10 16:06:14,402 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1730102774.453459897]: SELECTED BASE:Depth, 0
[ INFO] [1730102774.475609974]: RealSense Node Is Up!
[ WARN] [1730102774.525907144]: 
 28/10 16:06:14,526 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/10 16:06:14,577 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/10 16:06:14,628 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/10 16:06:14,679 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/10 16:06:14,818 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1730102774.822353546]: Hardware Notification:Motion Module failure,1.7301e+12,Error,Hardware Error
 28/10 16:06:14,868 WARNING [140154685830912] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
Is this part "Hardware Notification:Motion Module failure,1.7301e+12,Error,Hardware Error"what caused the problem? How can I solve this? Thanks !

dyb678, 2024-10-28:
[image]
This is my Firmware information.

MartyG-RealSense, 2024-10-28:
Hi @dyb678  It is not clear what causes a Motion Module Failure in the ROS wrapper when the camera is being accessed during launch.  

Does it still happen if you reset the camera at launch by adding **initial_reset:=true** to your roslaunch instruction, please?

dyb678, 2024-10-29:
yep，it still cann‘t work. I consider if it is caused by the version of fireware. Could u plz tell me its fireware version and its corresponding realsense version？And，because I have downloaded 5.16 version，how can I downgrade the version.

dyb678, 2024-10-29:
It should be noted that it works in realsense viewer And I can check rgb，depth ，gyro and accel data.But in ros launch imu topic has blank data

MartyG-RealSense, 2024-10-29:
Your current firmware is 5.16.0.1.  The recommended firmware to use with librealsense 2.50.0 and ROS1 wrapper 2.3.2 is 5.13.0.50. 

You can downgrade the firmware version to 5.13.0.50 in realsense-viewer using instructions at https://github.com/IntelRealSense/librealsense/issues/12599#issuecomment-1902680303

You should not make more than 20 firmware downgrade attempts though or the ability to downgrade permanently locks and you will only be able to upgrade to newer firmware versions than the one currently installed in your camera.

dyb678, 2024-10-30:
I have solved this problem and I can receive data.But I still have a problem: when I use “unite_imu_method =  linear_interpolation” I find the imu data is not stable. Do you have the method to solve this?

MartyG-RealSense, 2024-10-30:
Hi @dyb678  Setting unite_imu_method to **copy** instead of linear_interpolation can provide more stable IMU data.

MartyG-RealSense, 2024-11-15:
Case closed due to no further comments received.

barrylu457, 2024-11-30:
> Your current firmware is 5.16.0.1. The recommended firmware to use with librealsense 2.50.0 and ROS1 wrapper 2.3.2 is 5.13.0.50.
> 
> You can downgrade the firmware version to 5.13.0.50 in realsense-viewer using instructions at [IntelRealSense/librealsense#12599 (comment)](https://github.com/IntelRealSense/librealsense/issues/12599#issuecomment-1902680303)
> 
> You should not make more than 20 firmware downgrade attempts though or the ability to downgrade permanently locks and you will only be able to upgrade to newer firmware versions than the one currently installed in your camera.

I have the same problem: I use the D435i .  I have the topic"/camera/imu" in my rostopic list. But I cann't receive the data. I also checked  /camera/accel/sample, no data received.   I can see the IMU date on realsense-viewer.
my realsense2_camera 2.3.2 和librealsense v2.55.1, firmware version 5.16.0.1 . ubuntu20.04  /rosdistro: noetic  * /rosversion: 1.17.0. 

should I downgrade my firmware version or do some adjustment?  thank you very much.

[image]

MartyG-RealSense, 2024-11-30:
Hi @barrylu457  If you are using ROS1 Noetic then the ideal configuration will be librealsense 2.50.0, ROS1 wrapper 2.3.2 and firmware 5.13.0.50

I would recommend first downgrading to firmware 5.13.0.50 (but not any older than 5.13.0.50) and see if you can get IMU data without having to install librealsense 2.55.1 / wrapper 2.3.2 

If you still do not receive IMU data then you could uninstall librealsense 2.55.1 and wrapper 2.3.2, and then install correctly-matched 2.50.0 and 2.3.2 together at the same time using the package installation command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

************

May I also confirm with you that you are enabling the accel and gyro IMU topics?

`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`