realsenseai/realsense-ros #3237: No realsense device were found

amaraashishmahatokushwaha, 2024-10-28:
| Camera Model                       | D455f                        
| Firmware Version                   | 5.15.0.2                         
| Operating System & Version |   Ubuntu| 22.04
| Kernel Version (Linux Only)    | 5.15.136-tegra                                                            
| Platform                                 |NVIDIA Jetson Orin nano
| Librealsense SDK Version       |  2.55.1                                                    
| Segment			         |  robots, AMR                            
| ROS Distro			         |  humble                                   
| RealSense ROS Wrapper Version   | 4.51.1                                            

Hi @MartyG-RealSense ,
I hope you are doing well. i got stuck in this tried many possible ways you have mentioned in other issues too. but it doesnt work for me. Please help.

### Issue Description
I have installed librealsense from source and then ros wrapper from source too. and when i run realsense-viewer i get the output as stereo-module, rgb and other too. 
I have tried installing from libuvc but still it doesnt work. 
**But when i run or launch file from ros2 it shows error as no realsense device found.**
**Problem:**
Distro
[realsense2_camera_node-1] 01/01 10:45:33,664 WARNING [281472493676768] (d400-factory.cpp:1195) DS5 group_devices is empty.
[realsense2_camera_node-1] [WARN] [0000035133.664734880] [camera.camera]: No RealSense devices were found!

MartyG-RealSense, 2024-10-28:
Hi @amaraashishmahatokushwaha  I notice in the information at the top of this page that you are using librealsense SDK version 2.55.1, so the correct ROS wrapper version to match to it is **4.55.1**.  You have wrapper 4.51.1 installed though, which is designed for the earlier librealsense version 2.51.1.

As you installed the 4.51.1 ROS wrapper from source code, please completely delete the **/ros2_ws/src/** catkin workspace folder to uninstall the 4.51.1 wrapper and then install the 4.55.1 wrapper.  You will not need to install librealsense again, only the wrapper.

amaraashishmahatokushwaha, 2024-10-30:
Hi @MartyG-RealSense ,
Thank you for your observation regarding version and  reply. Actually that version compatibility is required to run on NVIDIA Isaac ros, you can go through [NVIDIA Isaac ROS Real-sense Setup Guide](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html). I have shifted to Nvidia isaac ros . I am currently running slam from isaac ros with the camera. And its working fine. If i need to run normally i'll remember your points to go ahead.
Thank you again.

MartyG-RealSense, 2024-10-30:
It's great to hear that you were successful.  Thanks very much for the update!

closed 2024-11-15 by MartyG-RealSense