realsenseai/realsense-ros #3242: Issue with setting TF namespace in ROS2 Realsense launch

Liu-Jinxin, 2024-11-03:
Hello,

I am trying to configure the namespace for TF in a ROS2 Realsense launch. While the launch file provides an option for setting the namespace, it appears to only affect the naming of camera topics and does not extend to the names of TF frames. In ROS1, this could be achieved using tf_prefix, and in ROS2, it seems it should be possible with frame_prefix, but this does not seem to be effective.

Here is the launch file configuration I am using:
```
configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'camera_namespace',             'default': 'top', 'description': 'namespace for camera'},
                           {'name': 'frame_prefix',                 'default': 'top', 'description': 'prefix for frame IDs'},
                           {'name': 'tf_prefix',                    'default': 'top', 'description': 'prefix for tf frame IDs'},

```
In ROS1, we could set tf_prefix to manage the frame names. How can we achieve similar functionality in ROS2? Any guidance or updates to the documentation would be greatly appreciated.

MartyG-RealSense, 2024-11-03:
Hi @Liu-Jinxin  There was a RealSense user at https://github.com/IntelRealSense/realsense-ros/issues/3119 who had problems with the naming of TF frames.  The approach they ended up taking at https://github.com/IntelRealSense/realsense-ros/issues/3119#issuecomment-2163458651 to achieve a working solution was to modify a source code file of the ROS2 wrapper version 4.55.1 called **tfs.cpp**

Liu-Jinxin, 2024-11-03:
Thanks @MartyG-RealSense for pointing me to #3119. I understand that the recommended approach involves modifying the source code of the ROS2 wrapper, specifically the tfs.cpp file, to effectively manage TF frame namespaces. While this method is effective, it might introduce maintenance challenges with future updates or custom configurations across different setups. 

Additionally, I came across issue #2888, which discusses the removal of the camera_name parameter and its impact on using frame_prefix in ROS2. I noticed in the description that frame_prefix seems like it should work properly. I would like to confirm whether my method of implementing frame_prefix might be incorrect, as it isn't having the desired effect.

MartyG-RealSense, 2024-11-04:
You should be able to freeze updates to the ROS wrapper and keep your edited wrapper in the same state long-term if you create a custom GitHub 'fork' of the wrapper and then edit the tfs.cpp file in that forked wrapper.

I do not have knowledge of the workings of frame_prefix, so I would recommend leaving a comment on https://github.com/IntelRealSense/realsense-ros/issues/2888 asking Arun-Presad-V about it.

MartyG-RealSense, 2024-11-15:
Hi @Liu-Jinxin  Do you require further assistance with this case, please?  Thanks!

Liu-Jinxin, 2024-11-15:
> Hi @Liu-Jinxin Do you require further assistance with this case, please? Thanks!

Thanks for your help, @MartyG-RealSense .

closed 2024-11-15 by Liu-Jinxin