realsenseai/realsense-ros #3243: PointCloud for local costmap

Cavalletta98, 2024-11-05:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                          | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.15.0                                                                     | 
| Platform                                 | PC                   |
| Librealsense SDK Version       |  2.55.1                                                 | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.55.1                                                  | 

### Issue Description
I'm trying to use the PointCloud from the camera to build a local costmap into Nav2. The plugin that i'm using is the voxel layer.
This is the nav2 configuration file:
[nav2.zip](https://github.com/user-attachments/files/17631702/nav2.zip)
The problem is that the costmap is empty even if there are obstacles in front of the camera.
Can someone could help me?

MartyG-RealSense, 2024-11-05:
Hi @Cavalletta98  There is a similar sounding /nav2 case at https://github.com/ros-navigation/navigation2/issues/3241 with an empty costmap that is not detecting obstacles.  Is there anything in that link that is helpful to you, please?

Cavalletta98, 2024-11-06:
Hi @MartyG-RealSense the discussion is very old. I noticed that, when i launch the nav2 stack, i'm getting this warning into the Realsense driver:
No stream match for pointcloud chosen texture Process - Color
Probably this is why i'm getting the empty costamp. Do you know how to solve?

MartyG-RealSense, 2024-11-06:
The message `No stream match for pointcloud chosen texture Process - Color` means that color frames are being dropped, typically because of instability of the color stream such as a fluctuating FPS speed.

Plese try adding the parameters **enable_infra1:=true pointcloud.stream_filter:=1** to your ROS2 launch instruction to texture the pointcloud with the infrared topic instead of the color topic to see whether the costmap can then be displayed.

`ros2 launch realsense2_camera rs_launch.py enable_infra1:=true pointcloud.stream_filter:=1`

Cavalletta98, 2024-11-06:
[realsense.zip](https://github.com/user-attachments/files/17646423/realsense.zip)
This is my configuration file for the camera. With this, the warning is no more present but still two problems:

1.  The rgb camera resolution is not the one that i specified: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
2. The costamp is still empty

MartyG-RealSense, 2024-11-06:
In your ROS launch log, does the line ** Device USB type** have 2.1 or 3.1 beside it?  If it is 2.1 then this would mean that the camera is being detected as being on a USB 2.1 connection.  

If you were trying to set the configuration 1280x720 at 30 FPS then this would be rejected as invalid by the launch on a USB 2.1 connection, as 2.1 only supports 6 FPS for that resolution on a D435i.

Cavalletta98, 2024-11-11:
I will make a try. In the meanwhile i discovered that when i launch nav2 and the slam, the cpu on the robot pc goes to 95% of usage. I'm now moving all the nav2 stuffs on an external pc. I will see if it will fix all the problems. I will let you know

MartyG-RealSense, 2024-11-11:
Thanks very much for the update.  I look forward to your next report.  Good luck!

Cavalletta98, 2024-11-14:
Hi @MartyG-RealSense i moved all the nav2 stuffs on another pc but still having the problem on empty costmap when using only the relasense camera. I shared with you the configuration used for the camera. Do you have some ideas? the same nav2 configuration is working with the simulated realsense
[realsense.zip](https://github.com/user-attachments/files/17750962/realsense.zip)
Moreover, there is no more the log that i sended to you on the camera driver

MartyG-RealSense, 2024-11-15:
I do not have any knowledge about nav2 or costmaps, unfortunately.

Cavalletta98, 2024-11-15:
No problem, thanks a lot

closed 2024-11-15 by Cavalletta98