realsenseai/realsense-ros #3244: [ROS1 WSL Ubuntu 20.04] Failed to open usb interface

Elektron97, 2024-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                        | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Ubuntu 20.04 (WSL) | 
| Platform                                 | Laptop (WSL)                         |
| ROS Distro			         |  Noetic                                         | 

In WSL2 I tried to launch:
```bash
roslaunch realsense2_camera rs_camera.launch
```
and I obtained this error:
```bash
[ INFO] [1730887040.305924387]:
 06/11 10:57:20,309 ERROR [140031503103744] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[ WARN] [1730887040.314327192]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1730887040.314409240]: The requested device with  is NOT found. Will Try again.
 06/11 10:57:20,312 ERROR [140031734470400] (sensor.cpp:572) acquire_power failed: failed to set power state
 06/11 10:57:20,314 WARNING [140031734470400] (rs.cpp:310) null pointer passed for argument "device"
```

I saw online that a file `99-realsense-libusb.rules` is required, so I pasted it from [here](https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules) as suggested in this [issue](https://github.com/IntelRealSense/realsense-ros/issues/1408) for Ubuntu 18.04. Unfortunately, it didn't work. Are there some different solutions for Ubuntu 20.04?

MartyG-RealSense, 2024-11-06:
Hi @Elektron97 I would not recommend using WSL2 with RealSense if possible.  It can be problematic to get it to work with librealsense, and getting it working with the RealSense ROS wrapper can be even more of a challenge.

In the past Microsoft have recommended using Windows 11 with WSL2 rather than Windows 10, as mentioned at https://github.com/IntelRealSense/realsense-ros/issues/2481#issuecomment-1250103682 - are you using Windows 10 or 11, please?

Elektron97, 2024-11-06:
HI @MartyG-RealSense , thank you for the answer. I'm currently using Windows 11.

Elektron97, 2024-11-10:
Thank you! I'm going to try in the next few days and I'll let you know asap!

MartyG-RealSense, 2024-11-11:
You are very welcome.  I look forward to your next report.  Good luck!

Elektron97, 2024-11-11:
I tried again with the firmware version suggested to me, downloaded [here](https://support.intelrealsense.com/hc/en-us/community/posts/6456192756371-Unable-download-firmware-5-13-0-50), because in the official page](https://www.intel.com/content/www/us/en/download/19242/firmware-for-intel-realsense-d400-product-family.html) is not listed.

However, I have the same error:
[image]
[image]

MartyG-RealSense, 2024-11-11:
Further advice about getting the camera working with WSL2 on Windows 11 can be found at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744

Elektron97, 2024-11-11:
I tried but still not working :(. I changed also the USB cable into a 3.2 USB cable. I paste the new settings from the realsense viewer.
[image]

To be sure, I pasted here also the procedure to link the device with WSL.
*Command Prompt*
```bash
usbipd bind --busid 2-13
usbipd attach --wsl --busid 2-13
# output
# usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions.
# usbipd: info: Using IP address 192.168.96.1 to reach the host.
```

In the ubuntu terminal
```bash
lsusb
# Bus 002 Device 003: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
# Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
# Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

MartyG-RealSense, 2024-11-12:
You are very welcome.  It's great to hear that you were successful.  Thanks very much for the update and for sharing the solution that worked for you!