realsenseai/realsense-ros #3245: Libusb Errors Leading to Node Crash

BryanBetancur, 2024-11-07:
<p dir="auto">Hello</p>

Required Info |  
-- | --
Camera Model | D435i
Firmware Version | 5.16.0.1
Operating System | Ubuntu 20.04.6 LTS
Platform | NVIDIA Jetson Orin AGX
Jetpack | 5.1.1 [L4T 35.3.1]
Kernel Version (Linux Only) | 5.10.104-tegra
Librealsense SDK Version | 2.56.1
ROS Distro | Humble
RealSense ROS Wrapper Version | 4.56.1

### Issue Description
After enabling and disabling the camera multiple times, the following errors occur randomly, causing the node to crash and subsequently respawn. I've documented three different cases (but all related with libusb), which I'll outline below:

1. **libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling:**
```
[nav_bottom-5] [INFO] [1730918083.523281362] [gary.cameras.nav_bottom]: Starting Sensor: RGB Camera
[nav_bottom-5] [INFO] [1730918083.611428773] [gary.cameras.nav_bottom]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[nav_bottom-5] 06/11 18:34:43,692 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:43,703 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:43,716 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:43,727 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:43,738 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:43,750 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:44,484 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:44,497 WARNING [281472913082672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_bottom-5] 06/11 18:34:59,916 ERROR [281472135014704] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[nav_bottom-5] 06/11 18:35:14,600 ERROR [281472135014704] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[nav_bottom-5] 06/11 18:35:15,938 ERROR [281472135014704] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[nav_bottom-5] 06/11 18:35:33,958 ERROR [281472135014704] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[nav_bottom-5] [INFO] [1730918139.783360474] [gary.cameras.nav_bottom]: Stopping Sensor: Depth Module
[nav_bottom-5] [INFO] [1730918139.783490012] [gary.cameras.nav_bottom]: Stop Sensor: Depth Module
[nav_bottom-5] [INFO] [1730918139.783534493] [gary.cameras.nav_bottom]: Close Sensor. 
[nav_bottom-5] [INFO] [1730918140.021006423] [gary.cameras.nav_bottom]: Close Sensor - Done. 
[nav_bottom-5] [INFO] [1730918140.065551972] [gary.cameras.nav_bottom]: Stopping Sensor: RGB Camera
[nav_bottom-5] [INFO] [1730918140.065680006] [gary.cameras.nav_bottom]: Stop Sensor: RGB Camera
[nav_bottom-5] [INFO] [1730918140.066495860] [gary.cameras.nav_bottom]: Close Sensor. 
[nav_bottom-5] [INFO] [1730918141.384667244] [gary.cameras.nav_bottom]: Close Sensor - Done. 
[nav_top-3] 06/11 18:36:08,472 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:37:17,589 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:37:22,124 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:37:30,129 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] [INFO] [1730918275.221905562] [gary.cameras.nav_top]: Stopping Sensor: RGB Camera
[nav_top-3] [INFO] [1730918275.222154910] [gary.cameras.nav_top]: Stop Sensor: RGB Camera
[nav_top-3] [INFO] [1730918275.222243456] [gary.cameras.nav_top]: Close Sensor. 
[nav_top-3] [INFO] [1730918276.413789795] [gary.cameras.nav_top]: Close Sensor - Done. 
[nav_top-3] [INFO] [1730918399.482503770] [gary.cameras.nav_top]: Stopping Sensor: RGB Camera
[nav_top-3] [INFO] [1730918399.517535260] [gary.cameras.nav_top]: Starting Sensor: RGB Camera
[nav_top-3] [INFO] [1730918399.781650015] [gary.cameras.nav_top]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[nav_top-3] 06/11 18:40:00,113 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:40:01,670 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:42:28,975 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:46:44,162 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:46:45,295 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:47:48,876 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:48:08,423 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:49:04,464 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:49:18,140 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:52:56,703 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:54:14,288 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:59:02,563 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 06/11 18:59:04,833 WARNING [281472225208624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[LAUNCH ERROR] [nav_top-3.launch]: process has died [pid 72794, exit code -11, cmd '/robot/additional_ros_packages/install/lib/realsense2_camera/realsense2_camera_node --ros-args -r __node:=nav_top -r __ns:=/gary/cameras --params-file /robot/build/ros_pkgs/install/gary_cameras/share/gary_cameras/config/gary_cameras.yaml --params-file /robot/config/general.config.yaml --params-file /robot/config/cameras/cameras.config.yaml -r /gary/cameras/nav_top/depth/color/points:=/gary/cameras/nav_top/points'].
[LAUNCH WARNING] [nav_top-3.launch]: Respawning...
[LAUNCH INFO] [nav_top-3.launch]: process started with pid [1640333]
[nav_top-3] [INFO] [1730919589.058724861] [gary.cameras.nav_top]: RealSense ROS v4.56.1
[... 172 lines cut ...]
[nav_bottom-5] [INFO] [1730917843.719654597] [gary.cameras.nav_bottom]: Device with serial number 044322073670 was found.
[... cut ...]
[nav_bottom-5] [WARN] [1730917852.117824405] [gary.cameras.nav_bottom]: re-enable the stream for the change to take effect.
[nav_bottom-5] [INFO] [1730917852.267727667] [gary.cameras.nav_bottom]: RealSense Node Is Up!

```

The only thing I’ve been able to identify is that it happens after the logs of `(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`, which occur quite frequently. I would appreciate it if you could guide me on how to resolve this issue.

MartyG-RealSense, 2024-11-07:
Hi @BryanBetancur  Does it make a difference if you add **initial_reset:=true** to the launch instruction to reset the camera at launch, please?

BryanBetancur, 2024-11-07:
Thanks for your quick response, I will check it during the day.

BryanBetancur, 2024-11-07:
It continues happening with the **initial_reset:=true**.
Last time I got:
```
[nav_top-3] [INFO] [1730996836.209927038] [gary.cameras.nav_top]: Starting Sensor: RGB Camera
[nav_top-3] [INFO] [1730996836.426728644] [gary.cameras.nav_top]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[nav_top-3] 07/11 16:27:16,726 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:27:16,737 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:27:16,752 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:27:17,647 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:28:07,483 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:29:08,795 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:32:05,723 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:32:21,334 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:33:02,965 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:34:43,170 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:35:13,259 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:35:45,483 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:39:41,320 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:40:24,819 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:40:36,028 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:40:50,571 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:41:48,746 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] [INFO] [1730997726.231472188] [gary.cameras.nav_top]: Stopping Sensor: RGB Camera
[nav_top-3] [INFO] [1730997726.231711840] [gary.cameras.nav_top]: Stop Sensor: RGB Camera
[nav_top-3] [INFO] [1730997726.231785729] [gary.cameras.nav_top]: Close Sensor. 
[nav_top-3] [INFO] [1730997727.742850137] [gary.cameras.nav_top]: Close Sensor - Done. 
[nav_top-3] [INFO] [1730997870.889564946] [gary.cameras.nav_top]: Stopping Sensor: RGB Camera
[nav_top-3] [INFO] [1730997870.910164796] [gary.cameras.nav_top]: Starting Sensor: RGB Camera
[nav_top-3] [INFO] [1730997871.131990332] [gary.cameras.nav_top]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[nav_top-3] 07/11 16:44:31,462 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:45:51,873 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:46:40,175 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:46:58,256 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:49:19,627 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:50:47,689 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nav_top-3] 07/11 16:54:45,862 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[nav_top-3] [WARN] [1730998766.110442700] [gary.cameras.nav_top]: re-enable the stream for the change to take effect.
[nav_top-3] [INFO] [1730998766.333801798] [gary.cameras.nav_top]: Stopping Sensor: RGB Camera
[nav_top-3] [INFO] [1730998766.425976218] [gary.cameras.nav_top]: Starting Sensor: RGB Camera
[nav_top-3] [INFO] [1730998766.481471697] [gary.cameras.nav_top]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[nav_top-3] [INFO] [1730998766.550470829] [gary.cameras.nav_top]: RealSense Node Is Up!
```

MartyG-RealSense, 2024-11-08:
Would it be possible for you to check Ubuntu's **htop** system monitoring tool to see whether the available memory capacity of the computer reduces after each start-stop?  If it does then this is called a 'memory leak' and once enough of the memory has been consumed then a program can become unstable or freeze / exit.

MartyG-RealSense, 2024-11-09:
Are the cameras plugged into USB ports on the computer or plugged into a USB hub?  The ports on a computer should have individual USB controllers for each port.  On a hub though, a single USB controller might be processing two of the hub's ports simultaneously.

BryanBetancur, 2024-11-11:
I initially had both cameras connected to a USB hub, but I also tested connecting them directly to the Jetson, carefully selecting ports to avoid internal Jetson hubs. Unfortunately, the issue still occurs under both setups.

Additionally, I tried connecting only one camera. While this setup took longer to fail, the issue eventually reappeared.

As another test, I activated the color, depth, and point cloud streams for one camera without performing continuous stop/start cycles—simply enabling them once and leaving them active. Even in this configuration, the problem reemerged after several hours with the difference of after the node respawn the camera doesn't stream correctly, the node shows the prints of the sensor starts but never send frames.

Let me know if there are any other troubleshooting steps I should try.

MartyG-RealSense, 2024-11-12:
Does the launch have less problems if you use **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true`

BryanBetancur, 2024-11-12:
Hi Marty,

It seems the issue is related to setting parameters. I conducted the following tests:

First Test:

- I set up three different Jetson Orin devices, each with two Realsense cameras. In two of the Jetsons, the cameras were connected via USB hubs, while in the third, the cameras were connected directly.
- I launched two Realsense nodes (one per camera) with color, depth, and pointcloud sensors initially set to `false` After the nodes launched successfully, I enabled the three sensors by setting parameters (color, depth, and pointcloud).
- **Result**: After a few hours, the libusb error appeared on all three Jetsons.

Second Test:

- I set up the same three Jetson Orin devices with the same Realsense camera setup (two cameras per Jetson, two using USB hubs, and one with direct connections).
- This time, I launched the Realsense nodes with color, depth, and pointcloud sensors set to `true` from the start.
- **Result**: After 12 hours, the libusb error did not occur on any of the Jetsons.

It appears that the issue might be triggered by enabling the sensors dynamically, rather than having them set to `true` from the beginning. Let me know if you need further details on the setup or if there’s anything else you'd like me to try.

closed 2024-12-07 by MartyG-RealSense