realsenseai/realsense-ros #3247: No IMU data from

hellovuong, 2024-11-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked at our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455f                          | 
| Firmware Version                   | 5.15.0.2                      | 
| Operating System & Version |   Ubuntu 20.04
| Kernel Version (Linux Only)    |  5.10.104-tegra                                                                       | 
| Platform                                 | NVIDIA Jetson                           |
| Librealsense SDK Version       |   2.54.1                                                    | 
| Language                               |  C++         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   |  ROS1-legacy (branch newest commit) run in Docker container   | 

### Issue Description
I installed SDK in Dockerfile as follows:
```bash
apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE

add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u && \
    apt-get update && \
    apt-get install librealsense2=2.54.1-0~realsense.1425 -y && \
    apt-get install -y librealsense2-gl=2.54.1-0~realsense.1425 librealsense2-dev=2.54.1-0~realsense.1425 librealsense2-utils=2.54.1-0~realsense.1425

```
The ROS1 wrapper is from ROS1-legacy with config as follows:
```bash
    <include file="$(find realsense2_camera)/launch/rs_aligned_depth.launch" >
        <arg name="enable_fisheye"            value="false"/>

        <arg name="depth_width"               value="848"/>
        <arg name="depth_height"              value="480"/>

        <arg name="enable_infra1"             value="false"/>
        <arg name="enable_infra2"             value="false"/>

        <arg name="color_width"               value="848"/>
        <arg name="color_height"              value="480"/>

        <arg name="accel_fps"                 value="200"/>

        <arg name="enable_pointcloud"         value="true"/>
    </include>
```
I checked the compatibility of SDK and firmware in https://dev.intelrealsense.com/docs/firmware-releases-d400 and it should be correct.
However, I can not receive IMU data. Can you help me to debug where is the problem?
Here is the output of realsense launch:

```bash
[ INFO] [1730983723.593692733]: Device with serial number 309622300196 was found.

[ INFO] [1730983723.593765600]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.0/video4linux/video1 was found.
[ INFO] [1730983723.593791201]: Device with name Intel RealSense D455F was found.
[ INFO] [1730983723.594373719]: Device with port number 2-2 was found.
[ INFO] [1730983723.594432506]: Device USB type: 3.2
[ INFO] [1730983723.600057936]: getParameters...
[ INFO] [1730983723.753303555]: setupDevice...
[ INFO] [1730983723.753416648]: JSON file is not provided
[ INFO] [1730983723.753492682]: ROS Node Namespace: camera
[ INFO] [1730983723.753525228]: Device Name: Intel RealSense D455F
[ INFO] [1730983723.753545132]: Device Serial No: 309622300196
[... 24 lines cut ...]
 07/11 12:48:44,406 WARNING [281472141349232] (ds-calib-parsers.cpp:41) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 07/11 12:48:44,410 WARNING [281472116171120] (backend-hid.h:68) write_fs_attribute Could not change accel_3d-dev15 to accel_3d-dev5 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/trigger/current_trigger
 07/11 12:48:44,414 WARNING [281472023916912] (backend-hid.h:68) write_fs_attribute Could not change gyro_3d-dev16 to gyro_3d-dev6 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/trigger/current_trigger
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
[ INFO] [1730983724.465908134]: SELECTED BASE:Depth, 0
[ INFO] [1730983724.492234418]: RealSense Node Is Up!
[ INFO] [1730983724.666834067]: publishing unordered pointcloud.
[ WARN] [1730983724.799555608]: No stream match for pointcloud chosen texture Process - Color
[ WARN] [1730983726.321769809]: No stream match for pointcloud chosen texture Process - Color
 07/11 12:48:49,422 WARNING [281472015524208] (backend-hid.cpp:633) iio_hid_sensor: Frames didn't arrived within the predefined interval
 07/11 12:48:49,422 WARNING [281472007131504] (backend-hid.cpp:633) iio_hid_sensor: Frames didn't arrived within the predefined interval

```
Out of those log I think this is interesting:
```bash
 07/11 12:48:44,410 WARNING [281472116171120] (backend-hid.h:68) write_fs_attribute Could not change accel_3d-dev15 to accel_3d-dev5 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/trigger/current_trigger
 07/11 12:48:44,414 WARNING [281472023916912] (backend-hid.h:68) write_fs_attribute Could not change gyro_3d-dev16 to gyro_3d-dev6 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/trigger/current_trigger
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
```

hellovuong, 2024-11-07:
Additionally, If I installed the SDK from source with parameters `-DFORCE_RSUSB_BACKEND=true` the error:
 ```bash
 07/11 12:48:44,410 WARNING [281472116171120] (backend-hid.h:68) write_fs_attribute Could not change accel_3d-dev15 to accel_3d-dev5 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/trigger/current_trigger
 07/11 12:48:44,414 WARNING [281472023916912] (backend-hid.h:68) write_fs_attribute Could not change gyro_3d-dev16 to gyro_3d-dev6 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/trigger/current_trigger
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
 07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
```
NOT appear any more but the frequency of color image is very low like only ~3Hz

MartyG-RealSense, 2024-11-07:
Hi @hellovuong  The message `No stream match for pointcloud chosen texture Process - Color` indicates that color frames are missing because they are being dropped.

The IMU topics are disabled by default in the ROS wrapper.  **enable_accel** and **enable_gyro** can be set to true in the launch file if you have not done so already, or you can enable them in the roslaunch by adding to the launch instruction the following parameters:

`enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

The unite_imu_method combines the two separate /accel and /gyro topics into a single combined `/imu` topic.

hellovuong, 2024-11-08:
Hi @MartyG-RealSense, Sorry I copied the wrong launch file. In the correct launch file, I did enable Gyro and accel but camera did not publish any data. I am pretty sure this warning are relevant but could not understand why it happened:
```bash
07/11 12:48:44,410 WARNING [281472116171120] (backend-hid.h:68) write_fs_attribute Could not change accel_3d-dev15 to accel_3d-dev5 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/trigger/current_trigger
07/11 12:48:44,414 WARNING [281472023916912] (backend-hid.h:68) write_fs_attribute Could not change gyro_3d-dev16 to gyro_3d-dev6 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/trigger/current_trigger
07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
07/11 12:48:44,417 WARNING [281472124563824] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200073.2.auto/iio:device15/buffer/enable
07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
07/11 12:48:44,418 WARNING [281472107778416] (backend-hid.cpp:717) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-2/2-2:1.5/0003:8086:0B5C.0043/HID-SENSOR-200076.3.auto/iio:device16/buffer/enable
```

MartyG-RealSense, 2024-11-08:
As you have a D455f and a Jetson, is your Jetson using JetPack 6 please?  

JetPack 6 removed a function called **hiddraw** that RealSense cameras equipped with an IMU (known as HID devices) make use of and so these cameras will not work with JP6.  

Possible fixes for this JP6 IMU issue include building librealsense from source code with RSUSB = true - as you found earlier at https://github.com/IntelRealSense/realsense-ros/issues/3247#issuecomment-2462175011 - or downgrading the JetPack version to **5.1.3**.

Is the 3 Hz frequency that you experienced with the color stream improved if you add the parameter **auto_exposure_priority:=false** to your roslaunch instruction?

hellovuong, 2024-11-08:
- I used JetPack 5.1.1 [L4T 35.3.1]. I didn't add to my comment  [#3247 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/3247#issuecomment-2464216080) that if I build from source with  RSUSB = true, the imu works perfectly just that the camera only send data at 3Hz
- I can try your suggestion auto_exposure_priority:=false and give you an update on Monday
Thanks!

MartyG-RealSense, 2024-11-08:
You are very welcome.  I look forward to your next update.  Good luck!

hellovuong, 2024-11-11:
Unfortunately, the parameter auto_exposure_priority:=false doesn't help to improve the fps of the color image.

MartyG-RealSense, 2024-11-11:
How does color perform if you use the roslaunch instruction below?

`roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=pointcloud enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation color_width:=848 color_height:=480 color_fps:=30`

hellovuong, 2024-11-11:
I ran couple of tests through out today: with SDK build from source with  RSUSB = true:
- `roslaunch realsense2_camera rs_aligned_depth.launch enable_infra1:=false enable_infra2:=false enable_fisheye:=false enable_pointcloud:=false accel_fps:=200 color_width:=848 depth_width:=848 enable_gyro:=true  enable_accel:=true` -> gyro/accel/color/depth works in correct fps 
-  `roslaunch realsense2_camera rs_aligned_depth.launch enable_infra1:=false enable_infra2:=false enable_fisheye:=false enable_pointcloud:=true accel_fps:=200 color_width:=848 depth_width:=848 enable_gyro:=true  enable_accel:=true` -> Enable pointcloud, no color or gyro/accel sends only depth image_raw and pointcloud are available
-  `roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=pointcloud enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation color_width:=848 color_height:=480 color_fps:=30` -> (your recommendation) result same as the 2nd command

hellovuong, 2024-11-11:
More detail output when I run this configuration:
`roslaunch realsense2_camera rs_aligned_depth.launch enable_infra1:=false enable_infra2:=false enable_fisheye:=false enable_pointcloud:=true accel_fps:=200 color_width:=848 depth_width:=848 enable_gyro:=true  enable_accel:=true`
color image had published for some seconds until the console log printed out:
` 11/11 14:55:18,669 ERROR [281472049062256] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132`
then no more color image :(

MartyG-RealSense, 2024-11-11:
I recommend not using rs_aligned_depth.launch due to the very old age of that file (over 5 years).  rs_camera.launch would be preferable.  However, when rs_camera.launch has problems with publishing a pointcloud, a method that often works is to use the `rs_rgbd.launch` file instead.

First you will need to install support for an RGBD launch on Noetic using the command below:

`sudo apt-get install ros-noetic-rgbd-launch`

Then launch rs_rgbd.launch with the following instruction:

`roslaunch realsense2_camera rs_rgbd.launch enable_accel:=true enable_gyro:=true color_width:=848 color_height:=480 color_fps:=30`

You do not need to enable the pointcloud and alignment with this launch file as they are already enabled by default.  This launch file also does not have the `unite_imu_method` parameter and will publish the IMU data to separate /accel and /gyro topics instead of combining them into a single /imu topic.

With the rs_rgbd launch file, the pointcloud is published to **depth_image_proc** instead of /camera/depth/color/points

hellovuong, 2024-11-13:
Hi, Thank you for your suggestion. I figured out that this problem can be related to our kernel, I will close the issue and investigate internally. Thank you again for your active support!

closed 2024-11-13 by hellovuong

MartyG-RealSense, 2024-11-13:
You are very welcome.  I'm glad to help.  Thanks very much for the update!