realsenseai/realsense-ros #3249: ROS Node Parameters set is success even if there was an error in the value

edgarcamilocamacho, 2024-11-11:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435 / D435i                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Ubuntu 20.04.6 LTS | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                                                       | 
| Platform                                 | NVIDIA Jetson AGX                           |
| Librealsense SDK Version       |  2.56.1                                                    | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.56.1                                                 | 

### Issue Description

I wanted to change the format of the RGB camera and did a typo like:

```
$ ros2 param set /gary/cameras/nav_bottom rgb_camera.color_format "BRG8"
Set parameter successful
```

As you can see the node responds that the setting of the parameter was success, but the logs of the camera's node says:

```
[nav_bottom-5] [ERROR] [1731349854.170183534] [gary.cameras.nav_bottom]: color stream doesn't support BRG8 format. Run 'ros2 param describe <your_node_name> rgb_camera.color_format' to get the list of supported formats. Setting the ROS param 'rgb_camera.color_format' back to: BGR8
```

If the parameter is not ok and the node itselfs has knowledge about it, it should return FAILED when trying to set the parameter, no?

MartyG-RealSense, 2024-11-12:
Hi @edgarcamilocamacho  Technically, RealSense cameras (and the librealsense SDK) support the BGR8 color format.  However, the RealSense ROS wrapper - which runs on top of librealsense - only provides the streams in one format.  For the color stream, that default format is RGB8. 

In 2021 a RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/1670 edited the source code of the ROS wrapper's **base_realsense_node.cpp** file to change the supported default format for a particular stream type.

edgarcamilocamacho, 2024-11-12:
But what I mean is that I made a typo and I didin't write "BGR" but "BRG", that's the point.

MartyG-RealSense, 2024-11-12:
Thanks for the clarification.  It seems like a problem with ROS2 rather than the RealSense ROS wrapper, as ROS2 allows it to register as successful but the wrapper catches the error.

closed 2024-12-07 by MartyG-RealSense