realsenseai/realsense-ros #3250: "Device is already streaming!" and no more frames after that

edgarcamilocamacho, 2024-11-11:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435 / D435i                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Ubuntu 20.04.6 LTS | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                                                       | 
| Platform                                 | NVIDIA Jetson AGX                           |
| Librealsense SDK Version       |  2.56.1                                                    | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.56.1                                                 | 

### Issue Description

When I try to enable all color, depth and pointcloud using the ROS parameters, sometimes (really random, I can't not have a way to control the replication) the node shows:

```

[INFO] [1731361496.428292897]: Stopping Sensor: RGB Camera
[INFO] [1731361496.438579189]: Starting Sensor: RGB Camera
[INFO] [1731361496.457483599]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[WARN] [1731361497.014352851]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[INFO] [1731361497.430367000]: Stopping Sensor: Depth Module
[INFO] [1731361497.438439322]: Starting Sensor: Depth Module
11/11 21:45:00,473 ERROR [281471772231984] (rs.cpp:255) [rs2_open_multiple( sensor:0xffff44738c60, profiles:0xffff447eb650, count:1 ) Invalid Value]
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361500.473344938]: /robot/additional_ros_packages/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:494:An exception has been thrown:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361500.473530380]: /robot/additional_ros_packages/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:376:An exception has been thrown:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361500.473603756]: Error updating the sensors:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[INFO] [1731361501.832015361]: Stopping Sensor: Depth Module
[INFO] [1731361501.847421727]: Starting Sensor: Depth Module
11/11 21:45:04,482 ERROR [281471772231984] (rs.cpp:255) [rs2_open_multiple( sensor:0xffff44738c60, profiles:0xffff447eb650, count:1 ) Invalid Value]
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361504.482235307]: /robot/additional_ros_packages/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:494:An exception has been thrown:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361504.482449453]: /robot/additional_ros_packages/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:376:An exception has been thrown:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[ERROR] [1731361504.482594702]: Error updating the sensors:
Failed to resolve the request:
Format: Z16, width: 848, height: 480
Into:
Device is already streaming!
[INFO] [1731361504.482739887]: Stopping Sensor: Depth Module
11/11 21:46:56,897 WARNING [281471604476208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

You know it's going to fail if you see `frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.`.

After those prints, the node never responds anymore the parameters update requests, and does not publish frames anymore. But the linux process itseld keeps alive.

Here my node's logs to understand times of setting parameters:

```
[INFO ] [1731361496.427011158]: Enabling camera!
[INFO ] [1731361496.427159799]: Setting Realsense Camera: color=true
[INFO ] [1731361496.432001023]: Parameter update request sent successfully
[INFO ] [1731361497.428737963]: Setting Realsense Camera: depth=true
[INFO ] [1731361497.441062895]: Parameter update request sent successfully
[INFO ] [1731361498.428722673]: Setting Realsense Camera: pointcloud=true
[INFO ] [1731361498.435508073]: Parameter update request sent successfully
[INFO ] [1731361498.435908396]: Waiting for frames...
[INFO ] [1731361500.547826753]: No depth frames received, trying again...
[INFO ] [1731361501.830840760]: Setting Realsense Camera: depth=true
[INFO ] [1731361501.834371221]: Parameter update request sent successfully
[INFO ] [1731361501.835908574]: Waiting for frames...
[INFO ] [1731361502.218599604]: No depth frames received, trying again...
[INFO ] [1731361502.378495026]: Setting Realsense Camera: depth=true
[INFO ] [1731361502.468394856]: Parameter update request sent successfully
[INFO ] [1731361502.673894957]: Waiting for frames...
[INFO ] [1731361503.436133466]: No depth frames received, trying again...
[INFO ] [1731361504.330695978]: Setting Realsense Camera: depth=true
[INFO ] [1731361504.485619206]: Parameter update request sent successfully
[ERROR] [1731361504.930669032]: Timeout!
```

MartyG-RealSense, 2024-11-12:
Hi @edgarcamilocamacho  Does it make a difference to whether the busy state ('already streaming') occurs if the camera is reset during launch by adding **initial_reset:=true**, as resetting the camera should disconnect and then reconnect it.

edgarcamilocamacho, 2024-11-12:
It already was in true at the moment of the testing

MartyG-RealSense, 2024-11-13:
Looking at your mention of knowing it will always fail if the message 'frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically' is received, does it make a difference if you disable a timestamping feature called 'Global Time' (which is enabled by default) by adding `global_time_enabled:=false` to your launch instruction?

closed 2024-12-01 by MartyG-RealSense