realsenseai/realsense-ros #3251: I can't get aligned depth images of my D435i from another PC with the same ROS_DOMAIN_ID.

iotakataoka, 2024-11-15:
Hello

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                           | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |  Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  6.8.0-48                                                                       | 
| Platform                                 | PC                          |
| Librealsense SDK Version       |  2.55.1                                                    | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.55.1                                                 | 

### Issue Description
I cannot get an aligned depth image from another PC with the same ROS_DOMAIN_ID.

### Actual Behavior
When viewed from a PC that does not have a realsenseD435i connected but has the same ROS_DOMAIN_ID, the `/camera/camera/aligned_depth_to_color/image_raw` topic is always published, but new values ​​are not sent only when rviz2 is running. /camera/camera/color/image_raw is sent even while rviz2 is running, and the image in rviz is also updated.
The command I'm using is: `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true pointcloud.enable:=true filters:=spatial initial_reset:=true` 
I also tried removing rgb_camera.profile and rgb_camera.profile, but there was no change.
I set rviz's Reliability Policy to Best Effort.

### Expected Behavior
I want to obtain an aligned depth image and a pointcloud and visualize them with rviz2.

iotakataoka, 2024-11-15:
It works fine on a PC connected to realsense as shown below

but it doesn't work well on a PC connected via DDS as shown below.

MartyG-RealSense, 2024-11-15:
Hi @iotakataoka  A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/3072#issuecomment-2060686363 who was using DDS and not getting an output from the aligned topics found that their problem was because the aligned images had higher bandwidth and their network could not handle it.  Is this case relevant to your own problem, please?

iotakataoka, 2024-11-15:
This is certainly similar to my problem. It's possible that the network can't handle large amounts of data. Could lowering the resolution or compressing the data be a solution?

MartyG-RealSense, 2024-11-15:
You might get better performance from compressed depth topics so long as you do not use the **compressedDepth** topic which has a much higher data size than the other compressed topics, as discussed at https://github.com/IntelRealSense/realsense-ros/issues/3105#issuecomment-2147886791

A list of the compressed topics available from the RealSense ROS2 wrapper can be found at the link below.

https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

iotakataoka, 2024-11-15:
I tried subscribing to the `/camera/camera/aligned_depth_to_color/image_raw/compressed` topic and `/camera/camera/aligned_depth_to_color/image_raw/compressed` topic using the `image-transport-plugin` package, but I was unable to receive the data properly, as shown in the image below.
[image]

I also confirmed that an error message was displayed stating ```[ERROR] [1731688823.753100743] [CompressedPublisher]: [16UC1] is not a color format. but [mono8] is. The conversion does not make sense```  The solution to this error was written in the location you showed me earlier, but I couldn't figure out how to set it up.

MartyG-RealSense, 2024-11-16:
Are you able to set RViz to use compressed data instead of raw data on its 'Create visualization' window as shown at https://github.com/IntelRealSense/realsense-ros/issues/1903#issuecomment-898522352 please?

iotakataoka, 2024-11-16:
The image above shows an attempt to display `/camera/camera/color/image_raw/compressed` and `/camera/camera/aligned_depth_to_color/image_raw/compressed` in this way.
[image]

MartyG-RealSense, 2024-11-16:
Looking at related cases including https://github.com/IntelRealSense/realsense-ros/issues/3105#issuecomment-2147886791 the users in those cases may be configuring their image_transport settings using the **rqt_reconfigure** settings interface, which can be launched with the commabd below

`ros2 run rqt_reconfigure rqt_reconfigure`

A ROS2 tutorial for image_transport provides information about configuring through this interface.

https://github.com/ros-perception/image_transport_tutorials?tab=readme-ov-file#changing-transport-specific-behavior

Specific settings can also be configured directly with **ros2 param set** commands like the example below.

`ros2 param set /camera/camera .aligned_depth_to_color.image_raw.compressed.format png`

iotakataoka, 2024-11-17:
I was able to change the format to png with `ros2 param set`. 

`$ ros2 param get /camera/camera .camera.color.image_raw.format`
```String value is: png```

`$ ros2 param get /camera/camera .camera.aligned_depth_to_color.image_raw.format`
```String value is: png```

As a result, the error ```[CompressedPublisher]: [16UC1] is not a color format. but [mono8] is. The conversion does not make sense``` disappeared, but it was not displayed in rviz. 
Also, the problem may be that data is not flowing to `/camera/camera/color/image_raw/camera_info`, as shown in the image below.

By the way, I found that if I stopped displaying the two types of camera data in rviz, the point cloud would appear and move.

[image]

MartyG-RealSense, 2024-11-17:
Thanks very much for the update.  I'm pleased to hear that you made significant progress.

Does not disp;aying the two types of camera data in order to make the pointcloud work cause problems for you, please?

iotakataoka, 2024-11-17:
There is a slight lag in the display of the point cloud, but this does not affect the topic itself. 
It is a little inconvenient that the images are not displayed at the same time, but I don't think it is a big problem.

MartyG-RealSense, 2024-11-17:
Ok, thanks very much.  It's good that this is a workable solution for you.

MartyG-RealSense, 2024-11-23:
Hi @iotakataoka  Do you require further assistance with this case, please?  Thanks!

iotakataoka, 2024-11-23:
The problem has been resolved for now. I will ask again if any other problems occur. Thank you.

closed 2024-11-23 by iotakataoka