realsenseai/realsense-ros #3254: slam with d455

karimrahimi76, 2024-11-17:
Hi all , 
i am trying to develop slam with intelRealsense camera d455 for my robot 
i converted the 3d point cloud data to 2d lidar data using depth_to_laser package 
now i want to develop nav2 but i am confused
for localization , i think amcl is not good choice because realsense han not 360 degree fov and amcl may not work correctly 
 also for odometry data i have wheel encoder and i want to fuse it with realsense imu data 
 is it good to fuse realsense imu with odometry for get accurate odometry or use something like rgdb odometry from rtabmap package 
 also for localization in nav2 should i use robot localization package instead of amcl ?
 and what  if i use robot localization , what about map->odom tf 
 is robot localization publish it correctly ? 
 thank you all

MartyG-RealSense, 2024-11-17:
Hi @karimrahimi76   Thanks very much for your questions.

I am not familiar with using nav2 unfortunately, so do not have advice to provide about questions involving nav2.  The recent nav2 related cases at https://github.com/IntelRealSense/librealsense/issues/13468 and https://github.com/IntelRealSense/realsense-ros/issues/3243 may be helpful references for you though.

RealSense 400 Series stereo depth cameras that have an IMU are not usually used with wheel encoders.  There was a now-retired camera product called the RealSense T265 Tracking Camera that supported that feature.

However, the link below has a guide to using a RealSense D435i with IMU data and wheel encoding.  The information should be applicable to the D455 camera model.

https://github.com/fazildgr8/realsense_explorer_bot

robot_localization should be able to publish odom, yes - please see https://github.com/IntelRealSense/realsense-ros/issues/3173

MartyG-RealSense, 2024-11-25:
Hi @karimrahimi76  Do you require further assistance with this case, please?  Thanks!

karimrahimi76, 2024-11-25:
Hi Marty ,
thank you for your reply , i didn't test my nav2 yet , i will test it with
realsense and robot localization package and let you know about it

MartyG-RealSense, 2024-12-07:
Hi @karimrahimi76  Do you have an update about this case that you can provide, please?  Thanks!

karimrahimi76, 2024-12-10:
Hi @MartyG-RealSense  , thank you for your reply , i tested the realsense d455 , depth to laser scan and now the mapping is ok , i just some problems with rgbd odometry that i think i should change it to use external imu and wheel odom and then start testing the nav2 , when i test that certainly come back and let you know about that

closed 2025-01-26 by MartyG-RealSense