realsenseai/realsense-ros #3255: dynamically set the exposure mode and exposure parameters in my RealSense ROS program

JasonWang9805, 2024-11-17:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {  **D435i**  }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Linux (Ubuntu 18/20/22) } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 |  NVIDIA Jetson                          |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description

Hello, I found that I can't use the "rosrun dynamic reconfigure dynparam" command to dynamically set the exposure mode and exposure parameters in my RealSense ROS program. Because I noticed that there are no topics like "/camera/color/enable_auto_exposure" and so on. What could be the cause of this? On another processor of mine, there are such topics and I can set them normally.

MartyG-RealSense, 2024-11-17:
Hi @JasonWang9805  If you use the command `rosrun rqt_reconfigure rqt_reconfigure` to access the graphical interface for changing the parameters, do you have an **rgb_camera** category in the side-panel of the rqt_reconfigure interface that has RGB options within it?

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#set-camera-controls-using-dynamic-reconfigure-params

[image]

JasonWang9805, 2024-11-18:
Thank you for your prompt reply. I've just checked and it can be adjusted in this way. However, the processor connected to my camera currently doesn't support visualization, and I can only do the debugging remotely. Is there a way to make the adjustment by using commands? For example, running "rosrun dynamic reconfigure dynparam" in the terminal. But I can't find the topic related to adjusting the exposure corresponding to this way on my current processor.

JasonWang9805, 2024-11-18:
I'm a Chinese and I use translation software to translate my language. I hope you can understand what I mean.

JasonWang9805, 2024-11-18:
Thank you for your prompt feedback. I know how to operate it now. I'd like to ask whether the parameters will revert to their default values after the next power - on startup.

MartyG-RealSense, 2024-11-18:
Yes, parameters changed dynamically during runtime will return to their defaults on the next launch.

JasonWang9805, 2024-11-18:
Is there a direct way to fix it? Similar to the camera frequency in the launch file, as long as the launch file is launched, there will be corresponding parameters.

MartyG-RealSense, 2024-11-18:
If a parameter can be set dynamically then it can also be manually added to the launch file as a `<rosparam> </rosparam>` code block if the parameter is not already present in the file.  The rosparam parameter will have the same parameter name as its `rosrun dynamic reconfigure dynparam` equivalent.

Here is an example of structuring a rosparam code block in the launch file, with a command that sets RGB auto-exposure to false (disabled).

```
<rosparam>
 /camera/rgb_camera/enable_auto_exposure: false
</rosparam>
```

MartyG-RealSense, 2024-11-18:
Please try the line below inside the rosparam code block in the launch file to set an RGB manual exposure value.

`/camera/rgb_camera/exposure: 78`

RGB exposure uses small values instead of the very large values such as '33000' that depth exposure uses.

If you want to set depth manual exposure in the rosparam block then try this:

`/camera/stereo_module exposure 33000`

MartyG-RealSense, 2024-11-19:
It's excellent news that you were successful.  As it works, I will close this case.  Thanks very much for the update!