realsenseai/realsense-ros #3256: Can't not display compressed RGB in RViz2

yeyiru, 2024-11-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435}                          | 
| Firmware Version                   | (5.16.0.1)                          | 
| Operating System & Version |   { Docker ubuntu22.04 in Jeston Orin } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | NVIDIA Jetson Orin                           |
| Librealsense SDK Version       |  { 2.55.1}                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Humble}                                           | 
| RealSense ROS Wrapper Version   |  {ros2-master}                                                  | 

### Issue Description
I encountered a similar issue to the one described in [https://github.com/IntelRealSense/realsense-ros/issues/3111#issue-2320092004](url), but I couldn't find a solution from the provided information.

Start Node use this command:
```
ros2 launch realsense2_camera rs_launch.py rgb_camera.color_profile:=1920x1080x30 pointcloud.enable:=false enable_depth:=false rgb_camera.color_format:='BGR8'
```

Here's the full output:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-11-18-12-25-54-407988-ubuntu-1626993
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1627052]
[realsense2_camera_node-1] [INFO] [1731903954.724520403] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1731903954.724775894] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1731903954.724830167] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1731903954.769832810] [camera.camera]: Device with serial number 825312071830 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1731903954.769963403] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1731903954.769992044] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1731903954.770310736] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1731903954.770344368] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1731903954.770457169] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1731903954.771488573] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1731903954.771531934] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1731903954.771556478] [camera.camera]: Device Serial No: 825312071830
[realsense2_camera_node-1] [INFO] [1731903954.771573822] [camera.camera]: Device physical port: /sys/devices/platform/bus@0/3610000.usb/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1731903954.771588991] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1731903954.771602239] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1731903954.771614943] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1731903954.945248608] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1731903954.947007893] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1731903954.957839924] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1731903954.973799921] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1731903954.974461400] [camera.camera]: re-enable the color stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1731903954.982440631] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1731903955.013739464] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1731903955.028787834] [camera.camera]: Open profile: stream_type: Color(0), Format: BGR8, Width: 1920, Height: 1080, FPS: 30
[realsense2_camera_node-1] [INFO] [1731903955.032407236] [camera.camera]: RealSense Node Is Up!
```
I can get these topics:
```
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/image_raw/camera_info
/camera/camera/color/image_raw/compressed
/camera/camera/color/image_raw/compressedDepth
/camera/camera/color/image_raw/theora
/camera/camera/color/metadata
/camera/camera/extrinsics/depth_to_color
```
If I `echo /camera/camera/color/image_raw/camera_info`  **no output**
If I `echo /camera/camera/color/camera_info` the camera's parameters can be output normally.

In rviz2, if I subscribe `/camera/camera/color/image_raw/compressed` nothing was shown
[image]

[image]
I got the warning:
`Camera Info
Expecting Camera Info on topic [/camera/camera/color/image_raw/camera_info]. No CameraInfo received. Topic may not exist.`

Here is the rviz2 output in terminal:
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1731904743.348072984] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1731904743.348235194] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1731904743.374858288] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1731904752.914804244] [rviz2]: Stereo is NOT SUPPORTED
```
I try to `echo /camera/camera/color/image_raw/camera_info` can't get some information.

If I subscribe `/camera/camera/color/image_raw/raw` it's normal:
[image]

Could you give me some advice, thank you very much!

yeyiru, 2024-11-18:
by the way, if I `echo /camera/camera/color/image_raw/compressed`, I can get some data, but if I `hz /camera/camera/color/image_raw/compressed`, there are some warning:
```
WARNING: topic [/camera/camera/color/compressed] does not appear to be published yet
```
but if I use `hz /camera/camera/color/image_raw/image_raw` it's normal

yeyiru, 2024-11-18:
In rqt I can see the topic
[image]

yeyiru, 2024-11-18:
> Hi @yeyiru Can you confirm please if you have installed the **image_transport** plugin for Humble with the command below so that compressed topics for the RealSense data are published?
> 
> `sudo apt install ros-humble-image-transport `
> 
> Further information about the compressed topics can be found at the official RealSense ROS2 documentation site here:
> 
> https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

Hi @MartyG-RealSense , Thanks for the quick reply.
Yes, I have installed the package. I also checked the document you mentioned, but I can't seem to find a solution for this problem.

MartyG-RealSense, 2024-11-18:
What happens if you set a lower color resolution such as 640x480x30 instead of 1920x1080x30

yeyiru, 2024-11-18:
> What happens if you set a lower color resolution such as 640x480x30 instead of 1920x1080x30

Bad news,  The same problem.

MartyG-RealSense, 2024-12-01:
Case closed due to no further comments received.

closed 2024-12-01 by MartyG-RealSense

yeyiru, 2024-12-01:
> In the Rviz2 'Create visualization' window, what happens if you change the drop-down selection from 'compressed' to 'raw' like in the image below?

Sorry I forgot to reply, this will show the screen properly

yeyiru, 2024-12-01:
@MartyG-RealSense Sorry I forgot to reply, this will show the screen properly

MartyG-RealSense, 2024-12-01:
@yeyiru  No problem at all.  I'm pleased to hear that it works for you!