realsenseai/realsense-ros #3262: ros2 wrapper for L515

jin-yiwei, 2024-11-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | L515                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Ubuntu 22 | 
| Kernel Version (Linux Only)    |  5.19.0                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  2.55.1                                                    | 
| Language                               |  python         | 
| Segment			         |  others                              | 
| ROS Distro			         |  humble                                           | 
| RealSense ROS Wrapper Version   |  i don't sure                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
i have 3 camera, two D435i and one L515, when i make: `ros2 launch realsense2_camera rs_launch`, d435i are ok, but l515 show "No realsense devices were found", do you know why?

MartyG-RealSense, 2024-11-21:
Hi @jin-yiwei  Support for L515 ended in librealsense version 2.54.1 because it is a retired camera model.  If you are using the RealSense ROS wrapper then I recommend installing the 2.50.0 or 2.51.1 version of librealsense, and using the 4.51.1 ROS2 wrapper in order for your L515 to be detected.

tonywong-1115, 2024-11-22:
Thanks for ur reply. But it seems the 2.54.1 version doesn't support ubuntu 22.04 because of the kernel version problem. The focal patches only support up to 5.11 kernel. Is there a better solution that I can launch l515 on 22.04?

MartyG-RealSense, 2024-11-22:
You could build 2.51.1 from source code using the **RSUSB Backend** procedure described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

The 2.51.1 librealsense version's source code can be downloaded from the 'Assets' file list at the bottom of the page linked to below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Once librealsense 2.51.1 has been successfully installed then you can build the **4.51.1** ROS2 wrapper from source code.  There is a custom 'forked' version of 4.51.1 created by a RealSense ROS user at the link below that you could build.

https://github.com/Unlimited-Robotics/realsense-ros-4.51.1

MartyG-RealSense, 2024-12-01:
Hi @jin-yiwei and @tonywong-1115  Do either of you require further assistance with this case, please?  Thanks!

jin-yiwei, 2024-12-06:
> Hi @jin-yiwei and @tonywong-1115 Do either of you require further assistance with this case, please? Thanks!

Thanks~ It seems solved

MartyG-RealSense, 2024-12-06:
You are very welcome.  I'm pleased to hear that.

closed 2024-12-13 by MartyG-RealSense