realsenseai/realsense-ros #3263: IMU data not being published on Raspberry Pi 4 and CPU not being able to handle depth data (Resource temporarily unavailable)

dfloreaa, 2024-11-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                          | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Linux (Ubuntu 20.04 Server) | 
| Kernel Version (Linux Only)    |  5.15.0-1065-raspi                                                                       | 
| Platform                                 |  Raspberry Pi 4 8GB                           |
| Librealsense SDK Version       |  v2.55.1                                                    | 
| Language                               |  C++/python        | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  v4.55.1                                                 | 

### Issue Description
I have been trying to use the RealSense D435 depth module for a few weeks now on a Raspberry Pi 4 connected to a server hosted on a Jetson Nano to publish depth data for a current scene as seen by a robot and have been fairly successful to some extent.

First off, there is no way to install the librealsense SDK on the Pi because there is no patch available for `uvcvideo` for the Pi's kernel (5.15.0-1065), only for regular Ubuntu distributions. Therefore, the only alternative is to build the library from source with the line
`cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_GRAPHICAL_EXAMPLES=false`

This way, rsusb backed is used instead of uvcvideo and it works fairly fine, with some bugs that had to be fixed, apparently, the Pi's CPU struggles with processing depth data at higher resolutions and framerates, with the error
`[realsense2_camera_node-1]  22/11 11:22:46,493 WARNING [281472779151584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`, which has been previously reported. Fortunately, the `camera/camera/depth/image_rect_raw` still gets published at a fair rate (~30fps) and that is well enough for what my current requirement needs. Still, it would be nice to get rid of that warning as it gets consistently logged in my console when I run the node (setting verbose to ERROR only does not solve this), if I could get some help with that it would be perfect.

On the other hand, I am now trying to access IMU information from the camera in order to tell it's angular orientation from the ground plane (in order to rotate the points to a top-down perspective) but struggle to get this data published even when I launch the node using the required parameters. My hypotheis is that this could have to do with the RSUSB backed being used instead of the intended uvcvideo one.

I am currently launching the node with the line:
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_sync:=true depth_module.depth_profile:=848x480x30 enable_infra:=false enable_infra1:=false enable_infra2:=false enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=2`

_I have also tried to just launch the node instead of the entire launchfile and it does not work either._

Whenever I list currently available topics I only get:
> /camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/image_rect_raw/compressed
/camera/camera/depth/image_rect_raw/compressedDepth
/camera/camera/depth/image_rect_raw/theora
/camera/camera/depth/metadata
/parameter_events
/rosout
/tf_static

Which does not include the `/camera/imu` topic that I need in order to access such information, this could be due to CPU not being able to process it because of it being busy with the depth data.

I mainly see two options to help me with my problem:

1. Add the patch for Ubuntu kernels on Raspberry Pi systems in order to be able to install the SDK the regular way
2. Somehow make the CPU process the data more efficiently, which could make the IMU data get published, I have tried with lowering the framerate and resolution and it does not solve the issue.

Here is the log for when I attempt to run the node with both depth and IMU information:

> robot@raspi-realsense:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_sync:=true depth_module.depth_profile:=848x480x30 enable_infra:=false enable_infra1:=false enable_infra2:=false enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=2
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2024-11-22-11-33-19-754296-raspi-realsense-4045
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4046]
[realsense2_camera_node-1] [INFO] [1732286000.328256296] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1732286000.328555931] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1732286000.328675326] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1732286016.157573259] [camera.camera]: Device with serial number 838212071138 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1732286016.157831478] [camera.camera]: Device with physical ID 2-1-3 was found.
[realsense2_camera_node-1] [INFO] [1732286016.157913021] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1732286016.158629625] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1732286016.158783697] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1732286016.158849710] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1732286022.679680889] [camera.camera]: Device with serial number 838212071138 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1732286022.679951330] [camera.camera]: Device with physical ID 2-1-4 was found.
[realsense2_camera_node-1] [INFO] [1732286022.680098296] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1732286022.681013968] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1732286022.681165247] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1732286022.681415991] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1732286022.682640812] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1732286022.682801143] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1732286022.682886409] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1732286022.682945999] [camera.camera]: Device Serial No: 838212071138
[realsense2_camera_node-1] [INFO] [1732286022.682997314] [camera.camera]: Device physical port: 2-1-4
[realsense2_camera_node-1] [INFO] [1732286022.683051961] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1732286022.683100943] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1732286022.683146427] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1732286025.776629379] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1732286026.189470967] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1732286026.219494685] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1732286026.388957326] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1732286026.518924755] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1732286026.530359565] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  22/11 11:33:46,531 WARNING [281472870967520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  22/11 11:33:46,987 WARNING [281472870967520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/11 11:33:47,101 WARNING [281472870967520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/11 11:33:47,213 WARNING [281472870967520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
(...)

MartyG-RealSense, 2024-11-22:
Hi @dfloreaa  The problem with IMU data my not be related to computing efficiency, as in most cases Pi boards will publish depth and color but not IMU data.  The issue of no access to a RealSense camera's IMU has previously been looked at by the Intel RealSense team and was thought to be due to the Raspberry Pi kernel - please see https://github.com/IntelRealSense/librealsense/issues/4979#issuecomment-540456439

In regard to reducing the logging rate, if you know how to write a ROS node script then you could try throttling messages to appearing only once every 'x' number of seconds, such as every 1 second.  The ROS2 Humble documentation has example C++ code here:

https://docs.ros.org/en/humble/Tutorials/Demos/Logging-and-logger-configuration.html#logging-throttled-all-but-the-first-time

dfloreaa, 2024-11-22:
Thank you for the quick reply, from what I read there was an attempt to enable IMU data logging on the Raspberry Pi, but it was not succesful due to the Ubuntu kernel used for raspberry systems, in my case, 5.15.0-1065-raspi.

I was considering manually patching the raspberry pi with the `patch-realsense-ubuntu-lts.sh` file (patching by hand using the same lines as the script, skipping the trailing changes verification) and installing the SDK using apt-install directly, that could possibly work (?)

As for the logs coming from libusb, has this issue not been accounted and/or fixed before? I have seen plenty of issues with people with the same problem but most of the time they are just told to ignore it, is there no way to disable this logging via the own SDK's code/the wrapper before building the library/package? I do find it weird that the CPU is struggling with processing the frames but still able to publish them at the desired rate.

Of course I can write a node that can throttle the messages but that just seems like a patch solution for a bigger underlying issue with the SDK, could this be accounted and hopefully fixed? It would be nice to either:

1.  Be able to disable these warnings from the launch (by logging them as warnings instead of regular prints, like they currently are)
2. Fix the underlying issue with the pi's CPU not being able to handle the processing of the frames (although it seems to be more than capable and there may be another issue that causes this warning to pop).

Thank you in advance for your help

MartyG-RealSense, 2024-11-23:
1.  Redirecting the ROS message output to log could mute the Warning type messages on screen whilst still allowing Error messages through, as described at https://github.com/IntelRealSense/realsense-ros/issues/1185#issuecomment-621218705

2.  Raspberry Pi is not an officially supported platform of the librealsense SDK (the primary focus is on x86/x64 architecture such as desktop / laptop PCs) and on the Arm-based Nvidia Jetson), so the SDK does not have Pi-specific updates and fixes deveoped for it.

It is possible to build the ROS wrapper from source code with colcon build in a mode that takes advantage of a GPU to provide acceleration to pointclouds and depth colorization.  If your project uses a colorized depth stream then it could potentially benefit from this mechanism.

***********

**accelerate_gpu_with_glsl**

Boolean: GPU accelerated with GLSL for processing PointCloud and Colorizer filters.

Note: To have smooth transition between the processing blocks when this parameter is updated dynamically, the node will:

1. Stop the video sensors
2. Do necessary GLSL configuration
3. And then, start the video sensors

To enable GPU acceleration, turn on **BUILD_ACCELERATE_GPU_WITH_GLSL** during build:

`colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'`

closed 2024-12-07 by MartyG-RealSense