realsenseai/realsense-ros #3265: Building librealsense from source but installing realsense-ros2 with sudo apt-get

ignacio-pm, 2024-11-26:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                         | 
| Firmware Version                   | 5.15.1                        | 
| Operating System & Version |   Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  5.15.148-tegra                                                                       | 
| Platform                                 | NVIDIA Jetson                     |
| Librealsense SDK Version       |  2.56.2                                             | 
| Language                               | ?         | 
| Segment			         |  Robot                           | 
| ROS Distro			         |  Humble                                       | 
| RealSense ROS Wrapper Version   |  Latest                                                 | 

### Issue Description
Hi,
I use a Realsense camera D455 with a Jetson Orin. In order to make it work, I need to install librealsense from source with the RSUSB patch (using [this script](https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh)). If installed from apt packages the camera D455 is not detected ( although a camera D435i is detected).

I would like to use the camera with the ROS 2 wrapper. If I install the wrapper from source, it works without problems. However, if I install it with debian packages, it does not recognize that librealsense was installed from source and it is overridden. This makes the wrapper to stop working because the librealsense that is installed through sudo apt-get does not have the RSUSB patch.

Is there a way to install the ROS 2 wrapper from debian packages after installing librealsense from source or to install librealsense from debian packages with the RSUSB patch?

Thanks in advance for the response.

MartyG-RealSense, 2024-11-26:
Hi @ignacio-pm  There should only be one installation of librealsense on your computer, a package installation or a source code build, but not both types on the same computer.  Otherwise, a conflict between the two librealsense installations could occur.  So if you build librealsense from source with the RSUSB method, it should not be installed a second time from packages afterwards.

**********

If you wish to attempt an installation of librealsense from packages on Jetson then the Jetson-specific packages on the Jetson instructions page at the link below must be used, not the ones on the distribution_linux.md instruction page.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

This approach is recommended by the ROS wrapper's installation instructions.

[image]

Once librelsense has been sucessfully installed from packages then you can attempt an install of the ROS wrapper from packages for ROS2 Humble with the command below:

`sudo apt install ros-humble-realsense2-*`

ignacio-pm, 2024-11-26:
Hi @MartyG-RealSense ,
I already tried installing from debian packages following the Jetson guide. When running rs-enumerate-devices in this case with the D455, it shows "No device detected. Is it plugged in?". This problem does not happen with the D435i. I fixed the problem by building the package from source using the [RSUSB backend](https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-rsusb-backend). Is there another possible workaround?

My first original question was if there is a way to install librealsense with the RSUSB backend from debian packages. I guess not because there nothing mentioned in the documentation.

My second and main question is if it is possible to install the ROS 2 wrapper from the debian packages, after installing librealsense from source. I tried installing ros-humble-realsense2-* and sudo apt-get does not understand that librealsense is already installed (from source). Therefore, sudo apt-get installs ros-humble-librealsense2, which duplicates the library, and as you mentioned, this should not happen. I would want to link the source built librealsense package to be detected by the debian package from ROS 2 wrapper so that ros-humble-librealsense2 is not installed and the wrapper uses the librealsense package, previously built from source.

closed 2024-11-27 by ignacio-pm