realsenseai/realsense-ros #3266: Call to set_parameters service hangs the realsense node

fbottarel, 2024-11-27:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                      | 
| Firmware Version                   | 5.14.0                       | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  6.8.0-49                                                                 | 
| Platform                                 | PC                           |
| Librealsense SDK Version       | ros-humble-librealsense2 2.55.1                                              | 
| Language                               |  Python        | 
| Segment			         |  Robot                          | 
| ROS Distro			         | Humble                      | 
| RealSense ROS Wrapper Version   | ros-humble-realsense-camera 4.55.1                                            | 

### Issue Description

One of my nodes has a callback that is supposed to toggle color and depth camera streams on and off. In order to do so, the python node makes a service call to the `/camera/camera/set_parameters` service in order to set parameters `enable_color` and `enable_depth` to `False`. The realsense node is launched with 

```shell
ros2 launch realsense2_camera rs_launch.py
```

<details>

<summary> Minimal working example: set_parameters call  </summary>

```python
from rcl_interfaces.msg import Parameter, ParameterType, ParameterValue
from rcl_interfaces.srv import SetParameters
import rclpy
from rclpy.node import Node
 
 
class SetParamsNode(Node):
 
 
    def __init__(self):
        super().__init__('setparamsnode')
        self.parameter_client = self.create_client(SetParameters, '/camera/camera/set_parameters')
        self.period = 1.0
        self.timer_ = self.create_timer(self.period, self.update)
        self.b_value = False
 
    def update(self):
        self.set_camera_parameters()
        self.b_value = not self.b_value
 
    def set_camera_parameters(self):
        param_color = Parameter(
            name='enable_color',
            value=ParameterValue(type=ParameterType.PARAMETER_BOOL, bool_value=self.b_value)
        )
        param_depth = Parameter(
            name='enable_depth',
            value=ParameterValue(type=ParameterType.PARAMETER_BOOL, bool_value=self.b_value)
        )
 
        # Send the request
        request = SetParameters.Request()
        request.parameters= [param_color, param_depth]
        self.parameter_client.call_async(request)
 
def main(args=None):
 
    rclpy.init()
    node = SetParamsNode()
    
    try:
        rclpy.spin(node)
    finally:
        node.destroy_node()
        rclpy.try_shutdown()
 
if __name__ == '__main__':
    main()
```

</details>

The first call to `set_parameters` works, turning off the streams, but it never returns. Moreover, the realsense node hangs and is unresponsive to any further service calls. However, the same code making the same call to `set_parameters_atomically` works as intended.

<details>

<summary> Minimal working example: set_parameters_atomically call  </summary>

```python
from rcl_interfaces.msg import Parameter, ParameterType, ParameterValue
from rcl_interfaces.srv import SetParametersAtomically
import rclpy
from rclpy.node import Node
 
 
class SetParamsNode(Node):
 
 
    def __init__(self):
        super().__init__('setparamsnode')
        self.parameter_client = self.create_client(SetParametersAtomically, '/camera/camera/set_parameters_atomically')
        self.period = 1.0
        self.timer_ = self.create_timer(self.period, self.update)
        self.b_value = False
 
    def update(self):
        self.set_camera_parameters()
        self.b_value = not self.b_value
 
    def set_camera_parameters(self):
        param_color = Parameter(
            name='enable_color',
            value=ParameterValue(type=ParameterType.PARAMETER_BOOL, bool_value=self.b_value)
        )
        param_depth = Parameter(
            name='enable_depth',
            value=ParameterValue(type=ParameterType.PARAMETER_BOOL, bool_value=self.b_value)
        )
 
        # Send the request
        request = SetParametersAtomically.Request()
        request.parameters= [param_color, param_depth]
        self.parameter_client.call_async(request)
 
def main(args=None):
 
    rclpy.init()
    node = SetParamsNode()
    
    try:
        rclpy.spin(node)
    finally:
        node.destroy_node()
        rclpy.try_shutdown()
 
if __name__ == '__main__':
    main()
```

</details>

The same behaviour can be observed by making the same calls via CLI. Is there something I am missing?

MartyG-RealSense, 2024-11-27:
Hi @fbottarel  The functions in the ROS2 wrapper that support device calls are limited to camera device information, such as serial number and firmware version.  The section of the wrapper documentation linked to below provides information about this.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#available-services 

A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/3067 had similar problems in April 2024 with trying to use service calls to enable and disable depth and color.

fbottarel, 2024-11-28:
Thanks @MartyG-RealSense, I am not sure I understand your answer. Could you elaborate a bit more? There is support for parameter reconfiguring, as I can disable streams through both methods I described. I can also re-enable them without the node freezing, but only if I use the `_atomically` service. I am not sure in which way your answer sheds light on the issue.

MartyG-RealSense, 2024-11-28:
There is no further advice that I can provide about this question unfortunately as the official wrapper documention only covers service calls related to camera information and not other functions such as controlling the on-off status of streams.

MartyG-RealSense, 2024-12-07:
Hi @fbottarel  Bearing in mind the response in the comment above, do you require further assistance with this case please?  Thanks!

closed 2024-12-13 by MartyG-RealSense