realsenseai/realsense-ros #3268: Installing a different librealsense version when using the libuvc script

thulioguilherme, 2024-12-02:
### System

| Required Info                         |                                                                                       |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                        | D435i |
| Firmware Version                    | 5.13.0.50  | 
| Operating System & Version   | Ubuntu 20.04 | 
| Kernel Version (Linux Only)    | 4.9.241 | 
| Platform                                  | Khadas |
| Librealsense SDK Version      | 2.55.1 | 
| Language                               | roslaunch | 
| Segment			         | Robot | 
| ROS Distro			         | Noetic | 
| RealSense ROS Wrapper Version | 2.2.23 | 

### Issue Description
Hi, I am working with a Realsense D435i using ROS1. I installed the librealsense 2.55.1 using the [libuvc script](https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh) because kernel patching is not possible with Khadas. I am trying to change the exposure and gain parameters using the `dynamic_reconfigure` node, but it just does not return.  Also, the `rqt_configure` node freezes when I select the `stereo_module`.

Since it seems that there is an incompatibility between the librealsense, ROS wrapper and firmware version (as discussed in [this issue](https://github.com/IntelRealSense/realsense-ros/issues/3255)), how can I install a different librealsense version using the mentioned script?

If not possible, is there a compatible version of camera firmware and realsense-ros (ROS1) that works with librealsense 2.55.1?

MartyG-RealSense, 2024-12-03:
Hi @thulioguilherme  The libuvc backend installation procedure that you are using to install librealsense is the best method to use in your particular situation, since it bypasses the kernel and so the application of a patch script is not required.

Development of the ROS1 wrapper ceased after librealsense 2.50.0 (though 2.51.1 also works with it) so the wrapper has not been updated for recent librealsense versions such as 2.55.1.  Some RealSense users have found that they are able to use 2.55.1 and firmware 5.16.0.1 with the 2.3.2 wrapper though, and it sounds as though 2.55.1 is working for you too.

When using dynamic_reconfigure with parameters, launch must have been completed first as they are commands that are used during runtime after launch.  Can you confirm that you have completed launch and the streams are being published when you use dynamic_reconfigure, please?

As well as being able to configure settings in the graphical interface of dynamic_reconfigure, you can also set individual parameters dynamically during runtime with rosrun commands, like these:

```
rosrun dynamic_reconfigure dynparam set /camera/stereo_module exposure 33000
rosrun dynamic_reconfigure dynparam set /camera/rgb_camera/ exposure 78
```

In regard to installing another librealsense version, you could try installing librealsense 2.50.0 and ROS1 wrapper 2.3.2 together at the same time from packages with the command below:

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

The packages installed by this command are based on **RSUSB = true**.  RSUSB is the successor of libuvc and works in a similar way by bypassing the kernel and not requiring the application of a patch script, so this installation method could also potentially work on your Khadas system.

thulioguilherme, 2024-12-03:
Hello again, the streams are being published. I can see the frequency using `rostopic hz` and also the video through `rosrun image_view image_view`, but I am unable to set the parameters in both cases (rqt_reconfigure and dynamic_reconfigure). 

I will try to install using the packages this time. The correct combination for ROS1 is librealsense 2.50.0, ROS1 wrapper 2.3.2, and camera firmware 5.13.0.50?

MartyG-RealSense, 2024-12-03:
> The correct combination for ROS1 is librealsense 2.50.0, ROS1 wrapper 2.3.2, and camera firmware 5.13.0.50?

Yes, that is the correct configuration for the 2.3.2 ROS1 wrapper.

thulioguilherme, 2024-12-03:
I installed the combination mentioned in the previous comment, and I was able to change the parameters in runtime. Thanks for the help.

closed 2024-12-03 by thulioguilherme