realsenseai/realsense-ros #3271: How to get multiple pointcloud mapped with infra1 and infra2?

DY-JANG0812, 2024-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-12-06:
Hi @DY-JANG0812  The Infrared 2 topic would be unavailable if the ROS launch was detecting your camera as being on a USB 2.1 connection, even if it was plugged into a USB 3.0 port.  This is because Infrared 2 is only supported on a USB 3 connection.

You may be able to achieve the results that you want simply by enabling the pointcloud filter by adding **pointcloud.enable:=true** to your launch instruction though.  For example:

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`

There is no need for you to do manual work to calibrate the two infrared topics to depth.  A RealSense depth frame is generated inside the camera hardware from raw left and right infrared frames (not the Infra2 and Infra2 topics) before the frames are even sent along the USB cable to the computer. 

Also, the left infrared camera has the benefit of always being pixel-perfect aligned, calibrated, and overlapped with the depth map, and also perfectly time-synchronized.

DY-JANG0812, 2024-12-06:
@MartyG-RealSense When referring to the left side, is it from the perspective of facing the camera from the front, or from looking at the back of the camera?

MartyG-RealSense, 2024-12-06:
Because the 0,0,0 origin point of depth is the center-line of the left infrared sensor, when depth is aligned to infrared the origin point will still be the left infrared sensor and so the same coordinate system will be used.

When depth is aligned to color the origin of depth changes to the center-line of the RGB sensor.

The camera uses the perspective of looking forwards from behind the back of the camera.  This is why when looking at the camera from the front, the left infrared sensor is on the right-side of the camera.

DY-JANG0812, 2024-12-09:
I'm currently using a RealSense camera for obstacle detection, but I'm having trouble extracting correct depth points(NOISE) from acrylic surfaces. I wanted to use infrared textures to remove incorrect depth points, but the issue seems to be that when the acrylic is warped or when infrared light is strongly detected or NOT reflected to camera, the depth calculation is incorrect because the light is not properly detected. I'm considering using filters like hole filling filtering or toggling the emitter on/off. Is there an internal feature that ensures only the filled depth points are outputted during hole filling, instead of noise points?

MartyG-RealSense, 2024-12-09:
As you are using ROS, the best way to strengthen the checking of depth values for confidence in their accuracy and excluding values with low confidence in their accuracy would likely be to load in a **json** camera configuration file (such as 'high_accuracy') using the ROS parameter **json_file_path** parameter, as discussed at https://github.com/IntelRealSense/realsense-ros/issues/2445

I would recommend 'medium_density' over the high_accuracy preset file, as medium_density provides a good balance between accuracy and the amount of detail on the depth image, whilst high_density tends to over-strip the depth detail when eliminating low-confidence values and therefore leaving the image looking sparse.

closed 2025-01-20 by MartyG-RealSense