realsenseai/realsense-ros #3273: XArm robot and RealSense D400 calibration - compatibility issues

luciargonza2, 2024-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {  D435if }                          | 
| Firmware Version                   | (?)                          | 
| Operating System & Version |   { Win (11) but I could instal ubuntu  } | 
| Kernel Version (Linux Only)    |         ?                                                             | 
| Platform                                 | ?                       |
| Librealsense SDK Version       |  { ?}                                                    | 
| Language                               |  { }         | 
| Segment			         |  {Robot XArm6 from UFACTORY }                             | 
| ROS Distro			         |  {?}                                           | 
| RealSense ROS Wrapper Version   |  {?}                                                  | 

### Issue Description

Good morning,

I’m reaching out for guidance on how to use an XArm 6 robot from UFACTORY with the RealSense D435i camera. I’ve been trying to follow the documentation from both companies but am encountering many compatibility issues.

Firstly, UFACTORY primarily supports ROS 1 and hasn’t updated its ROS 2 packages since 2022. On the other hand, Intel seems to focus on the opposite, which makes it challenging to determine the best approach.

In my case, I’m working on a Windows 11 computer, but I’ve also tried installing the necessary packages on Ubuntu. Regardless of the setup, I keep running into compatibility errors related to kernel versions, Ubuntu versions, the Librealsense SDK, and other dependencies.

I know there are people who use this exact combination of robots and RealSense cameras from the D400 series successfully. I’d like to ask if anyone could guide me on the specific versions I should install to avoid compatibility issues.

For example, I need advice on the following:

Firmware version
Operating system and version
Kernel version (if using the Linux subsystem on Windows)
Platform
Librealsense SDK version
ROS distribution
xArm ROS SDK version
RealSense ROS Wrapper version
Anything else that might be important to ensure everything works smoothly on my system would be incredibly helpful. At this point, I’ve tried many different combinations, and I’m unsure whether the issues are due to compatibility problems or other errors.

Thank you so much for your guidance.

Best regards,
Lucia

MartyG-RealSense, 2024-12-10:
Hi @luciargonza2  RealSense does have a compatibility wrapper for ROS1.  It has not been updated for a while as only the RealSense ROS2 wrapper is now actively developed.  The ROS1 wrapper supports ROS1 versions up until Noetic (the final ROS1 version) though so it is probable that it could work with UFACTORY's ROS1 interface.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

The recommended configuration for the ROS1 wrapper is librealsense 2.50.0, ROS1 wrapper 2.3.2 and camera firmware driver version 5.13.0.50.  In regard to the Ubuntu version to use, the ROS1 wrapper is able to work with 18.04 for ROS1 Melodic or 20.04 for ROS1 Noetic.

If you install librealsense 2.50.0 and ROS1 wrapper 2.3.2 from packages together at the same time then the installation will be based on the **RSUSB backend**, meaning that it should not matter what the kernel version is because when RSUSB = true, the kernel is bypassed and so the installation is not dependent on a particular Linux version or kernel version and does not require a kernel patch script to be applied to the kernel.

You can install librealsense and the wrapper from packages with the command below:

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

In regard to Windows, whilst it is technically possible to install the ROS wrapper on Windows or use WSL2, it can be very difficult to get working and so I would recommend installing on a computer using the Ubuntu OS if possible for you to do so.

luciargonza2, 2024-12-11:
Hello Marty,

Thank you for your reply. I followed your advice and installed ROS1 wrapper 2.3.2, with the corresponding librealsense 2.50.0, in ubuntu 20.4 with Noetic. Now I am trying to run this command to check the installation "roslaunch realsense2_camera rs_camera.launch", but the RealSense camera is not been detected.

 Also, I tried to enumerate the devices and I get this error "No device detected. Is it plugged in?". 

The kernel version that I am using is "5.15.167.4-microsoft-standard-WSL2" in Windows 11.

But Im still getting the same error again:

Could you help me taking a look at it once again? Thank you

[image]

luciargonza2, 2024-12-11:
Hello Marty,
I've successfully got the ubuntu to recognize the realsense camera; the problem remains, and when I run 
[image]
but when I run rs enumerate to see if the software recognize the device
[image]
and im using WSL 2 but when I switched to WSL 1, it works but returning a new error message that udev failed
Thank you

MartyG-RealSense, 2024-12-12:
Detection of the D435if model was not added to the RealSense SDK and RealSense Viewer until version **2.54.1**.  

[image]

So version 2.50.0 would not be able to detect the D435if when it is plugged in because the SDK was not yet programmed with this camera model's product ID (PID) number.

luciargonza2, 2024-12-17:
Hello Marty,

This time I installed the sdk version 2.54.1 and the wrapper 2.3.2 for ROS1 in Ubuntu Noetic 20.04. I also checked that the device was shared and attached from my Windows 11 to the WSL. The problem remains "No RealSense devices were found!". When I try to open Realsense-viewer, I get an info message saying "Cannot access /sys/class/video4linux" I believe I might be missing something bwhen installing librealsense 2.54.1 instead of 2.50.0. And when trying to enumerate the devices, no device is recognized either. Would getting a 435i camera solve this issue?

MartyG-RealSense, 2024-12-17:
A D435i can work with old RealSense versions such as 2.50.0.  However, if you installed 2.54.1 and your D435if was still not detected then it suggests that the problem might be with WSL2 rather than the camera.  

You can test the camera in 2.54.1 in Windows instead of Ubuntu to confirm this by visiting the link below and then downloading and running the file **Intel.RealSense.Viewer.exe** from the 'Assets' file list at the bottom of the page at that link.  If the 2.54.1 version of the RealSense Viewer tool detects the D435if correctly then it is likely that it is a WSL2 or Ubuntu issue.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

[image]

There was a WSL2 user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 who was able to get RealSense working with WSL2 on Windows 11 by "completely unplugging and replugging the camera and also restarting WSL".

Another RealSense WSL2 user who tried this method of getting the camera detected said that sometimes it worked for them and sometimes it didn't.

luciargonza2, 2024-12-17:
Hi Marty, 

Now I am trying toinstall in ROS2. I installed ROS2 distro humble with Intel RealSense v2.55.1 SDK and 4.55.1 wrapper in Ubuntu 22.04 jummy  5.15.167.4-microsoft-standard-WSL2. I also checked that the camera was connected and shared with the wsl. However, I still get the problem that the camera is nor recognized. Any idea of why this happens and how can I solve the problem? Also, I realised of the following error, but I do not know how to solve this problem or if it's even significant.

Best regards,
Lucia
[image]
[image]

MartyG-RealSense, 2024-12-18:
As Intel do not provide official support for using RealSense with WSL2, problems with it tend to only be solvable with advice from other RealSense WSL2 users.

For example, a WSL2 user at the link below succeeded with instructions that recommended inputting the following command to copy the **udev** device handling rules before using the camera.

`sudo ln -s ~/repos/librealsense/config/99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules`

https://stackoverflow.com/questions/75335203/realsense-sdk-2-0-installation-in-ubuntu

luciargonza2, 2024-12-21:
Hello Marty,

Since you do not provide support with WSL2, I installed a native Linux system with 20.04 Ubuntu, ROS1 Noetic and got the camera D435i so that there is no further compatibility issue. Now I am trying to build from source the old librealsense-2.50.0 version by downloading the .zip folder, but I keep getting an error when running the file in ./script/patch-realsense-ubuntu-lts.sh. The kernel I am using is 5.4, although I also tried the 5.8 and got the same error. Also I have the firewall inactive so that shouldn't be the problem. I believe I should be getting the patch for focal, but since it cannot access the repository because of a connection issue I cannot proceed with the installation. How could I solve this? 
[image]

luciargonza2, 2024-12-21:
Also, the first message I got was that the repository was empty. I checked the script inside the patch-realsense-ubuntu-lts.sh I tried to look for the specific repository that I thought was trying to get connected to in this line:
git remote add origin git://kernel.ubuntu.com/ubuntu/ubuntu-${ubuntu_codename}.git
and in my case focal as ubuntu_codename. However, I believe that page does not exist anymore. Could you confirm that and in case it is so provide a solution? Thank you.

MartyG-RealSense, 2024-12-21:
If you are using 2.50.0 and ROS1 Noetic then it is easiest to install librealsense 2.50.0 and ROS1 wrapper 2.3.2 together at the same time with the command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

You do not need to apply a kernel patch script with this method because the librealsense that it installs is based on the RSUSB Backend, which bypasses the kernel.

I am not familiar with how the kernel patch scripts work, so I do not have advice to offer about that subject.  I do apologize.

luciargonza2, 2024-12-23:
Hello Marty,
Thank you for your response. This is the new update:

1. I remove any installation first to avoid having it twice:
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
2. I reinstalled in Ubuntu 20.05, kernel 5.8, Ros1 Noetic
export ROS_VER=noetic
sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
3. This time the camera was recognized, however, I run into this problem: 
WARNING [140201154311936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
4. I checked that the ports were ok, I changed the usb ports as well and the problem was the same, connecting the camera directly into the pc.
5. Looking into other issues [I](https://github.com/IntelRealSense/realsense-ros/issues/1510) realized that it is probable that many useful tools were not installed when using this first method installation strategy, because I could not use many things. First of all realsense-viewer was a command not found, as well as rs-enumerate-devices. That is why I tried to follow the second method and build everything from source. However, I always arrive to the same point in which I cannot install the patch when installing librealsense SDK 2.50 following this instructions 
[image]
That is why I followed this other instructions and instead of downloading the .zip file for the 2.50 SDK, i used the server key to download it from Github as in this instructions: 
[image]
As a result, I am not sure if I installed the version2.50 or 2.55.1 because when I open now realsense viewer that is the version I get ( 2.55.1). Either way, I preceded with the installation and installed the ros wrapper 2.3.2 also from source, by downloading the .zip file. After that, I tried again to use the example and I arrived to de same error as in step 3. Although by reading at the details I see that the sdk version is 2.50 and wrapper is 2.3.2, i believe it was built on top of 2.55.1 and that is why the realsense viewer keeps being that version.

Please, could you provide some solution for this? either tell me how to instal sdk 2.50 from source without getting the patch issue or solving the warning error I have with this latest instalation. Here I provide the details of the error:

roslaunch realsense2_camera rs_camera.launch
... logging to /home/ubuntu/.ros/log/d3d5c71a-c0d6-11ef-8a7b-b30cb1237c0f/roslaunch-ubuntu-System-Product-Name-58945.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-System-Product-Name:41111/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
[... 101 lines cut ...]
[ INFO] [1734921441.126644274]: setupStreams...
 22/12 18:37:21,263 WARNING [139729152493312] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1734921441.314073250]: SELECTED BASE:Depth, 0
[ INFO] [1734921441.319615190]: RealSense Node Is Up!
[ WARN] [1734921441.381641844]: 
 22/12 18:37:21,381 WARNING [139729152493312] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 22/12 18:37:22,370 WARNING [139729152493312] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 22/12 18:37:22,420 WARNING [139729152493312] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

luciargonza2, 2024-12-23:
I also tried to include initial_reset:=true and I got the same error. And when using RViz, this was the result
[image]

MartyG-RealSense, 2024-12-23:
The realsense-viewer tool is not installed when using `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` and it is therefore necessary to install librealsense and the ROS wrapper separately if you need to use the librealsense tools and examples.

** control_transfer returned** warnings indicate that there is a communication problem between the camera and computer which is usually caused by the USB system (the port or the cable).  Sometimes topics can display normally in RViz even when the warnings are generating continuously in the log, but apparently not in your particular case, as shown when you performed the initial_reset.

What happens if you change Fixed Frame in RViz from 'map' to **camera_link**

librealsense can be installed from source code in a mode that bypasses the kernel if the **libuvc backend** installation procedure at the link below is used.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc method has a pre-made build script called 'libuvc_installation.sh'.  

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

To install it with tools and examples such as realsense-viewer included, edit line 46 of the script from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

to this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

After using this method and testing that librealsense and its tools are working correctly, you would afterwards have to build the ROS wrapper from source code using 'Method 2' of the ROS1 wrapper instructions.

closed 2025-01-20 by MartyG-RealSense