realsenseai/realsense-ros #3275: ROS topic Camera Intrinsic Parameters are not updated

belalhmedan90, 2024-12-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D415                           | 
| Firmware Version                   |  5.15.0.2                         | 
| Operating System & Version |   Linux Ubuntu 20 | 
| Kernel Version (Linux Only)    |  5.15.0-107-generic                                                                       | 
| Platform                                 | PC                         |
| Librealsense SDK Version       |  2.54.1                                           | 
| Language                               |     ROS      | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Noetic                                          | 
| RealSense ROS Wrapper Version   |  2.50.0-1                                                 | 

### Issue Description
Camera Intrinsic parameters coming from ROS1 via the `/camera/color/camera_info` topic doesn't match the values from the `realsense-viewer`,   even after running on-chip calibration.

MartyG-RealSense, 2024-12-10:
Hi @belalhmedan90  If you are viewing the **Calibration Data** pop-up panel in the realsense-viewer tool, this will not have information that is relevant to the RGB color sensor.  

[image]

The best way to obtain the camera's color sensor intrinsics is to launch librealsense's **rs-enumerate-devices** tool in calibration information mode with the command below.

`rs-enumerate-devices -c`

In the image below are the intrinsics for the RGB color sensor of my own D415 at 1280x720 and 1920x1080 resolutions, though the intrinsics for each individual camera are unique due to the manufacturing process at the factory.

[image]

If you use rs-enumerate-devices - c to retrieve the color intrinsics for your own D415, how closely do they compare to the ones accessed in ROS1 please?

belalhmedan90, 2024-12-10:
Sorry, I forgot to mention that I have depth and color aligned, and my concern is Depth Intrinsic parameters as I need them to get accurate point-cloud from depth, thanks for your kind support.

```sh
Intrinsic of "Depth" / 1280x720 / {Z16}
  Width:        1280
  Height:       720
  PPX:          641.368225097656
  PPY:          357.280181884766
  Fx:           912.178955078125
  Fy:           912.178955078125
  Distortion:   Brown Conrady
  Coeffs:       0    0    0    0    0  
  FOV (deg):    70.11 x 43.07

 Intrinsic of "Color" / 1280x720 / {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y8/Y16}
  Width:        1280
  Height:       720
  PPX:          633.499755859375
  PPY:          374.307586669922
  Fx:           911.388488769531
  Fy:           910.107055664062
  Distortion:   Inverse Brown Conrady
  Coeffs:       0    0    0    0    0  
  FOV (deg):    70.15 x 43.15
```
`camera_info` is same for color and Depth after alignment:
```sh
rostopic echo -n 1 /camera/aligned_depth_to_color/camera_info
header: 
  seq: 95
  stamp: 
   secs: 1733843005
   nsecs: 471710920
  frame_id: "camera_color_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [911.3884887695312, 0.0, 633.499755859375, 0.0, 910.1070556640625, 374.3075866699219, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [911.3884887695312, 0.0, 633.499755859375, 0.0, 0.0, 910.1070556640625, 374.3075866699219, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

MartyG-RealSense, 2024-12-10:
When depth is aligned to color, color intrinsics should be used for the aligned image because the origin point of depth changes from the center-line of the left infrared sensor to the center-line of the RGB sensor after alignment.  And it looks as though the color intrinsic values provided by rs-enumerate-devices (PPX, PPY, Fx, Fy) are the same as the aligned intrinsics in camera_info in the K and P rows.

closed 2024-12-10 by belalhmedan90