realsenseai/realsense-ros #3277: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

CYJ00, 2024-12-13:
[ INFO] [1734061888.050640114]: RealSense ROS v2.3.2
[ INFO] [1734061888.050656726]: Built with LibRealSense v2.50.0
[ INFO] [1734061888.050662036]: Running with LibRealSense v2.50.0

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Operating System & Version |  Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  Linux 5.15.0-125-generic                                                                      | 
| Platform                                 | PC                           |
| ROS Distro			         |  noetic                                           | 

### Issue Description
I install the package using apt-get install command.
I used `roslaunch realsense2_camera rs_camera.launch` command.

It's the log.
```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
[... 99 lines cut ...]
 13/12 12:51:32,375 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:32,426 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1734061892.476904749]: SELECTED BASE:Depth, 0
 13/12 12:51:32,477 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1734061892.488778735]: RealSense Node Is Up!
[ WARN] [1734061892.764581188]: 
 13/12 12:51:32,765 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:32,816 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:32,867 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:32,917 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:32,970 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:33,021 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:33,072 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
^C 13/12 12:51:33,324 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
 13/12 12:51:33,375 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:33,431 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 13/12 12:51:33,581 WARNING [140640532006656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
shutting down processing monitor...
... shutting down processing monitor complete
done
```
As a result, RGB Image is successfully received. But I can't receive `/camera/aligned_depth_to_color/image_raw` topic.
When I used `roslaunch realsense2_camera rs_camera.launch command`, there was no problem. How can I get /camera/aligned_depth_to_color/image_raw topic?

MartyG-RealSense, 2024-12-13:
Hi @CYJ00  I would not recommend using the `rs_aligned_depth.launch` launch file as it is extremely old.  It is beter to use the rs_camera roslaunch instruction below to publish the aligned topics, `/camera/aligned_depth_to_color/image_raw` and `/camera/aligned_depth_to_color/camera_info`

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`

mewescott, 2025-01-09:
I am having an issue with this too.  I am using the recommended launch parameter and I am getting all zeros when I echo the topic.  I can see the image in rviz though but when I try to use the data from it for a downstream process the process acts like it is getting nothing from the topic.  I am using the ros-development branch.

MartyG-RealSense, 2025-01-10:
Hi @mewescott  It is recommended that the **ros2-master** branch is used, as ros2-development is the development-in-progress version that will become the next master version of the wrapper.

Could you paste your launch log into a comment, please?  Thanks!

mewescott, 2025-01-10:
I was advised to use the development branch by the guys working on rotation.  I later found out that they are only rotating the depth image when I need both so I am not using it until they finish rotation for both.  If you think I need to switch back to get depth alignment to work I can try it.  Do I need to switch both SDK and wrapper?

sentry3@ubuntu:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py enable_rgbd:=true enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:=true depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30
[INFO] [launch]: All log files can be found below /home/sentry3/.ros/log/2025-01-10-09-29-15-834817-ubuntu-4758
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4759]
[realsense2_camera_node-1] [INFO] [1736522956.441364122] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736522956.441687105] [camera.camera]: Built with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1736522956.441750274] [camera.camera]: Running with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1736522956.475815602] [camera.camera]: Device with serial number 234322304152 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736522956.475989685] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1736522956.476031958] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1736522956.476397246] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1736522956.476432543] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736522956.693975934] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1736522956.788416193] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1736522956.788549284] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1736522956.788593605] [camera.camera]: Device Serial No: 234322304152
[realsense2_camera_node-1] [INFO] [1736522956.788628614] [camera.camera]: Device physical port: /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1736522956.788662439] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1736522956.788690279] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1736522956.788714408] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [WARN] [1736522957.035782288] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736522957.037547671] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1736522957.068126360] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1736522957.068348732] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1736522957.086352937] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736522957.086668879] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736522957.128536809] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736522957.143605780] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1736522957.181587320] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736522957.195607212] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1736522957.213663802] [camera.camera]: RealSense Node Is Up!

I am also getting a really bad depth cloud image.

[image]

Point cloud image is bad too:

[image]

Here is the terminal from it.  I have a message popping up I don't know how to correct yet:

sentry3@ubuntu:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py enable_rgbd:=true enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:=true depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30 pointcloud.enable:=true initial_reset:=true
[INFO] [launch]: All log files can be found below /home/sentry3/.ros/log/2025-01-10-09-41-29-134730-ubuntu-5339
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5340]
[realsense2_camera_node-1] [INFO] [1736523689.518846437] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736523689.519085063] [camera.camera]: Built with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1736523689.519124519] [camera.camera]: Running with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1736523689.547731068] [camera.camera]: Device with serial number 234322304152 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736523689.547895613] [camera.camera]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video12 was found.
[... cut ...]
[realsense2_camera_node-1] [INFO] [1736523696.324540464] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1736523696.590166839] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1736523697.623495643] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1736523698.698436641] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

MartyG-RealSense, 2025-01-11:
@mewescott  Switching to the master SDK version is not likely to improve your issues, so I recommend sticking with the development version as it appears to be functioning, though with non-ideal performance.

Unless you need the aligned topic to be published then you should not need to set `align_depth.enable` to true in your launch instruction, as depth and color should be automatically mapped together on the pointcloud when depth and color are true and `pointcloud.enable:=true`

However, the message `No stream match for pointcloud chosen texture Process - Color` indicates that the color frames are being dropped and are therefore not able to be displayed.

Is the pointcloud displayed correctly if you disable the color topic, enable Infra1 and set the pointcloud to use infrared to texture the pointcloud instead of RGB?

`ros2 launch realsense2_camera rs_launch.py enable_sync:=true align_depth.enable:=true enable_color:=false enable_infra1:=true enable_depth:=true depth_module.depth_profile:=1280x720x30 depth_module.infra_profile:=1280x720x30 pointcloud.enable:=true pointcloud.stream_filter:=1  initial_reset:=true`

closed 2025-01-26 by MartyG-RealSense