realsenseai/realsense-ros #3279: imu data not showing in rviz2

timbojones91, 2024-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { DD455}                          | 
| Firmware Version                   | 5.13.0.50                        | 
| Operating System & Version |   Ubuntu 22.04 Jetpack 6.0 r2 | 
| Kernel Version (Linux Only)    |  5.15.135-tegra                                                                   | 
| Platform                                 | Jetson orin nx 16gb                    |
| Librealsense SDK Version       |         2.55.1                                            | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  humble                          | 
| RealSense ROS Wrapper Version   |  {4.55.1 latest master                                                 | 

### Issue Description

I ive installed the realsense sdk with libuvc driver and the wrapper from source on an Jetson orin nx. 

Realsense Viewer runs without any problems. 

ive launched the wrapper with ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true enable_gyro:=true enable_accel:=true

rviz shows some topics but ihe imu is not functional

there is the following output in the terminal: 

nvu@nvu:~$ cd ~/ros2_ws
nvu@nvu:~/ros2_ws$ . install/local_setup.bash
nvu@nvu:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true enable_gyro:=true enable_accel:=true
[INFO] [launch]: All log files can be found below /home/nvu/.ros/log/2024-12-16-13-24-36-072980-nvu-12049
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [12050]
[realsense2_camera_node-1] [INFO] [1734351876.321039805] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1734351876.321209312] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1734351876.321237536] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1734351876.927082601] [camera.camera]: Device with serial number 031422250402 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1734351876.927215947] [camera.camera]: Device with physical ID 2-1.1-3 was found.
[realsense2_camera_node-1] [INFO] [1734351876.927243788] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1734351876.927662035] [camera.camera]: Device with port number 2-1.1 was found.
[realsense2_camera_node-1] [INFO] [1734351876.927697171] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1734351876.927840534] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1734351876.928482689] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1734351876.928527905] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1734351876.928549058] [camera.camera]: Device Serial No: 031422250402
[realsense2_camera_node-1] [INFO] [1734351876.928562338] [camera.camera]: Device physical port: 2-1.1-3
[realsense2_camera_node-1] [INFO] [1734351876.928573826] [camera.camera]: Device FW version: 5.13.0.50
[realsense2_camera_node-1] [INFO] [1734351876.928584802] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1734351876.928594563] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1734351881.378384226] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1734351881.379648503] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1734351882.348011617] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1734351882.350116229] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1734351882.350616429] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1734351882.362089071] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1734351882.362266002] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1734351882.362314995] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1734351882.494193282] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1734351882.595000294] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  16/12 13:24:42,596 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,607 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,617 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,628 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,639 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,650 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,661 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,671 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,683 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:42,694 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... 31 lines cut ...]
[realsense2_camera_node-1]  16/12 13:24:44,608 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,619 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,630 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,641 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,651 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,662 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:24:44,673 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  16/12 13:26:14,551 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:26:14,561 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/12 13:26:14,572 WARNING [281472313125088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2024-12-16:
Hi @timbojones91  The log suggests that the accel and gyro topics are being published.  The **/imu** topic, which combines accel and gyro messages into a single topic, would not be published though.  To publish the /imu topic, you should add to your launch instruction the parameter below:

`unite_imu_method:=2`

MartyG-RealSense, 2025-01-12:
Hi @timbojones91  Do you require further assistance with this case, please?  Thanks!

timbojones91, 2025-01-12:
Im good thank you 🙂

MartyG-RealSense, 2025-01-12:
That's great to hear.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2025-01-12 by MartyG-RealSense