realsenseai/realsense-ros #3282: xioctl error with the realsense D455

tillb98, 2024-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {D455}                          | 
| Firmware Version                   | (5.16.0.1)                          | 
| Operating System & Version |   {Ubuntu 22 Server} | 
| Kernel Version (Linux Only)    |  (5.15.0-1067-raspi)                                                                       | 
| Platform                                 | Raspberry Pi 4                          |
| Librealsense SDK Version       |  { 2.55.1 }                                                    | 
| Language                               |  {C++/python }         | 
| Segment			         |  {Robot}                             | 
| ROS Distro			         |  {Humble}                                           | 
| RealSense ROS Wrapper Version   |  {4.55.1}                                                  | 

### Issue Description
Dear Sir or Madam,

I am currently trying to use my Realsense 455 together with a Raspberrypi 4 for my mobile robot. The Raspberrypi is flashed with an Ubuntu Server 22.04 OS. I had already managed to read depth data from the Realsense with the wrapper for ROS 2 Humble and display it in RVIZ on my computer. 

Recently, the operation with the Pi has unfortunately stopped working for reasons that are not clear to me (see attached screenshot with log/error messages from the terminal). To rule out a hardware error, I have also successfully tested the function on my Ubuntu 22 laptop with ROS 2 Humble. I have already reinstalled the SDK and the ROS 2 wrapper several times. I read in another issue that the LibUVC-backend installation could possibly help. Unfortunately, this has not been successful either. 

Unfortunately, I can't think of any other solution strategies from this point on. Would it be possible to get help with this? 

Kind regards
[image]

MartyG-RealSense, 2024-12-18:
Hi @tillb98  A **xioctl** error typically indicates that there is a conflict between the librealsense SDK and the kernel.  This is why the **libuvc backend** method of installing the SDK (which bypasses the kernel) is the best way to install it on a Raspberry Pi.  I note though that you have tried the libuvc method and it has not been successful for you.

Usually, the best that can be achieved from using RealSense with a Pi board is a basic depth and color stream, with no additional functions such as pointcloud, alignment or post-processing filters.  IMU data also usually will not be accessible on a Pi.

Your USB connection type is being detected as the slow USB 2.1, which will further degrade performance.  If the USB cable is inserted in a USB 3.1 port on the Pi then a 2.1 detection could occur if a USB 2.1 cable is being used instead of the official USB3 cable supplied with the camera, or if the camera is plugged into a USB2 hub instead of a USB3 one.

Let's try a custom launch configuration with only depth and color enabled to see whether it makes a difference.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 rgb_camera.profile:=640x480x30`

If it does not make a difference, please then try adding **initial_reset:=true** onto the end of the launch command to reset the camera at launch.

MartyG-RealSense, 2025-01-12:
Hi @tillb98  Do you require further assistance with this case, please?  Thanks!

tillb98, 2025-01-12:
Hi Marty, 

First of all, thank you for your quick help!
As suggested, I tried changing the resolution and using a USB 3.0 cable between Christmas and New Year. Neither had any effect. Apparently, it seems to be a pure performance problem of the Raspberry Pi 4. To test this assumption, I ran the same script on a Raspberry Pi 5 with an Ubuntu 24.10 desktop OS. OS, ROS Humble and the Realsense packages were installed and executed via the M2 Hat for the Raspberry Pi 5 on an SSD hard disk. With this combination, the data transfer ran just as smoothly and error-free as when the camera was connected directly to the PC. Streaming the data and visualizing it in RVIZ 2 utilizes one of the Pi's 4 cores at 100%.

I would therefore say that you can close the issue. For my robot, I will upgrade the hardware from a Pi 4 to a Pi 5 accordingly.

Kind regards

closed 2025-01-12 by tillb98

MartyG-RealSense, 2025-01-13:
I'm pleased to hear that a Pi 5 and SSD hard drive provided significantly better performance for you.  Thanks very much for the update!