realsenseai/realsense-ros #3285: Multiple Cameras using RSUSB Backend

kushal2507, 2025-01-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Ubuntu 22.04 and 24.04 | 
| Kernel Version (Linux Only)    | 6.8                                     | 
| Platform                                 | PC  |
| SDK Version                            | 2.56.2                         | 
| ROS Version | Humble running 4.55.1, I have also tried 4.56.2 |
| Language                            |  ROS2/C++                        | 
| Segment			| Robot                       | 

### Issue Description

I have installed the librealsense SDK using the method described [here](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692). I am being able to launch only 1 camera out of the 2. This the terminal output 
```
ros2 launch bgl_hardware_launcher multi_realsense.launch.py 
[INFO] [launch]: All log files can be found below /home/bgldev/.ros/log/2025-01-08-15-29-07-299699-proto-bos-v1-chewbacca-27637
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [27638]
[INFO] [realsense2_camera_node-2]: process started with pid [27640]
[realsense2_camera_node-2] [INFO] [1736350147.726404815] [camera_rear]: RealSense ROS v4.55.1
[realsense2_camera_node-2] [INFO] [1736350147.726659728] [camera_rear]: Built with LibRealSense v2.56.2
[realsense2_camera_node-2] [INFO] [1736350147.726724705] [camera_rear]: Running with LibRealSense v2.56.2
[realsense2_camera_node-1] [INFO] [1736350147.733443198] [camera_front]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736350147.733656510] [camera_front]: Built with LibRealSense v2.56.2
[realsense2_camera_node-1] [INFO] [1736350147.733705291] [camera_front]: Running with LibRealSense v2.56.2
[realsense2_camera_node-1]  08/01 15:29:07,836 WARNING [137977860720192] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  08/01 15:29:07,846 WARNING [137977860720192] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  08/01 15:29:07,911 WARNING [137977860720192] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  08/01 15:29:07,921 WARNING [137977860720192] (handle-libusb.h:110) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-2] [INFO] [1736350147.926735875] [camera_rear]: Device with serial number 025222070751 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1736350147.927134068] [camera_rear]: Device with physical ID 2-3-27 was found.
[realsense2_camera_node-2] [INFO] [1736350147.927561561] [camera_rear]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-2] [INFO] [1736350147.928536876] [camera_rear]: Device with port number 2-3 was found.
[realsense2_camera_node-2] [INFO] [1736350147.928701103] [camera_rear]: Device USB type: 3.2
[realsense2_camera_node-2] [INFO] [1736350147.928804840] [camera_rear]: Resetting device...
[realsense2_camera_node-1]  08/01 15:29:07,931 ERROR [137977860720192] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[realsense2_camera_node-1]  08/01 15:29:07,932 ERROR [137977892177472] (uvc-sensor.cpp:427) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  08/01 15:29:07,934 ERROR [137977892177472] (rs.cpp:255) [rs2_create_device( info_list:0x7d7d50013ef0, index:0 ) UNKNOWN] failed to set power state
[realsense2_camera_node-1]  08/01 15:29:07,935 ERROR [137977892177472] (rs.cpp:255) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  08/01 15:29:07,935 WARNING [137977892177472] (rs.cpp:391) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1736350147.935208477] [camera_front]: Device 1/2 failed with exception: failed to set power state
[realsense2_camera_node-1] [INFO] [1736350148.102537714] [camera_front]: Device with serial number 031522070524 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736350148.102759345] [camera_front]: Device with physical ID 2-5-25 was found.
[realsense2_camera_node-1] [INFO] [1736350148.102800443] [camera_front]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736350148.103256030] [camera_front]: Device with port number 2-5 was found.
[realsense2_camera_node-1] [INFO] [1736350148.103313947] [camera_front]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736350148.103343365] [camera_front]: Resetting device...
[realsense2_camera_node-2]  08/01 15:29:13,977 WARNING [137901681673792] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-2]  08/01 15:29:14,043 WARNING [137901681673792] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-2] [INFO] [1736350154.085783227] [camera_rear]: Device with serial number 025222070751 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1736350154.086140187] [camera_rear]: Device with physical ID 2-3-28 was found.
[realsense2_camera_node-2] [INFO] [1736350154.086261429] [camera_rear]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-2] [INFO] [1736350154.087048411] [camera_rear]: Device with port number 2-3 was found.
[realsense2_camera_node-2] [INFO] [1736350154.087180774] [camera_rear]: Device USB type: 3.2
[realsense2_camera_node-2] [INFO] [1736350154.087542449] [camera_rear]: getParameters...
[realsense2_camera_node-2] [WARN] [1736350154.090739327] [camera_rear]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-2] [INFO] [1736350154.090938917] [camera_rear]: Publish diagnostics every 1 seconds.
[realsense2_camera_node-2] [INFO] [1736350154.094321334] [camera_rear]: JSON file is not provided
[realsense2_camera_node-2] [INFO] [1736350154.094422294] [camera_rear]: Device Name: Intel RealSense D435I
[realsense2_camera_node-2] [INFO] [1736350154.094463722] [camera_rear]: Device Serial No: 025222070751
[realsense2_camera_node-2] [INFO] [1736350154.094504345] [camera_rear]: Device physical port: 2-3-28
[realsense2_camera_node-2] [INFO] [1736350154.094533068] [camera_rear]: Device FW version: 5.16.0.1
[realsense2_camera_node-2] [INFO] [1736350154.094573664] [camera_rear]: Device Product ID: 0x0B3A
[... 77 lines cut ...]
[realsense2_camera_node-1]  08/01 15:29:16,157 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156157.208008
[realsense2_camera_node-1]  08/01 15:29:16,190 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156190.333984
[realsense2_camera_node-1]  08/01 15:29:16,223 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156222.954102
[realsense2_camera_node-1]  08/01 15:29:16,257 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156257.102295
[realsense2_camera_node-1]  08/01 15:29:16,290 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156290.681396
[realsense2_camera_node-1]  08/01 15:29:16,321 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156321.369873
[realsense2_camera_node-1]  08/01 15:29:16,355 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156355.440918
[realsense2_camera_node-1]  08/01 15:29:16,387 WARNING [137977881691712] (uvc-sensor.cpp:131) Frame received with streaming inactive,Depth0, Arrived,0.000000 1736350156386.989746
[... cut ...] iple). I have tried a different motor controller and the freeze still happens which got to conclude that it was something with the intel cameras. This was not happening on an older (I think 10th gen) PC so I figured it was something kernel related. This is why I decided to try and use the RSUSB backend method which supposedly bypasses the kernel drivers. But I am not being able to run multiple cameras through that. I have no issues running a single camera.

Can I get some help please. Thank you!

kushal2507, 2025-01-09:
Update -  I tried running 2 Intel cameras and the motor controller on Windows and the system freezes there as well. Only way to recover is to do a hard power reset.

kushal2507, 2025-01-09:
hi @MartyG-RealSense I can confirm I am experiencing the same issue. Is what I am trying to do with the RSUSB backend method possible? Can I try some other steps?

MartyG-RealSense, 2025-01-20:
Hi @kushal2507  Do you require further assistance with this case, please?  Thanks!

sebastianserna30, 2025-01-24:
> In librealsense 2.56.2, the **patch-realsense-ubuntu-lts-hwe.sh** patch script was updated with support for kernel 6.8 and Ubuntu 24.04.

Hello @MartyG-RealSense, 
would this work to patch kernel 6.8 in **ubuntu 22.04** ?

MartyG-RealSense, 2025-01-25:
Hi @sebastianserna30  The **2.56.2** version or newer of the librealsense SDK has kernel 6.8 patch support in `patch-realsense-ubuntu-lts-hwe.sh` for Ubuntu Jammy (22.04) and Noble (24.04), as described at https://github.com/IntelRealSense/librealsense/pull/13236