realsenseai/realsense-ros #3286: Low FPS and errors on the RealSense D435i on the Jetson AGX Orin

Vishnu-AP, 2025-01-15:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Linux (Ubuntu 22.04) L4T | 
| Kernel Version (Linux Only)    |  5.15.136-tegra                                                                 | 
| Platform                                 | NVIDIA Jetson AGX Orin      |
| Librealsense SDK Version       |  2.55.1                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.55.1                                                 | 

### Issue Description
I have installed done the following to setup the RealSense D435i:
1. Installed RealSense 2.55.1 SDK using the RSUSB installation mentioned in [jetson installation page](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
2. Built realsense-ros 4.55.1 from source as mentioned [here](https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#option-2-install-from-source)

Despite the firmware version and SDK version being inline, I still am not able to acheive stable FPS and I get multiple errors when I launch the realsense node with the following command: 
ros2 launch realsense2_camera rs_launch.py rgb_camera.color_profile:='640,360,60' depth_module.depth_profile:='640,360,60'

I keep getting the following errors and I am never able to acheive stable FPS as well as the actual mentioned FPS. If 60 fps is mentioned in the arguments I get around 15 only: 
[realsense2_camera_node-1]  15/01 15:40:20,511 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:40:22,594 ERROR [281472351004896] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132

I would also deeply appreciate if you can tell me the best way to setup the RealSense D345i on the Jetson AGX Orin so as to achieve the maximum resolution and FPS supported since I plan on working with high resolution data

Here is the entire output for your reference:
~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py rgb_camera.color_profile:='640,360,60' depth_module.depth_profile:='640,360,60'
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-15-15-37-51-198570-rigbetellabs-46749
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [46750]
[realsense2_camera_node-1] [INFO] [1736935671.489464629] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736935671.489669043] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736935671.489706451] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736935672.153022497] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736935672.153187584] [camera.camera]: Device with physical ID 2-3.4-4 was found.
[realsense2_camera_node-1] [INFO] [1736935672.153214528] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736935672.153581438] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1736935672.153614269] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736935672.153766108] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1736935672.155100659] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1736935672.155247123] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1736935672.155274418] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1736935672.155290802] [camera.camera]: Device physical port: 2-3.4-4
[realsense2_camera_node-1] [INFO] [1736935672.155301810] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1736935672.155313138] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1736935672.155323730] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1736935676.284217255] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736935676.285427583] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1736935677.043948904] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736935677.047550288] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1736935677.047798543] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1736935677.056386070] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736935677.057519086] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736935677.206519984] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736935677.359337336] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 360, FPS: 60
[realsense2_camera_node-1]  15/01 15:37:57,360 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1736935677.369709587] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736935677.395314665] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 360, FPS: 60
[realsense2_camera_node-1] [INFO] [1736935677.399813611] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  15/01 15:37:57,683 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:57,694 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:57,705 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:57,716 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:57,727 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:58,860 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:37:58,871 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:38:24,803 ERROR [281472351004896] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132
[... 41 lines cut ...]
[realsense2_camera_node-1]  15/01 15:40:04,441 WARNING [281473181149408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 15:40:14,454 ERROR [281472351004896] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132
[realsense2_camera_node-1]  15/01 15:40:16,779 ERROR [281472351004896] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132
[... cut ...]
[realsense2_camera_node-1] [INFO] [1736935987.175593238] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1736935987.380530450] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1736935987.380775856] [camera.camera]: Stop Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736935987.380829776] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1736935988.087544263] [camera.camera]: Close Sensor - Done. 
rs-fw[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 46750]

MartyG-RealSense, 2025-01-15:
Hi @Vishnu-AP  When the warning **control_transfer returned** generates continuously in the log, it indicates that there is a communication problem between the camera and computer / computing board.  This is typically related to a problem in the USB system, such as a glitch on the computer's USB port or a bad USB cable.

1.  Does it make a difference if you add **initial_reset:=true** to your launch instruction to reset the camera at launch?

2.  Does it make a difference if you unplug the micro-sized USB cable connector from the side of the camera, turn it around the opposite way and re-insert it (USB-C cables are two-way insertion at the micro-sized end)

3.  Does it make a difference if you add **auto_exposure_priority:=false** to your launch instruction?

4.  Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice?

Vishnu-AP, 2025-01-15:
I am not sure if this information below would help:
I tried connecting the RealSense to my laptop and I do not get the errors mentioned above, but I am not able to acheive maximum FPS when I go for a resolution like 640x360x60, I get somewhere around 40.

Trying out the steps you mentioned in order:
1. This actually solved the issue to a great extent, I was able to acheive 15 FPS with this but when I try for 30FPS and 60FPS I do not acheive complete 30 or 60 FPS, it usually caps to somewhere between 15 - 17 FPS with error message showing up less frequently (I have attached the output below)
ros2 launch realsense2_camera rs_launch.py rgb_camera.color_profile:='640,360,60' depth_module.depth_profile:='640,360,60' initial_reset:=true
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-15-17-58-38-374876-rigbetellabs-26978
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [26979]
[realsense2_camera_node-1] [INFO] [1736944118.665414276] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736944118.665610182] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736944118.665644358] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736944119.373791179] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736944119.373965101] [camera.camera]: Device with physical ID 2-3.4-8 was found.
[realsense2_camera_node-1] [INFO] [1736944119.373995437] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736944119.374355440] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1736944119.374392849] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736944119.374417873] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1736944126.246834723] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736944126.247005988] [camera.camera]: Device with physical ID 2-3.4-9 was found.
[realsense2_camera_node-1] [INFO] [1736944126.247048004] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736944126.247595177] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1736944126.247645706] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736944126.247847436] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1736944126.248650803] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1736944126.248709843] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1736944126.248750484] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1736944126.248787956] [camera.camera]: Device physical port: 2-3.4-9
[realsense2_camera_node-1] [INFO] [1736944126.248818068] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1736944126.248845621] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1736944126.248872725] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1736944130.240560454] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736944130.241740177] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1736944130.995333973] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736944130.996757826] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1736944130.996895299] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1736944131.002658231] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736944131.003229820] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736944131.111385872] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736944131.265249923] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 360, FPS: 60
[realsense2_camera_node-1]  15/01 17:58:51,268 WARNING [281471956805856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1736944131.279067104] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736944131.294254729] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 360, FPS: 60
[realsense2_camera_node-1] [INFO] [1736944131.297647688] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  15/01 17:58:51,586 WARNING [281471956805856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 17:58:51,597 WARNING [281471956805856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  15/01 17:59:08,848 WARNING [281471956805856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 17:59:14,057 ERROR [281471679916256] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132

2. Did not make any difference - still the same

3. Now when I set the FPS at 30 and 60, it caps at ~21FPS with the errors showing up less frequently (similar to the above output)
4.  I have not changed the cable, it is the black 1m cable (not sure if it is the original one or not because I am not the first one to use it)

MartyG-RealSense, 2025-01-15:
Thanks very much for the detailed feedback.  

Do you have the same problems if you use 640x480x60 resolution instead of 640x360x60

Vishnu-AP, 2025-01-15:
I tried with 640,480,60 as you mentioned, there was no change in the FPS (still ~15) but I did not get the watchdog error despite running it for a good amount of time.

Additional info: I also tried a higher resolution - 1280,720,15 and 1280,720,6 (both RGB and Depth) but the FPS caps to 6

**ros2 launch realsense2_camera rs_launch.py rgb_camera.color_profile:='640,480,60' depth_module.depth_profile:='640,480,60' initial_reset:=true auto_exposure_priority:=false**
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-15-18-20-43-836677-rigbetellabs-39363
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [39364]
[realsense2_camera_node-1] [INFO] [1736945444.124843701] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736945444.125043569] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736945444.125080592] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736945444.846350804] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736945444.846599920] [camera.camera]: Device with physical ID 2-3.4-11 was found.
[realsense2_camera_node-1] [INFO] [1736945444.846638639] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736945444.847055143] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1736945444.847097862] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736945444.847124197] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1736945451.724075046] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736945451.724280290] [camera.camera]: Device with physical ID 2-3.4-12 was found.
[realsense2_camera_node-1] [INFO] [1736945451.724341601] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1736945451.725142417] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1736945451.725250831] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736945451.725860292] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1736945451.727452325] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1736945451.727541923] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1736945451.727600130] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1736945451.727644834] [camera.camera]: Device physical port: 2-3.4-12
[realsense2_camera_node-1] [INFO] [1736945451.727685921] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1736945451.727725888] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1736945451.727761407] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1736945455.724390459] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736945455.725233995] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1736945456.480917508] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1736945456.482274507] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1736945456.482399400] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1736945456.489336550] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736945456.489689119] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736945456.625520417] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736945456.745562636] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 60
[realsense2_camera_node-1]  15/01 18:20:56,746 WARNING [281472258533600] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1736945456.751849429] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  15/01 18:20:56,758 WARNING [281472258533600] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1736945456.778845145] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 60
[realsense2_camera_node-1] [INFO] [1736945456.782395574] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  15/01 18:20:57,075 WARNING [281472258533600] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 18:20:57,086 WARNING [281472258533600] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  15/01 18:22:34,541 WARNING [281471881242848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 18:22:34,552 WARNING [281471881242848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 18:22:34,563 WARNING [281471881242848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2025-01-15:
Thanks.  And how does it perform if depth FPS is set to 30 and color FPS to 60?  Because the shutter on the D435i's RGB color sensor is slower than the shutter on the depth sensor, setting RGB to 2x the depth FPS can help to equalize performance of the two.

Vishnu-AP, 2025-01-16:
I was getting a cap of ~20FPS because I was subscribing to the topics over a wireless network (which was limiting communication). I checked the FPS local to the machine and I was getting the expected FPS! I think the initial_reset:=true argument did the job because I was getting 1 FPS on both local and remote before that, which is why I ignored the possibility that the wireless network could be a bottleneck.
Setting both depth FPS and color FPS to 60 also works just fine (~58 FPS)
So I think as of now the issue is fixed!
Thank you @MartyG-RealSense !

Vishnu-AP, 2025-01-16:
I noticed one issue, when I am running the command: **ros2 topic hz /camera/camera/color/image_raw** locally (from the same machine as the publisher), I get the FPS as passed into the arguments. But the second I run the command from a different machine (Adding the ROS_DOMAIN_ID), I see uvc streamer watchdog errors and the FPS drops to ~5
Not sure if this is an issue with the librealsense installation / ros-wrapper or if it has something to do with the DDS

Vishnu-AP, 2025-01-16:
I will try the methods mentioned in the discussion. For now I will close this issue with this comment since it has resolved the problem currently.
Thanks @MartyG-RealSense !

Vishnu-AP, 2025-01-27:
Hi There!
When I try to enable the imu on the realsense by enabling gyro and accel at 200 fps each, I do not get any data from the realsense, even the infrared images that were working perfectly without enabling the gyro and accel now fail to show any data.
Am I making a mistake in the arguments passed to the launch file? I have attached the output below - the same control transfer error which occurs multiple times a second

ros2 launch realsense2_camera rs_launch.py initial_reset:='true' enable_infra1:='true' enable_infra2:='true' depth_module.infra_profile:='640,480,30' enable_accel:='true' enable_gyro:='true' gyro_fps:='200' accel_fps:='200' unite_imu_method:='2'
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-27-11-51-47-046371-rigbetellabs-19329
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [19330]
[realsense2_camera_node-1] [INFO] [1737958907.338550642] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1737958907.338786632] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1737958907.338815975] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1737958908.087076178] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1737958908.087376070] [camera.camera]: Device with physical ID 2-3.4-5 was found.
[realsense2_camera_node-1] [INFO] [1737958908.087420997] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1737958908.088035180] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1737958908.088091530] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1737958908.088127848] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1737958914.920333683] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1737958914.920608648] [camera.camera]: Device with physical ID 2-3.4-5 was found.
[realsense2_camera_node-1] [INFO] [1737958914.920652518] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1737958914.921089300] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1737958914.921128210] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1737958914.921381544] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1737958914.922007150] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1737958914.922058027] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1737958914.922088074] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1737958914.922116009] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1737958914.922137032] [camera.camera]: Device physical port: 2-3.4-5
[realsense2_camera_node-1] [INFO] [1737958914.922153671] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1737958914.922170567] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1737958914.922185414] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1737958918.715642694] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1737958918.717490904] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1737958919.421570132] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [WARN] [1737958919.423042965] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1737958919.429269770] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1737958919.429496129] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1737958919.429569950] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1737958919.683889020] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1737958919.794058770] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1737958919.794209676] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1737958919.794239082] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  27/01 11:51:59,795 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1737958919.798149059] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  27/01 11:51:59,806 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  27/01 11:52:00,237 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  27/01 11:52:00,249 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  27/01 11:52:00,262 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2025-01-27:
Your launch instruction looks fine and the log is okay except for the **control_transfer returned** messages.  

If you are using 200 for Accel then I recommend trying 400 for Gyro instead of 200.  This is because it is best practice to configure Accel and Gyro in specific pairs of values (100 Accel and 200 Gyro, or 200 Accel and 400 Gyro) rather than setting the same value for both.

You could also try leaving the Infra1 and Infra2 = true parameters out of the launch instruction to test whether enabling depth, RGB, Infrared 1 & 2 and Motion Module all at the same time is over-burdening the Jetson and causing 'Resource temporarily unavailable' (which may occur if the CPU cannot keep up with the volume of data arriving at the computer's USB port).

Vishnu-AP, 2025-01-28:
I tried having gyro@400 FPS and accel@200FPS, I saw no change. I also tried disabling the IR streams and running only the RGBD streams @640,480,30 with initial_reset. I still saw no change. I have attached the outut below
I tried the same command which caused the issue with all the streams running on a linux machine (Laptop) and it works fine. Only on my Jetson Orin with RSUSB installation it does not work.

ros2 launch realsense2_camera rs_launch.py initial_reset:='true' rgb_camera.color_profile:='640,480,30' depth_module.depth_profile:='640,480,30' enable_accel:='true' enable_gyro:='true' gyro_fps:='400' accel_fps:='200' unite_imu_method:='2'
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-28-13-09-52-563330-rigbetellabs-14611
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [14612]
[realsense2_camera_node-1] [INFO] [1738049992.852500060] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1738049992.852719996] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738049992.852753692] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738049993.542153398] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1738049993.542330902] [camera.camera]: Device with physical ID 2-3.4-4 was found.
[realsense2_camera_node-1] [INFO] [1738049993.542356342] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1738049993.542757141] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1738049993.542797365] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1738049993.542821429] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1738050000.322044638] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1738050000.322313693] [camera.camera]: Device with physical ID 2-3.4-4 was found.
[realsense2_camera_node-1] [INFO] [1738050000.322350077] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1738050000.322774172] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1738050000.322817820] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1738050000.323056379] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1738050000.323690969] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1738050000.323731641] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1738050000.323759961] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1738050000.323782073] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1738050000.323802361] [camera.camera]: Device physical port: 2-3.4-4
[realsense2_camera_node-1] [INFO] [1738050000.323821753] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1738050000.323838905] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1738050000.323854329] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1738050004.336137565] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1738050004.337381401] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1738050005.056558168] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1738050005.057886293] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1738050005.057991413] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1738050005.062945768] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738050005.063353575] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1738050005.063415078] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1738050005.207860107] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738050005.354021099] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1]  28/01 13:10:05,355 WARNING [281472308734176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738050005.357620737] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  28/01 13:10:05,368 WARNING [281472308734176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738050005.379378824] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[... cut ...]
[realsense2_camera_node-1]  28/01 13:10:05,637 WARNING [281472308734176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/01 13:10:05,650 WARNING [281472308734176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/01 13:10:05,662 WARNING [281472308734176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2025-01-28:
Are you using JetPack 6 on your Jetson, please?  If you are then it is likely that the problem with IMU data is related to that JetPack.  

For some RealSense users, installing the librealsense SDK from source code - like you did - can sometimes fix the problem but not in all cases.  One user found that downgrading their camera's firmware driver solved it.

Would it be possible to downgrade your camera's firmware to the 5.13.0.50 version (but not one older than that) to see if it resolves your IMU problem?

Vishnu-AP, 2025-01-30:
Yes, I am using jetpack 6. I have built librealsense from source using the RSUSB backend.
I will try downgrading the firmware and update

Vishnu-AP, 2025-01-30:
I tried downgrading to 5.13.0.50 firmware version without downgrading the SDK (Currently at 2.55.1) and it worked!!
I only get the control transfer error in the very beginning, is that something to worry about? I have attached the output for reference. (I ran it for about 1min to check all the FPS values through ROS2)

ros2 launch realsense2_camera rs_launch.py initial_reset:='true' enable_infra1:='true' enable_infra2:='true' depth_module.infra_profile:='640,480,30' enable_accel:='true' enable_gyro:='true' gyro_fps:='200' accel_fps:='200' unite_imu_method:='2'
[INFO] [launch]: All log files can be found below /home/diadem/.ros/log/2025-01-30-10-30-09-034211-rigbetellabs-9689
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [9690]
[realsense2_camera_node-1] [INFO] [1738213209.323482805] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1738213209.323694992] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738213209.323724879] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738213209.992318221] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1738213209.992595589] [camera.camera]: Device with physical ID 2-3.4-6 was found.
[realsense2_camera_node-1] [INFO] [1738213209.992652835] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1738213209.993429486] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1738213209.993494316] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1738213209.993539211] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1738213216.758108972] [camera.camera]: Device with serial number 233522070955 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1738213216.758354853] [camera.camera]: Device with physical ID 2-3.4-7 was found.
[realsense2_camera_node-1] [INFO] [1738213216.758411171] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1738213216.759168623] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1738213216.759240141] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1738213216.759601251] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1738213216.760951550] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1738213216.761049275] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1738213216.761115257] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1738213216.761169080] [camera.camera]: Device Serial No: 233522070955
[realsense2_camera_node-1] [INFO] [1738213216.761206551] [camera.camera]: Device physical port: 2-3.4-7
[realsense2_camera_node-1] [INFO] [1738213216.761235286] [camera.camera]: Device FW version: 5.13.0.50
[realsense2_camera_node-1] [INFO] [1738213216.761266037] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1738213216.761293908] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1738213220.630674282] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1738213220.631579953] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1738213221.329913008] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [WARN] [1738213221.331305354] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1738213221.337021293] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738213221.337195145] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1738213221.337256583] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1738213221.633748395] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738213221.742416905] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1738213221.742575077] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1738213221.742608580] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  30/01 10:30:21,743 WARNING [281472048752864] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738213221.746008423] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  30/01 10:30:21,755 WARNING [281472048752864] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738213221.769808315] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[... cut ...]
[realsense2_camera_node-1] [INFO] [1738213292.791636193] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1738213293.127395505] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1738213293.127632875] [camera.camera]: Stop Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1738213293.151540901] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1738213293.400275077] [camera.camera]: Close Sensor - Done. 
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 9690]

MartyG-RealSense, 2025-01-30:
It's great to hear that a firmware downgrade was a successful solution for you!

If **control_transfer returned** warnings only occur at the beginning, or once or twice a minute, then they can be safely ignored.

Vishnu-AP, 2025-01-30:
Okay thats great to hear!
I will now be closing the issue with this comment as it is resolved

Thank you @MartyG-RealSense !

closed 2025-01-30 by Vishnu-AP