realsenseai/realsense-ros #3287: D405 does not publish /camera/camera/color/image_raw topic

athnzc, 2025-01-17:
**Setup 1** 

| Required Info                 |                    |
|-------------------------------|--------------------|
| Camera Model                  | D405               |
| Firmware Version              | 5.15.1             |
| Operating System & Version    | Ubuntu 22.04       |
| Kernel Version (Linux Only)   | 5.15.0-130-generic |
| Platform                      | x86_64 (Docker)    |
| Librealsense SDK Version      | v2.56.0            |
| Language                      | None               |
| Segment                       | None               |
| ROS Distro                    | Humble             |
| RealSense ROS Wrapper Version | v4.55.1            |

**Setup 2**

|  Required Info           |                            |
|-------------------------------|---------------------|
| Firmware Version              | 5.16.0.1      |
| Operating System & Version    | Ubuntu 22.04.05     |
| Kernel Version (Linux Only)   | ?                   |
| Platform                      | x86_64 (bare metal) |
| Librealsense SDK Version      | v2.55.1             |
| Language                      | None                |
| Segment                       | None                |
| ROS Distro                    | Humble              |
| RealSense ROS Wrapper Version | v4.55.1             |

Hello! We tested two different D405 devices on the above setups (one device on each setup) and in neither was the `/camera/camera/color/image_raw topic` being published. We also tested two different **D455** cameras using the same setups and they were able to publish the /camera/camera/color/image_raw topic. All devices were connected on **USB 3.2** ports. Could someone please indicate what we are doing wrong? Is the particular topic simply not available for this camera? According to [the documentation](https://dev.intelrealsense.com/docs/ros2-wrapper#published-topics), it should be published...

MartyG-RealSense, 2025-01-17:
Hi @athnzc  The D405 camera model does not have an RGB sensor like the D455 does.  Instead, it obtains its RGB color image from the depth sensor.  When programming in the librealsense SDK, this means that when querying the color stream, commands have to be directed to the depth sensor instead of the color sensor.

Something similar can be done in the ROS2 wrapper.  In May 2024, thanks to a contribution to the ROS2 wrapper - https://github.com/IntelRealSense/realsense-ros/pull/3096 - from another D405 user, the ability to access the color stream on D405 via `depth_module.color_profile` was added to the version of the rs_launch.py file on the **ros2_development** wrapper.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L51C38-L52

Changing the launch file does not require reinstalling the ROS wrapper as the launch file is separate from the wrapper.  So you could perhaps use the above launch file to perform your launch instead of the default rs_launch.py file in the **ros2-master** branch of the wrapper.

For example, you could try this with the modified launch file:

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=848x480x30 depth_module.color_profile:=848x480x30`

athnzc, 2025-01-22:
Hello @MartyG-RealSense , thank you for the swift reply!  I tried the solution you suggested. More specifically I included the modified launch file in <ros2 workspace>/src/realsense-ros/realsense2_camera/launch, built the workspace again with colcon otherwise the launch file was not recognized, sourced the workspace, and ran the command you mentioned, but I still don't see any /camera/camera/color/image_raw topic. I should note that I had already installed the ros2-development branch of the ROS wrapper. 

```
ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_rect_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color

```

athnzc, 2025-01-24:
Can you please confirm whether it is possible to get the image_raw topic from D405? I also have another question: image_rect_raw generally refers to an undistorted image stream. Is this the case for `/camera/camera/color/image_rect_raw` ?

MartyG-RealSense, 2025-01-25:
I cannot confirm it, unfortunately.  Whilst the documentation states that all RealSense cameras that can provide RGB can provide the `/camera/camera/color/image_raw` topic, unlike the other models listed there the D405 does not have a dedicated RGB sensor component and instead obtains its RGB from the depth sensor.

https://dev.intelrealsense.com/docs/ros2-wrapper#d400-stereo-depth-cameras

[image]

I note that you have `/camera/camera/color/image_rect_raw `in your ros2 topic list though, and I think that this is what you are going to have to use.

Almost all RealSense streams have a distortion model automatically applied to them (the main exception being the Y16 infrared format, which is used for camera calibration).  The distortion is calculated within the camera hardware's 'Vision Processor D4' circuit board as the data is captured and applied to the data before it is sent to the computer as a distorted, or 'rectified', stream.

closed 2025-02-13 by athnzc