realsenseai/realsense-ros #3288: Unknown FOV when aligning depth and rgb camera on D455

hlundheim, 2025-01-23:
I'm using the D455 to track the location of a laser pointer relative to a room and am using the FOV to calculate the angle to laserpoint relative to camera. After aligning depth and rgb camera using align_depth.enable:=true, output of rgb camera and aligned depth are both 640x480. Is the FOV of these outputs known? Are they the same? Are they constant? Have tried looking into this, but found no definitive answer. Can provide info about system if that's of use.

MartyG-RealSense, 2025-01-25:
Hi @hlundheim After depth to color alignment is performed when align_depth.enable is set to true, the FOV size of the depth image will be resized to the FOV of the color image.

You can find the FOV in degrees of the color stream at a specific resolution by launching the rs-enumerate-devices tool in calibration information with the command `rs-enumerate-devices -c` and scrolling through the long list of information that is output in the terminal until you find the color intrinsics for 640x480.  The values will be unique to each individual RealSense camera due to the manufacturing process at the factory.

Below is an image from my own D455 of the color intrinsics at 640x480 and its FOV.

[image]

Intrinsic values change as the resolution is changed.

hlundheim, 2025-01-29:
Thank you @MartyG-RealSense This worked as you said, no further issues.

closed 2025-01-29 by hlundheim

MartyG-RealSense, 2025-01-29:
You are very welcome, @hlundheim - I'm pleased that I could help.  Thanks very much for the update!