realsenseai/realsense-ros #3290: Unsupported device! Product ID: 0x0B5B

Akumar201, 2025-01-30:
When I am trying to run my launch file inside my docker container, I am getting following error , I am not sure is this a version issue in the firmware or I have to downgrade the SDK. I am using Intel realsense 405, also I am using docker to install these packages

```
RUN apt-get update && apt-get install -y \
    python3-pip \
    # gnome-terminal \
    # can-utils \
    # ros-noetic-socketcan-interface \
    # python3-serial \
    ros-noetic-actionlib \
    ros-noetic-control-msgs \
    ros-noetic-tf \
    ros-noetic-cv-bridge \
    ros-noetic-image-geometry \
    ros-noetic-ddynamic-reconfigure \
    ros-noetic-image-transport \
    ros-noetic-camera-info-manager \
    ros-noetic-nodelet \
    ros-noetic-diagnostic-updater \
    ros-noetic-librealsense2 \
    libeigen3-dev \
    libuvc-dev \
    libusb-1.0-0-dev \
    libgoogle-glog-dev \
    usbutils \
    # librealsense2-dev \
    curl -y \
```

```

[ INFO] [1738269548.228459113]: Device with physical ID 2-6-13 was found.
[ INFO] [1738269548.228469554]: Device with name Intel RealSense D405 was found.
[ INFO] [1738269548.228633821]: Device with port number 2-6 was found.
[ INFO] [1738269548.228656807]: Device USB type: 3.2
[FATAL] [1738269548.228701588]: Unsupported device! Product ID: 0x0B5B
[ INFO] [1738269548.232797568]: Device with serial number 218722270728 was found.

[ INFO] [1738269548.232815920]: Device with physical ID 2-4.4.3-14 was found.
[ INFO] [1738269548.232820675]: Device with name Intel RealSense D405 was found.
[ INFO] [1738269548.232972878]: Device with port number 2-4.4.3 was found.
[ INFO] [1738269548.232984286]: Device USB type: 3.2
[FATAL] [1738269548.233006231]: Unsupported device! Product ID: 0x0B5B
[cam_high/realsense2_camera_manager-2] process has died [pid 2055, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_high-realsense2_camera_manager-2.log].
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_high-realsense2_camera_manager-2*.log
[cam_left_wrist/realsense2_camera_manager-4] process has died [pid 2057, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_left_wrist-realsense2_camera_manager-4.log].
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_left_wrist-realsense2_camera_manager-4*.log
[cam_high/realsense2_camera-3] process has finished cleanly
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_high-realsense2_camera-3*.log
[cam_left_wrist/realsense2_camera-5] process has finished cleanly
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_left_wrist-realsense2_camera-5*.log
[ INFO] [1738269554.199468634]:  
[ INFO] [1738269554.221523264]: Device with serial number 218622272567 was found.

[ INFO] [1738269554.221537709]: Device with physical ID 2-4.4.1-15 was found.
[ INFO] [1738269554.221541923]: Device with name Intel RealSense D405 was found.
[ INFO] [1738269554.221650611]: Device with port number 2-4.4.1 was found.
[ INFO] [1738269554.221655467]: Device USB type: 3.2
[FATAL] [1738269554.221675408]: Unsupported device! Product ID: 0x0B5B
[cam_right_wrist/realsense2_camera_manager-6] process has died [pid 2059, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_right_wrist-realsense2_camera_manager-6.log].
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_right_wrist-realsense2_camera_manager-6*.log
[cam_right_wrist/realsense2_camera-7] process has finished cleanly
log file: /root/.ros/log/3d1af592-df4a-11ef-809b-b047e9369f84/cam_right_wrist-realsense2_camera-7*.log

```

MartyG-RealSense, 2025-01-31:
Hi @Akumar201   If you are using ROS1 Noetic then librealsense version **2.51.1** and camera firmware driver version **5.13.0.50** should be used.  

The ROS wrapper version that should be used with Noetic is ros1-legacy wrapper version 2.3.2.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

I recommend librealsense 2.51.1 rather than 2.50.0 because 2.51.1 was the version in which official support for D405 was first introduced.

I also recommend building the ROS1 wrapper from source code with installation 'Method 2'.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#method-2-the-realsense-distribution

This is because support for D405 was not included in the original release of the 2.3.2 ROS1 wrapper but was added later in the source code.  There was also a 'forked' version of the wrapper with D405 support created by a RealSense user - please see https://github.com/IntelRealSense/realsense-ros/issues/2951#issuecomment-1864509361

Akumar201, 2025-02-01:
@MartyG-RealSense , Thank you for your prompt replies, I really appreciate it.  I resolved it by building the following [rjwb1]([https://github.com/rjwb1/realsense-ros]) https://github.com/rjwb1/realsense-ros , it's working perfect, please let me know if you have any suggestions.

closed 2025-02-01 by Akumar201

MartyG-RealSense, 2025-02-01:
It's great to hear that the forked wrapper enabled your D405 to work.  Thanks very much for the update!