realsenseai/realsense-ros #3292: [realsense2_camera]: JSON file is not provided when using realsense D435 camera in isaac_ros_centerpose

monajalal, 2025-02-05:
https://forums.developer.nvidia.com/t/isaac-ros-centerpose-inference-with-intel-realsense-d435-stuck-at-unable-to-get-power-limit-for-gpu-0/322715/2?u=mona.jalal

Hi @MartyG-RealSense 
I saw https://github.com/IntelRealSense/realsense-ros/issues/2884 but I am not sure how it exactly applies to fixing my issue?

```
admin@DOS:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_examples isaac_ros_examples.launch.py launch_fragments:=realsense_mono_rect,centerpose,centerpose_visualizer model_name:=centerpose_shoe model_repository_paths:=[${ISAAC_ROS_WS}/isaac_ros_assets/models]
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2025-02-05-15-02-12-500591-DOS-15739
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [15752]
[component_container_mt-1] [INFO] [1738785732.813857183] [isaac_ros_examples.container]: Load Library: /opt/ros/humble/lib/libresize_node.so
[component_container_mt-1] [INFO] [1738785732.827032738] [isaac_ros_examples.container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::image_proc::ResizeNode>
[component_container_mt-1] [INFO] [1738785732.827056585] [isaac_ros_examples.container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::image_proc::ResizeNode>
[component_container_mt-1] [INFO] [1738785735.348455726] [centerpose_encoder.resize_node]: [NitrosNode] Initializing NitrosNode
[component_container_mt-1] [INFO] [1738785735.348656678] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/std/libgxf_std.so
[component_container_mt-1] [INFO] [1738785735.349608794] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_gxf_helpers.so
[component_container_mt-1] [INFO] [1738785735.351550108] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_sight.so
[component_container_mt-1] [INFO] [1738785735.353611199] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_atlas.so
[component_container_mt-1] 2025-02-05 15:02:15.358 WARN  gxf/std/program.cpp@538: No GXF scheduler specified.
[component_container_mt-1] [INFO] [1738785735.359016937] [centerpose_encoder.resize_node]: [ResizeNode] Set output data format to: "nitros_image_rgb8"
[component_container_mt-1] [INFO] [1738785735.359103145] [centerpose_encoder.resize_node]: [NitrosNode] Starting NitrosNode
[component_container_mt-1] [INFO] [1738785735.365014286] [centerpose_encoder.resize_node]: [NitrosNode] Loading extensions
[component_container_mt-1] [INFO] [1738785735.365063246] [centerpose_encoder.resize_node]: [NitrosContext] Loading extension: gxf/lib/multimedia/libgxf_multimedia.so
[component_container_mt-1] [INFO] [1738785735.366534825] [centerpose_encoder.resize_node]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_message_compositor.so
[component_container_mt-1] [INFO] [1738785735.366807747] [centerpose_encoder.resize_node]: [NitrosContext] Loading extension: gxf/lib/cuda/libgxf_cuda.so
[component_container_mt-1] [INFO] [1738785735.368373252] [centerpose_encoder.resize_node]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_tensorops.so
[component_container_mt-1] [INFO] [1738785735.370403984] [centerpose_encoder.resize_node]: [NitrosNode] Loading graph to the optimizer
[component_container_mt-1] [INFO] [1738785735.372631660] [centerpose_encoder.resize_node]: [NitrosNode] Running optimization
[component_container_mt-1] [INFO] [1738785735.480341516] [centerpose_encoder.resize_node]: [NitrosNode] Obtaining graph IO group info from the optimizer
[component_container_mt-1] [INFO] [1738785735.482505191] [centerpose_encoder.resize_node]: [NitrosPublisherSubscriberGroup] Pinning the component "image_sink/sink" (type="nvidia::isaac_ros::MessageRelay") to use its compatible format only: "nitros_image_rgb8"
[component_container_mt-1] [INFO] [1738785735.488608579] [centerpose_encoder.resize_node]: [NitrosNode] Starting negotiation...
[component_container_mt-1] [INFO] [1738785735.489255523] [isaac_ros_examples.container]: Load Library: /opt/ros/humble/lib/librealsense2_camera.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/centerpose_encoder/resize_node' in container '/isaac_ros_examples/container'
[component_container_mt-1] [INFO] [1738785735.559955594] [isaac_ros_examples.container]: Found class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[component_container_mt-1] [INFO] [1738785735.560014323] [isaac_ros_examples.container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[component_container_mt-1] [INFO] [1738785735.574216809] [realsense2_camera]: RealSense ROS v4.51.1
[component_container_mt-1] [INFO] [1738785735.574592880] [realsense2_camera]: Built with LibRealSense v2.55.1
[component_container_mt-1] [INFO] [1738785735.574604803] [realsense2_camera]: Running with LibRealSense v2.55.1
[component_container_mt-1] [INFO] [1738785735.575161031] [isaac_ros_examples.container]: Load Library: /opt/ros/humble/lib/libimage_format_converter_node.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense2_camera' in container '/isaac_ros_examples/container'
[component_container_mt-1] [INFO] [1738785735.576271082] [isaac_ros_examples.container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::image_proc::ImageFormatConverterNode>
[component_container_mt-1] [INFO] [1738785735.576295046] [isaac_ros_examples.container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::image_proc::ImageFormatConverterNode>
[component_container_mt-1] [INFO] [1738785735.583623185] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Initializing NitrosNode
[component_container_mt-1] [INFO] [1738785735.583879479] [centerpose_encoder.image_format_converter_node]: [ImageFormatConverterNode] Set output data format to: "nitros_image_rgb8"
[component_container_mt-1] [INFO] [1738785735.583944306] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Starting NitrosNode
[component_container_mt-1] [INFO] [1738785735.606913634] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Loading extensions
[component_container_mt-1] [INFO] [1738785735.608928260] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Loading graph to the optimizer
[component_container_mt-1] [INFO] [1738785735.611928192] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Running optimization
[component_container_mt-1] [INFO] [1738785735.663628175] [realsense2_camera]: Device with serial number 819112070232 was found.
[component_container_mt-1] 
[component_container_mt-1] [INFO] [1738785735.664320267] [realsense2_camera]: Device with physical ID 3-7.2-71 was found.
[component_container_mt-1] [INFO] [1738785735.664330537] [realsense2_camera]: Device with name Intel RealSense D435 was found.
[component_container_mt-1] [INFO] [1738785735.664472386] [realsense2_camera]: Device with port number 3-7.2 was found.
[component_container_mt-1] [INFO] [1738785735.664480336] [realsense2_camera]: Device USB type: 2.1
[component_container_mt-1] [WARN] [1738785735.664486124] [realsense2_camera]: Device 819112070232 is connected using a 2.1 port. Reduced performance is expected.
[component_container_mt-1] [INFO] [1738785735.665557935] [realsense2_camera]: getParameters...
[component_container_mt-1] [INFO] [1738785735.665758982] [realsense2_camera]: JSON file is not provided
[component_container_mt-1] [INFO] [1738785735.665768182] [realsense2_camera]: Device Name: Intel RealSense D435
[component_container_mt-1] [INFO] [1738785735.665773148] [realsense2_camera]: Device Serial No: 819112070232
[component_container_mt-1] [INFO] [1738785735.665777700] [realsense2_camera]: Device physical port: 3-7.2-71
[component_container_mt-1] [INFO] [1738785735.665782564] [realsense2_camera]: Device FW version: 5.16.0.1
[component_container_mt-1] [INFO] [1738785735.665786669] [realsense2_camera]: Device Product ID: 0x0B07
[component_container_mt-1] [INFO] [1738785735.665790873] [realsense2_camera]: Sync Mode: Off
[component_container_mt-1] [INFO] [1738785735.682216615] [centerpose_encoder.image_format_converter_node]: [NitrosNode] Obtaining graph IO group info from the optimizer
[... cut ...]
2025-02-05 15:17:37 [mona@DOS ~]$ docker ps
CONTAINER ID   IMAGE                                                 COMMAND                  CREATED             STATUS                 PORTS                                                 NAMES
bc06d010ceba   isaac_ros_dev-x86_64                                  "/usr/local/bin/scri…"   About an hour ago   Up About an hour                                                             isaac_ros_dev-x86_64-container

```

Please let me know if you may need more info for debugging this issue?

MartyG-RealSense, 2025-02-06:
Hi @monajalal  It looks as though the GPU message is caused by TensorRT and not librealsense. 

The RealSense ROS wrapper launch completes successfully with the log message **RealSense Node Is Up!** but then multiple copies of the **control_transfer returned** warning are generated in the log.  

`control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

A couple of causes of this message can be:

1.  A communication problem between the camera and the computer, typically related to a problem with the USB port or USB cable;   or

2.  Data is arriving at the USB port faster than the computer's CPU can process it.  Reducing the resolution or FPS speed of the RealSense streams may reduce the number of control_transfer warnings that are published in the log.

I note that your camera connection is being detected as slow USB 2.1 instead of fast USB 3.2.  If your camera is plugged into a USB 3 port on the Jetson, please try unplugging the micro-sized end of the USB cable from the side of the camera, turning the connector around the other way and re-inserting it to see whether the camera is then detected as USB 3.2.

monajalal, 2025-02-06:
@MartyG-RealSense 

I have a Thinkpad laptop that has two USB ports. Both USBs on the side looks like to be USB 2.1 (older looking usb ports). I also used a StarTech.com USB hub and still shows USB 2.1. The USB-c behind the laptop should have been 3.2 but I don't know what may have been the problem that still showed 3.2.

It looks like with usb 3.2 the fps for color frame of D435 is 30 however with usb 2.1 it will be lower.

The usb-c looking ports also connected to camera show 2.1.

```
2025-02-06 16:44:34 [mona@DOS ~]$ lspci -v | grep -i 'usb controller\|xhci'
00:0d.0 USB controller: Intel Corporation Tiger Lake-H Thunderbolt 4 USB Controller (rev 05) (prog-if 30 [XHCI])
	Subsystem: Lenovo Tiger Lake-H Thunderbolt 4 USB Controller
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:0d.2 USB controller: Intel Corporation Tiger Lake-H Thunderbolt 4 NHI #0 (rev 05) (prog-if 40 [USB4 Host Interface])
00:14.0 USB controller: Intel Corporation Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller (rev 11) (prog-if 30 [XHCI])
	Subsystem: Lenovo Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci

```

I can see 3.2 here though: 	`Subsystem: Lenovo Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller`

Here is my USB port speeds:
```
2025-02-06 16:46:27 [mona@DOS ~]$ sudo lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 20000M/x2
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 4: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 2, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 3, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 4, Class=Application Specific Interface, Driver=, 480M
    |__ Port 7: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 14: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
```

```
2025-02-06 16:47:40 [mona@DOS ~]$ sudo dmidecode -s system-product-name
20YS004SUS

```

Is there a certain cable I should buy for the usb-c connector to camera? It's possible that USB-c to USB-c cable I used yesterday was the problem?!

MartyG-RealSense, 2025-02-07:
Thanks very much for the information that you are using a ThinkPad with USB 2.1 ports with isaac-ros and not an Nvidia Jetson.

Ports that are USB 3 can typically be identified by a small SS (super speed) logo beside them.

[image]

Sometimes a RealSense camera will be mis-detected as USB 2.1 even if inserted into a USB 3 port.  

Ideally the official 1 meter USB cable supplied with the camera will be used.  If you choose your own USB-C cable then it should be a high quality one that is designed for data transfer and not an inexpensive cable meant only for device charging or syncing a device to a computer.

USB C to C cables do often have problems with RealSense cameras.

MartyG-RealSense, 2025-02-07:
My understanding is that **usb 3-5** is an indicator that it is a USB 3 port, with each port having a different 3-number to differentiate them from each other.

closed 2025-02-07 by monajalal

monajalal, 2025-02-10:
@MartyG-RealSense thanks for your input. We had multiple new D435 cameras, I used a new everything, including its usb-3 cable. I also ended up using a Thunderbolt 3 Dock Gen 2 that has blue USB connections.

[image]

Could you please link to outdoor/amazon vendors who have the same exact cable but longer for our usecase?