realsenseai/realsense-ros #3295: purpose of "re-enable the stream for the change to take effect"

christian-rauch, 2025-02-07:
During the launch of the `realsense2_camera_node` with a custom configuration, I see
```
re-enable the stream for the change to take effect.
```
twice in my screen logs. I am manually setting the `color_profile` and `depth_profile` to `640x480x60` each.

I assume this message is shown since the camera's default `1280x720x30` and `848x480x30` respectively is changed to `640x480x60`.

From the message itself, it is not really clear how this is caused and what the effect of not "re-enable the stream" will be.

Starting the node with a custom `color_profile`:
```sh
ros2 run realsense2_camera realsense2_camera_node --ros-args -p rgb_camera.color_profile:="640x480x60"
```
shows the `re-enable the stream for the change to take effect.` message.

However, inspecting the parameter (`ros2 param get /camera/camera rgb_camera.color_profile`) and the topic (`ros2 topic echo --no-arr /camera/camera/color/image_raw`) shows me that the parameter is set (`String value is: 640x480x60`) and that the image has the expected dimension:
```
height: 480
width: 640
encoding: rgb8
is_bigendian: 0
step: 1920
data: '<sequence type: uint8, length: 921600>'
```

So, what is the purpose of this warning? And, if it is important to "re-enable the stream", why is this not done automatically?

MartyG-RealSense, 2025-02-07:
Hi @christian-rauch  Typically, in the ROS2 launch log the message `re-enable the stream for the change to take effect` is followed by messages such as these:

```
Stopping Sensor: Stereo Module
Starting Sensor: Stereo Module
```

So the launch is automatically stopping and then re-starting sensors in order to apply the change.  It is normal and you do not need to take any action.

christian-rauch, 2025-02-07:
Indeed, I see this in the full log:
```
re-enable the stream for the change to take effect.
Stopping Sensor: Depth Module
Stopping Sensor: RGB Camera
Stopping Sensor: Motion Module
Starting Sensor: Depth Module
Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
Starting Sensor: RGB Camera
Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 60
```
I think this is then the same for setting `unite_imu_method` other than `0`, which will cause:
```
For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
```

but also seems to be handled automatically:
```
Stopping Sensor: Motion Module
Stopping Sensor: Motion Module
```

Since I always see the `Stopping` / `Starting` messages, also with the defaults:
```
Stopping Sensor: Depth Module
Stopping Sensor: RGB Camera
Stopping Sensor: Motion Module
Starting Sensor: Depth Module
[...]
Starting Sensor: RGB Camera
[...]
Starting Sensor: Motion Module
```
it is not really clear if this is done automatically because of the warnings or if a user should still stop/starting them again manually.

I honestly find these warnings confusing as it appears the user is supposed to do something. What purpose do these warnings server if they are handled automatically? Can you just remove them?

MartyG-RealSense, 2025-02-07:
There have not been previous requests to remove the 're-enable the stream for the change to take effect' warning message so users of the ROS wrapper seem to be okay with it.  If a number of users reported that it caused confusion for them then it could certainly be looked at in future, and if any users reading this issue feel confusion about it then they are very welcome to leave a comment here.

MartyG-RealSense, 2025-02-21:
Hi @christian-rauch  Do you require further assistance with this case, please?  Thanks!

christian-rauch, 2025-02-21:
> Hi [@christian-rauch](https://github.com/christian-rauch) Do you require further assistance with this case, please? Thanks!

It's clear to me now that the warning can be ignored. But it sounded as if you wanted to leave this issue open for others to comment on. But I am also happy to close this now.

MartyG-RealSense, 2025-02-21:
Okay, thanks very much.  As there have not been any responses from others in the last 2 weeks, let's close it and if others comment here in future to report a problem like you had then we can re-open it.  Thanks again!

closed 2025-02-21 by christian-rauch