realsenseai/realsense-ros #3296: I tried executing roslaunch realsense2_camera rs_camera.launch But the following error is displayed. How to solve it? I use 20.04.6 and D435.

Chiawei0824isme, 2025-02-11:
[ERROR] [1739241526.491897238]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1739241526.492033626]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1739241526.492476641]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 20202, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/mini2/.ros/log/4eff8a22-e821-11ef-8f43-d39dc1adc5a5/camera-realsense2_camera-3.log].
log file: /home/mini2/.ros/log/4eff8a22-e821-11ef-8f43-d39dc1adc5a5/camera-realsense2_camera-3*.log

Chiawei0824isme, 2025-02-11:
I am using Shumei Pi 4 to plug in USB3.0.

Chiawei0824isme, 2025-02-11:
I entered this command

cd ~/catkin_ws
catkin_make clean
catkin_make
source devel/setup.bash

He made a new mistake.

roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/mini2/.ros/log/8ee77074-e83e-11ef-abf8-ad41633dd9cd/roslaunch-mini2-pi-70945.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.22.11.2:42499/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 30 lines cut ...]
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://172.22.11.2:11311

process[camera/realsense2_camera_manager-1]: started with pid [70985]
process[camera/realsense2_camera-2]: started with pid [70986]
[ INFO] [1739255335.880207170]: Initializing nodelet with 4 worker threads.
[ INFO] [1739255336.265419505]: RealSense ROS v2.3.2
[ INFO] [1739255336.265540411]: Built with LibRealSense v2.50.0
[ INFO] [1739255336.265605040]: Running with LibRealSense v2.50.0
[ WARN] [1739255336.376466863]: No RealSense devices were found!

MartyG-RealSense, 2025-02-11:
Hi @Chiawei0824isme  What method did you use to install the librealsense SDK before you performed the ROS1 wrapper installation on your Raspberry Pi 4, please?

Chiawei0824isme, 2025-02-11:
sudo apt install -y cmake git libusb-1.0-0-dev libssl-dev libgtk-3-dev pkg-config
cd ~
git clone https://github.com/IntelRealSense/librealsense
cd librealsense
mkdir build
cd build
cmake ..
make
sudo make install

Chiawei0824isme, 2025-02-11:
To supplement my download, use the following command

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

MartyG-RealSense, 2025-02-11:
Thanks very much for the information.

Your ROS1 wrapper installation procedure looks fine.  However, for librealsense installation on a Raspberry Pi I would recommend changing your **cmake ..** line to the one below.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

Chiawei0824isme, 2025-02-12:
I still have the following problem, how should I deal with it?

mini2@mini2-pi:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/mini2/.ros/log/214c3144-e8e8-11ef-aa53-c9b5b3a73c5f/roslaunch-mini2-pi-10819.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.22.11.2:46871/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 28 lines cut ...]
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10866]
ROS_MASTER_URI=http://172.22.11.2:11311

setting /run_id to 214c3144-e8e8-11ef-aa53-c9b5b3a73c5f
process[rosout-1]: started with pid [10878]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [10885]
process[camera/realsense2_camera-3]: started with pid [10890]
[ INFO] [1739326919.916746971]: Initializing nodelet with 4 worker threads.
[ INFO] [1739326921.581590706]: RealSense ROS v2.3.2
[ INFO] [1739326921.581713213]: Built with LibRealSense v2.50.0
[ INFO] [1739326921.581781596]: Running with LibRealSense v2.50.0
[ WARN] [1739326921.682429358]: No RealSense devices were found!

Chiawei0824isme, 2025-02-12:
我重新插usb並輸入lsusb，新增了以下裝置
Bus 001 Device 009: ID 2bc5:055c  
Bus 001 Device 010: ID 2bc5:065c  USB2.0 Hub
Bus 001 Device 008: ID 1a40:0101 Terminus Technology Inc. Hub

Chiawei0824isme, 2025-02-12:
I typed dmesg | grep uvc
It shows the following error, how to deal with it?

[    7.950743] uvcvideo: Found UVC 1.00 device Orbbec Gemini E RGB Camera (2bc5:055c)
[    7.953936] uvcvideo: Failed to query (GET_INFO) UVC control 7 on unit 1: 0 (exp. 1).
[    7.954152] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 1: 0 (exp. 1).
[    7.954317] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 1: 0 (exp. 1).
[    7.954509] uvcvideo: Failed to query (GET_INFO) UVC control 12 on unit 1: 0 (exp. 1).
[    7.954741] uvcvideo: Failed to query (GET_INFO) UVC control 14 on unit 1: 0 (exp. 1).
[    7.958077] uvcvideo: Failed to query (GET_INFO) UVC control 16 on unit 1: 0 (exp. 1).
[    7.966454] usbcore: registered new interface driver uvcvideo
[ 2371.687483] usbcore: deregistering interface driver uvcvideo
[ 3411.108314] uvcvideo: loading out-of-tree module taints kernel.
[ 3411.113544] uvcvideo: Found UVC 1.00 device Orbbec Gemini E RGB Camera (2bc5:055c)
[ 3411.114487] uvcvideo: Failed to query (GET_INFO) UVC control 7 on unit 1: 0 (exp. 1).
[ 3411.114653] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 1: 0 (exp. 1).
[ 3411.114809] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 1: 0 (exp. 1).
[ 3411.114980] uvcvideo: Failed to query (GET_INFO) UVC control 12 on unit 1: 0 (exp. 1).
[ 3411.115143] uvcvideo: Failed to query (GET_INFO) UVC control 14 on unit 1: 0 (exp. 1).
[ 3411.115478] uvcvideo: Failed to query (GET_INFO) UVC control 16 on unit 1: 0 (exp. 1).
[ 3411.120309] Modules linked in: uvcvideo(OE+) rfcomm xt_MASQUERADE xt_state xt_conntrack ipt_REJECT nf_reject_ipv4 algif_hash iptable_filter xt_tcpudp bpfilter nf_nat_h323 nf_conntrack_h323 aes_neon_blk crypto_simd nf_nat_pptp cryptd nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp algif_skcipher nf_nat_sip af_alg nf_conntrack_sip bnep nf_nat_irc nf_conntrack_irc nf_nat_ftp nf_conntrack_ftp iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hci_uart btqca btbcm btintel vc4 binfmt_misc drm_kms_helper snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_bcm2835(CE) snd_pcm_oss snd_mixer_oss btsdio snd_pcm bluetooth snd_seq_dummy snd_seq_oss snd_seq_midi snd_seq_midi_event ecdh_generic ecc snd_rawmidi ftdi_sio cp210x usbserial brcmfmac snd_seq bcm2835_codec(CE) bcm2835_v4l2(CE) bcm2835_isp(CE) brcmutil v4l2_mem2mem snd_seq_device bcm2835_mmal_vchiq(CE) videobuf2_dma_contig videobuf2_vmalloc snd_timer cfg80211 videobuf2_memops videobuf2_v4l2 videobuf2_common crct10dif_ce
[ 3411.120369]  videodev raspberrypi_hwmon snd mc vc_sm_cma(CE) fb_sys_fops rpivid_mem syscopyarea sysfillrect sysimgblt uio_pdrv_genirq uio sch_fq_codel ppdev lp parport ramoops reed_solomon ip_tables x_tables autofs4 v3d gpu_sched drm spidev phy_generic [last unloaded: uvcvideo]
[ 3411.120424] lr : uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3411.120492]  uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3411.120502]  uvc_register_video_device+0x5c/0x1a0 [uvcvideo]
[ 3411.120511]  uvc_probe+0x1e68/0x1f58 [uvcvideo]
[ 3411.120557]  uvc_init+0x38/0x1000 [uvcvideo]
[ 3411.120855] Modules linked in: uvcvideo(OE+) rfcomm xt_MASQUERADE xt_state xt_conntrack ipt_REJECT nf_reject_ipv4 algif_hash iptable_filter xt_tcpudp bpfilter nf_nat_h323 nf_conntrack_h323 aes_neon_blk crypto_simd nf_nat_pptp cryptd nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp algif_skcipher nf_nat_sip af_alg nf_conntrack_sip bnep nf_nat_irc nf_conntrack_irc nf_nat_ftp nf_conntrack_ftp iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hci_uart btqca btbcm btintel vc4 binfmt_misc drm_kms_helper snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_bcm2835(CE) snd_pcm_oss snd_mixer_oss btsdio snd_pcm bluetooth snd_seq_dummy snd_seq_oss snd_seq_midi snd_seq_midi_event ecdh_generic ecc snd_rawmidi ftdi_sio cp210x usbserial brcmfmac snd_seq bcm2835_codec(CE) bcm2835_v4l2(CE) bcm2835_isp(CE) brcmutil v4l2_mem2mem snd_seq_device bcm2835_mmal_vchiq(CE) videobuf2_dma_contig videobuf2_vmalloc snd_timer cfg80211 videobuf2_memops videobuf2_v4l2 videobuf2_common crct10dif_ce
[ 3411.120906]  videodev raspberrypi_hwmon snd mc vc_sm_cma(CE) fb_sys_fops rpivid_mem syscopyarea sysfillrect sysimgblt uio_pdrv_genirq uio sch_fq_codel ppdev lp parport ramoops reed_solomon ip_tables x_tables autofs4 v3d gpu_sched drm spidev phy_generic [last unloaded: uvcvideo]
[ 3411.120953] lr : uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3411.121016]  uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3411.121025]  uvc_register_video_device+0x5c/0x1a0 [uvcvideo]
[ 3411.121034]  uvc_meta_register+0x50/0x60 [uvcvideo]
[ 3411.121043]  uvc_probe+0x1e74/0x1f58 [uvcvideo]
[ 3411.121083]  uvc_init+0x38/0x1000 [uvcvideo]
[ 3411.121249] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 3 was not initialized!
[ 3411.121256] uvcvideo 1-1.1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 3411.121262] uvcvideo 1-1.1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 3411.121692] usbcore: registered new interface driver uvcvideo
[ 3416.973461] uvcvideo: Found UVC 1.00 device Orbbec Gemini E RGB Camera (2bc5:055c)
[ 3416.974379] uvcvideo: Failed to query (GET_INFO) UVC control 7 on unit 1: 0 (exp. 1).
[ 3416.974553] uvcvideo: Failed to query (GET_INFO) UVC control 9 on unit 1: 0 (exp. 1).
[ 3416.974721] uvcvideo: Failed to query (GET_INFO) UVC control 10 on unit 1: 0 (exp. 1).
[ 3416.974888] uvcvideo: Failed to query (GET_INFO) UVC control 12 on unit 1: 0 (exp. 1).
[... cut ...]
[ 3422.100242] lr : uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3422.100307]  uvc_queue_init+0x7c/0xd0 [uvcvideo]
[ 3422.100314]  uvc_register_video_device+0x5c/0x1a0 [uvcvideo]
[ 3422.100321]  uvc_meta_register+0x50/0x60 [uvcvideo]
[ 3422.100328]  uvc_probe+0x1e74/0x1f58 [uvcvideo]
[ 3422.100662] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 3 was not initialized!
[ 3422.100673] uvcvideo 1-1.1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 3422.100678] uvcvideo 1-1.1.3:1.0: Entity type for entity Camera 1 was not initialized!

Chiawei0824isme, 2025-02-13:
I used the sudo realsense-viewer command and it still didn't appear.

Chiawei0824isme, 2025-02-13:
My Platform camera will disappear if I unplug the D435. My raspberry-pi-4 only has one D435 camera and no other cameras. I use the USB3.0 transmission cable provided by the factory.

Chiawei0824isme, 2025-02-13:
I used the cheese command and the D435 screen was displayed, but when I used the sudo realsense-viewer command, it still couldn't be read.

Chiawei0824isme, 2025-02-13:
Replenish
I use roslaunch realsense2_camera rs_camera.launch initial_reset:=true
The command is to display No RealSense devices were found!

... logging to /home/mini2/.ros/log/11bb48fe-e9cf-11ef-88d2-c53eee98bf1d/roslaunch-mini2-pi-8300.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.22.11.2:44453/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 29 lines cut ...]
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8337]
ROS_MASTER_URI=http://172.22.11.2:11311

setting /run_id to 11bb48fe-e9cf-11ef-88d2-c53eee98bf1d
process[rosout-1]: started with pid [8363]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8366]
process[camera/realsense2_camera-3]: started with pid [8368]
[ INFO] [1739426108.615985487]: Initializing nodelet with 4 worker threads.
[ INFO] [1739426110.673884367]: RealSense ROS v2.3.2
[ INFO] [1739426110.674870864]: Built with LibRealSense v2.50.0
[ INFO] [1739426110.675425806]: Running with LibRealSense v2.50.0
[ WARN] [1739426110.852416407]: No RealSense devices were found!

Chiawei0824isme, 2025-02-13:
I use lsusb display
Bus 001 Device 009: ID 2bc5:055c  
Bus 001 Device 010: ID 2bc5:065c  USB2.0 Hub
Bus 001 Device 008: ID 1a40:0101 Terminus Technology Inc. Hub

Chiawei0824isme, 2025-02-13:
I've found the problem, thank you.

athnzc, 2025-02-13:
Hello @Chiawei0824isme , could you please include the solution you found because other people may need it in the future? Thank you!

MartyG-RealSense, 2025-02-13:
@Chiawei0824isme  I'm pleased to hear that you found a solution.  Thanks very much for letting me know!

AldoPenaGa, 2025-02-19:
@Chiawei0824isme could you provide the solution? I have the exact same problem

Chiawei0824isme, 2025-02-20:
I refer to this installation method.
https://richard98hess444.medium.com/realsense%E5%AE%89%E8%A3%9D-7b10392fad4b

Chiawei0824isme, 2025-02-20:
> [@Chiawei0824isme](https://github.com/Chiawei0824isme)你能提供解決方案嗎？我有同樣的問題

我參考的是這個安裝方法。
https://richard98hess444.medium.com/realsense%E5%AE%89%E8%A3%9D-7b10392fad4b

closed 2025-03-08 by MartyG-RealSense