realsenseai/realsense-ros #3297: Problem with colcon build

pramodcoboticca, 2025-02-11:
@MartyG-RealSense 

I am getting an error while starting the intel real sense D435i in Ros2

Starting >>> realsense2_camera_msgs
Starting >>> helper_bot
Starting >>> rplidar_ros
Finished <<< rplidar_ros [0.65s]                                           
Finished <<< realsense2_camera_msgs [0.82s]                                  
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [0.11s]              
Finished <<< helper_bot [1.34s]                          
--- stderr: realsense2_camera                         
CMake Error at CMakeLists.txt:143 (message):
  Unsupported ROS Distribution: humble

---
Failed   <<< realsense2_camera [1.09s, exited with code 1]

Summary: 4 packages finished [2.52s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera

**ROS2 topic list**

ros2 topic list
/parameter_events
/rosout

MartyG-RealSense, 2025-02-11:
Hi @pramodcoboticca  Which version of the RealSense ROS2 wrapper are you installing with **colcon build**, please?  I note that the message `Unsupported ROS Distribution: humble` is displayed.  Humble was first supported in the **4.51.1** version of the wrapper.

MartyG-RealSense, 2025-02-11:
Usually if you were performing a colcon build of the RealSense ROS2 wrapper then it would install the current latest version of the wrapper.  A list of versions is available at the link below.

https://github.com/IntelRealSense/realsense-ros/releases

Following the wrapper's instructions for a colcon build should cause the latest version to be installed from the **ros2-master** GitHub branch of the wrapper.

If colcon build installs the latest available version on a particular wrapper branch but your build is not supporting Humble, then I wonder if you might have colcon-built an earlier generation of the RealSense ROS2 wrapper on the **ros2-legacy** GitHub branch.  Do the instructions at the link below look like the ones that you used?

https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy?tab=readme-ov-file#step-3-install-intel-realsense-ros2-wrapper-from-sources

pramodcoboticca, 2025-02-11:
Hi @MartyG-RealSense i installed the latest version wrapper and the intel real sensor is working now on intel real sense viewer but i am getting error with rviz and i am getting ros topic 
ros2 topic list
/parameter_events
/rosout
robot@r

[image]

[image]

MartyG-RealSense, 2025-02-11:
It's great to hear that you were able to get the camera working in the RealSense Viewer tool.

In regard to the ROS wrapper, if you compare the above list of topics in your image to the expected list of topics in the wrapper documentation at the link below then they match up closely, suggesting that the wrapper is publishing the ROS2 topics correctly.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#published-topics

I note that the **metadata** topics are absent, which suggests that camera metadata might not be enabled on your installation of the librealsense SDK.  Metadata is enabled when installing librealsense from DKMS packages, or when building librealsense from source code with CMake and doing one of the following:

1.  including the build flags -dforce_libuvc=true OR -dforce_rsusb_backend=true in the CMake build instruction;  or
2. applying a librealsense kernel patch script to the Linux kernel.

Compressed topics will not be published unless the **image_transport** plugin is installed using the command below.

`sudo apt install ros-humble-image-transport`

https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

closed 2025-02-28 by MartyG-RealSense