realsenseai/realsense-ros #3300: How to modify the. launch language to change the resolution of the D435 RGB image from the default 640 x 480

wenmingxiaohuo, 2025-02-12:


MartyG-RealSense, 2025-02-12:
Hi @wenmingxiaohuo   There are different methods in the ROS1 and ROS2 wrappers for defining a custom resolution.

**ROS1**

You need to specify three factors for a stream - height, width and FPS.  For example, to define 848x480 at 30 FPS for the RGB color stream in the launch instruction::

`roslaunch realsense2_camera rs_camera.launch color_width:=848 color_height:=480 color-fps:-30`

**ROS2**

In the ROS2 wrapper you define an RGB camera 'profile'.

`realsense2_camera/launch/rs_launch.py rgb_camera.color_profile:=848x480x30`

wenmingxiaohuo, 2025-02-17:
Thank you very much. With your guidance, my question has been answered. I am using ros1. Perhaps changing the color_fps: -30 in the above command to color_fps:=30 would be more appropriate. Thank you again and best wishes

MartyG-RealSense, 2025-02-17:
Yes, it should be **color_fps:=30**.  The -30 was a typing mistake on my part.  I have edited the comment above to fix it.  My apologies!

closed 2025-03-08 by MartyG-RealSense