realsenseai/realsense-ros #3301: AttributeError: module 'em' has no attribute 'Interpreter'

Debactor, 2025-02-13:
When I colcon build the ros2 humble workspace in Ubuntu 22.04 on Nvidia Jetson Orin Nano, the error occured:

robot@robot:~/prj/realsense_ws$ colcon build
Starting >>> realsense2_camera_msgs
--- stderr: realsense2_camera_msgs                         
CMake Error at /opt/ros/humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(/home/robot/miniconda3/bin/python3 -m rosidl_adapter
  --package-name realsense2_camera_msgs --arguments-file
  /home/robot/prj/realsense_ws/build/realsense2_camera_msgs/rosidl_adapter__arguments__realsense2_camera_msgs.json
  --output-dir
  /home/robot/prj/realsense_ws/build/realsense2_camera_msgs/rosidl_adapter/realsense2_camera_msgs
  --output-file
  /home/robot/prj/realsense_ws/build/realsense2_camera_msgs/rosidl_adapter/realsense2_camera_msgs.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
      _interpreter = em.Interpreter(

  AttributeError: module 'em' has no attribute 'Interpreter'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "/home/robot/miniconda3/lib/python3.10/runpy.py", line 196, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/home/robot/miniconda3/lib/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:39 (rosidl_generate_interfaces)

---
Failed   <<< realsense2_camera_msgs [0.90s, exited with code 1]
                                
Summary: 0 packages finished [1.49s]
  1 package failed: realsense2_camera_msgs
  1 package had stderr output: realsense2_camera_msgs
  2 packages not processed
 
May I ask how to deal with it?

MartyG-RealSense, 2025-02-13:
Hi @Debactor  I have not seen this particular error `AttributeError: module 'em' has no attribute 'Interpreter'` before in relation to RealSense.  However, my research of your question indicated that some people in non-RealSense discussions who also encountered it were able to resolve it by inputting the two commands below.

```
pip uninstall em
pip install empty
```

closed 2025-02-28 by MartyG-RealSense

MartyG-RealSense, 2026-01-16:
Please next try installing an older version of empy:

```
pip uninstall em
pip install empy==3.3.4
```