realsenseai/realsense-ros #3302: D455 Fails to Start – ROS 2 Humble & LibRealSense v2.55.1

Ragnar7982, 2025-02-18:
Hello, I'm using Ubuntu22.04 ROS2 Humble, RealSense ROS v4.55.1, LibRealSense v2.55.1 and with an Intel RealSense D455.
Since yesterday, my D455 has not been able to start properly. It used to work fine under the same setup, so this issue is unexpected. I have ruled out cable-related problems since I have been using the same setup for a long time without issues. The error log is as follows:

[d455_problems.txt](https://github.com/user-attachments/files/18838498/d455_problems.txt)

The error messages keep repeating, but the Realsense Node never successfully starts. I would appreciate any help in identifying and resolving this issue. Thank you!

MartyG-RealSense, 2025-02-18:
Hi @Ragnar7982  Thanks very much for the log.  The message `Device or resource busy` keeps being displayed when the wrapper attempts to access the camera.

1.  Is realsense-viewer closed at the time that the ROS2 wrapper launch is performed?

2.  Does it make a difference if you add `initial_reset:=true` to your RO2 launch instruction to reset the camera at launch?

3.  Does it make a difference if you unplug the micro-sized USB connector from the base of the D455 camera, turn the connector around the other way and re-insert it?

Ragnar7982, 2025-02-18:
Thank for prompt reply!
1. Yes.
2. No.
3. Sometimes it works, sometimes it doesn't.
It might be D455 camera problems?

MartyG-RealSense, 2025-02-18:
Sometimes RealSense cameras can work better when the micro-sized connector of the USB cable is inserted in one particular direction of the two available.  Once you have found the insertion orientation of the micro-connector that the camera works best in then you do not need to remove it from the base of the camera again.

Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

Ragnar7982, 2025-02-20:
My own one, but now I prob know where might be the problems, thank you for the help!

closed 2025-02-20 by Ragnar7982