realsenseai/realsense-ros #3306: RealSense D405 in ROS2 - Ignoring Resolution & FPS Parameters, High Latency

gitbothubredbot, 2025-02-21:
I’m working on a ROS2-based Turtlebot3 using the Intel RealSense D405 for real-time object detection and navigation. My setup includes:

    **Raspberry Pi 5 (Ubuntu 24, ROS2 Jazzy) → Handles the camera data acquisition.
    Workstation (Ubuntu 22, ROS2 Humble) → Performs object detection & processing, minimizing the load on the Pi5. (Cant upgrade as TB3 packages arent supported for ROS2 Jazzy till now) 
    Dockerized RealSense ROS2 Wrapper → Running on the Pi5 to stream camera data to the workstation.
    Navigation Stack (Nav2) → Uses detected objects for obstacle avoidance and path planning.**

Challenges & Issues
    **High-Latency Camera Data from Pi5 to Workstation

    The Pi5 streams RealSense depth and color data, but when viewed in RViz, the frame rate is extremely low (~1 FPS or worse).
    Viewing in RQT is slightly better, but it’s still not fast enough for real-time processing.
    Even basic ROS2 nodes (like talker/listener) communicate smoothly between Pi5 and the workstation, so the issue seems to be camera-specific.**

    **Device: Intel RealSense D405 + Raspberry Pi 5 8GB  (Ubuntu 24.01 ROS2 Jazzy - Cant Downgrade to Humble as only Ubuntu 24 Supported on Pi5)
    OS: Ubuntu 22.04
    ROS2 Distro: Humble
    Containerized: Yes (Docker)
    librealsense Version: v2.56.0
    ROS Wrapper Version: 4.55.1
    USB Type: 3.2**

Despite setting the correct parameters (both through launch commands and YAML configuration), the camera always starts with 848x480 resolution at 30 FPS instead of the desired 424x240 at 5 FPS. This is causing huge delays and very slow frame rates (1 FPS or worse).

```
docker run -it --rm \
  --privileged \
  --net=host \
  -v /dev/bus/usb:/dev/bus/usb \
  -e ROS_DOMAIN_ID=30 \
  realsense2-humble-rsusb \
  bash -c "
    source /opt/ros/humble/setup.bash &&
    source /root/ros2_ws/install/setup.bash &&
    ros2 launch realsense2_camera rs_launch.py \
      camera_name:=d405 \
      enable_color:=true \
      enable_depth:=true \
      depth_module.profile:=424x240x5 \
      rgb_camera.profile:=424x240x5 \
      global_time_enabled:=false \
      enable_infra1:=false \
      enable_infra2:=false \
      enable_fisheye:=false \
      enable_sync:=false \
      enable_pointcloud:=false
  "
```

But in logs, it still shows:
```

[INFO] [camera.d405]: Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 30
[INFO] [camera.d405]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30

```
Created a yaml file too : 
```
camera:
  ros__parameters:
    depth_module.profile: "424x240x5"
    rgb_camera.profile: "424x240x5"
    global_time_enabled: false
    enable_infra1: false
    enable_infra2: false
    enable_fisheye: false
    enable_sync: false
    enable_pointcloud: false
```
Passed it in the launch command:
``
ros2 launch realsense2_camera rs_launch_get_params_from_yaml.py config_file:=/root/lowres_params.yaml``

Also manually set parameters at runtime : 

```
ros2 param set /camera depth_module.profile "424x240x5"
ros2 param set /camera rgb_camera.profile "424x240x5"

```

Checked camera capabilities with rs-enumerate-devices

    It confirms 424x240 at 5 FPS is supported, but it never applies.

1. Why is the camera ignoring resolution and FPS parameters, even when set explicitly in both launch and YAML?
2. Are there any known issues with the D405 + ROS2 Docker setup where parameters don’t persist?
3. Is there a proper workaround to reliably get low resolution (424x240) and lower FPS (5) for real-time processing?
4. Could this be a firmware issue, and should I attempt a downgrade or upgrade?

All in all, I want the camera topics on rviz or lets say just rqt streaming uninterrupted data to the workstation. They are both on the same ROS Domain ID - connected to the same Wifi, basic ROS2 communications work, i also see the camera topics but then i get some errors  when i echo the ros2 image_rect_raw topic like : 
```
- 164
- 115
- 114
- 171
- 126
- 113
- 155
- 110
- 97
- 151
- 106
- 93
- 154
- 109
- 96
- 154
- 109
- 96
- 155
- 110
- 97
- 155
- 110
- 97
- 155
- 110
- 97
- 157
- 111
- 95
- 159
- 113
- 97
- 168
- 120
- 101
- 169
- 121
- 102
- 162
- 114
- 95
- 161
- 113
- 94
- 163
- 115
- 107
- 167
- 119
- 111
- 163
- 114
- 113
- 163
- 114
- 113
- 159
- 109
- 106
- 164
- 114
- 111
- 167
- 114
- 108
- 168
- 115
- 109
- 159
- 113
- 95
- 168
- 122
- 104
- 163
- 118
- 105
- 165
- 120
- 107
- 160
- 122
- 115
- 152
- 114
- 107
- 169
- 128
- 122
- 163
- 122
- 116
- 152
- 102
- 95
- 161
- 111
- 104
- 166
- 118
- 112
- 154
- 106
- 100
- 171
- 123
- 117
- 164
- 116
- 110
- 155
- 109
- 103
- 154
- 108
- 102
- 155
- 109
- 103
- 157
- 111
- 105
- 161
- 115
- '...'
---
A message was lost!!!
	total count change:1
	total count: 2---
A message was lost!!!
	total count change:1
	total count: 3---
A message was lost!!!
	total count change:1
	total count: 4---
A message was lost!!!
	total count change:1
	total count: 5---
A message was lost!!!
	total count change:1
	total count: 6---
A message was lost!!!
	total count change:1
	total count: 7---
A message was lost!!!
	total count change:1
	total count: 8---
A message was lost!!!
	total count change:1
	total count: 9---
^Ckshitij@kshitij:~$ ros2 topic echo /camera/camera/color/image_rect_raw 
A message was lost!!!
	total count change:142
	total count: 142---
A message was lost!!!
	total count change:1
	total count: 143---
A message was lost!!!
	total count change:3
	total count: 146---
A message was lost!!!
	total count change:3
	total count: 149---
A message was lost!!!
	total count change:4
	total count: 153---
kshitij@kshitij:~$ ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_rect_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/parameter_events
/rosout
/tf_static
kshitij@kshitij:~$ ros2 topic echo /camera/camera/color/image_rect_raw 
A message was lost!!!
	total count change:1
	total count: 1---
header:
  stamp:
    sec: 1740090517
    nanosec: 621034180
  frame_id: camera_color_optical_frame
height: 480
width: 848
encoding: rgb8
is_bigendian: 0
step: 2544
data:
- 149
- 100
- 111
- 152
- 103
- 114
- 157
- 109
- 115
- 158
- 110
- 116
- 154
- 107
- 108
- 155
- 108
- 109
- 157
- 108
- 109
- 157
- 108
- 109
- 164
- 115
- 114
- 160
- 111
- 110
- 161
- 113
- 105
- 163
- 115
- 107
- 163
- 117
- 109
- 163
- 117
- 109
- 156
- 110
- 106
- 161
- 115
- 111
- 164
- 118
- 114
- 162
- 116
- 112
- 163
- 115
- 109
- 164
- 116
- 110
- 167
- 120
- 109
- 158
- 111
- 100
- 158
- 111
- 98
- 161
- 114
- 101
- 165
- 118
- 105
- 159
- 112
- 99
- 157
- 110
- 97
- 157
- 110
- 97
- 153
- 106
- 95
- 154
- 107
- 96
- 155
- 113
- 110
- 161
- 119
- 116
- 153
- 113
- 118
- 153
- 113
- 118
- 158
- 118
- 123
- 163
- 123
- 128
- 161
- 120
- 130
- 159
- 118
- 128
- 156
- 114
- 130
- 159
- 117
- 133
- 169
- 121
- 127
- 159
- 111
- 117
- 162
- 112
- '...'
---
A message was lost!!!
	total count change:2
	total count: 3---
A message was lost!!!
	total count change:1
	total count: 4---
A message was lost!!!
	total count change:1
	total count: 5---
A message was lost!!!
	total count change:1
	total count: 6---
A message was lost!!!
	total count change:1
	total count: 7---
A message was lost!!!
	total count change:1
	total count: 8---
A message was lost!!!
	total count change:1
	total count: 9---
A message was lost!!!
	total count change:2
	total count: 11---
A message was lost!!!
	total count change:1
	total count: 12---
A message was lost!!!
	total count change:1
	total count: 13---
header:
  stamp:
    sec: 1740090518
    nanosec: 54485596
  frame_id: camera_color_optical_frame
height: 480
width: 848
encoding: rgb8
is_bigendian: 0
step: 2544
data:
- 165
- 121
- 105
- 157
- 113
- 97
- 156
- 110
- 90
- 163
- 117
- 97
- 152
- 108
- 92
- 151
- 107
- 91
- 153
- 110
- 101
- 155
- 112
- 103
- 155
- 114
- 108
- 154
- 113
- 107
- 157
- 120
- 115
- 156
- 119
- 114
- 162
- 125
- 122
- 158
- 121
- 118
- 157
- 120
- 117
- 156
- 119
- 116
- 154
- 111
- 113
- 153
- 110
- 112
- 155
- 108
- 107
- 156
- 109
- 108
- 162
- 112
- 109
- 161
- 111
- 108
- 164
- 114
- 111
- 166
- 116
- 113
- 164
- 118
- 114
- 165
- 119
- 115
- 164
- 118
- 114
- 161
- 115
- 111
- 159
- 113
- 107
- 163
- 117
- 111
- 159
- 113
- 109
- 158
- 112
- 108
- 160
- 117
- 112
- 161
- 118
- 113
- 162
- 116
- 112
- 155
- 109
- 105
- 152
- 108
- 108
- 153
- 109
- 109
- 157
- 112
- 117
- 156
- 111
- 116
- 158
- 111
- 112
- 158
- 111
- 112
- 162
- 112
- '...'
---
A message was lost!!!
	total count change:1
	total count: 14---

```
, I need to offload the image processing from the Pi because of computational load, so I need the data and stream coming to workstation. 

I would appreciate your response and thank you for looking into the problem!

MartyG-RealSense, 2025-02-21:
Hi @gitbothubredbot  Support for setting the RGB color profile on a D405 camera is in the **ros2-development** branch of the RealSense ROS wrapper rather than ros2-master, as described at https://github.com/IntelRealSense/realsense-ros/pull/3284#event-15780023403

https://github.com/IntelRealSense/realsense-ros/tree/ros2-development

This branch has to be built from source code.

gitbothubredbot, 2025-02-21:
I built it from source, 
I created a yaml file, setting the depth_module as mentioned in the above comments --
```

camera:
  ros__parameters:
    camera_name: d405
    initial_reset: true
    global_time_enabled: false
    enable_color: true
    enable_depth: true
    enable_infra1: false
    enable_infra2: false
    enable_infra: false
    enable_fisheye: false
    enable_pointcloud: false
    enable_sync: false
    depth_module:
      profile: "424x240x5"
      color_profile: "424x240x5"
      infra_profile: "424x240x5"
      color_format: "RGB8"

```
launched the node as :

`ros2 launch realsense2_camera rs_launch_get_params_from_yaml.py config_file:=/root/lowres_params.yaml `
Terminal Output
```

Set ROS param depth_module.depth_profile to default: 848x480x30
Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 30
Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
```

What I’ve Tried:
Using CLI parameter overrides in rs_launch.py
Using a YAML file with a top-level key of camera: (to match the node name)
Manually setting parameters via ros2 param set after node startup
Disabling global time synchronization (global_time_enabled:=false)
Disabling extra streams (infrared, pointcloud, etc.)

MartyG-RealSense, 2025-02-21:
If you have built the ros2-development wrapper from source code, let's try the ros2 launch instruction below which sets RGB via the **depth** module profile - as mentioned in the linked-to PR - to see whether it configures the D405 depth and color streams to the desired settings.

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=424x240x5 depth_module.color_profile:=424x240x5`

gitbothubredbot, 2025-02-21:
Terminal output : 

```
akkodis@akkodis:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=424x240x5 depth_module.color_profile:=424x240x5
[INFO] [launch]: All log files can be found below /home/akkodis/.ros/log/2025-02-21-13-13-11-476423-akkodis-11596
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [11599]
[realsense2_camera_node-1] [INFO] [1740139991.739835977] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1740139991.739964440] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740139991.739985255] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740139992.028406041] [camera.camera]: Device with serial number 218622273877 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1740139992.028538208] [camera.camera]: Device with physical ID 5-1-2 was found.
[realsense2_camera_node-1] [INFO] [1740139992.028568004] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1740139992.028892430] [camera.camera]: Device with port number 5-1 was found.
[realsense2_camera_node-1] [INFO] [1740139992.028933041] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1740139992.029051523] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1740139992.029519283] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1740139992.029560691] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1740139992.029584654] [camera.camera]: Device Serial No: 218622273877
[realsense2_camera_node-1] [INFO] [1740139992.029605394] [camera.camera]: Device physical port: 5-1-2
[realsense2_camera_node-1] [INFO] [1740139992.029628154] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1740139992.029652357] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1740139992.029671802] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1740139994.562628310] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1740139994.564045535] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1740139994.575833202] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740139994.766225606] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740139994.815477112] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1740139994.815602557] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 424, Height: 240, FPS: 5
[realsense2_camera_node-1]  21/02 13:13:14,816 WARNING [281472393076800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1740139994.818726951] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  21/02 13:13:14,827 WARNING [281472393076800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

Terminal output on the workstation : 

kshitij@kshitij:~$ ros2 topic echo /camera/camera/color/image_rect_raw 
header:
  stamp:
    sec: 1740140101
    nanosec: 261906006
  frame_id: camera_color_optical_frame
height: 480
width: 848
encoding: rgb8
is_bigendian: 0
step: 2544
data:
- 114
- 119
- 120
- 117
- 121
- 122
- 116
- 120
- 121
- 116
- 120
- 121
- 114
- 121
- 121
- 114
- 121
- 121
- 113
- 120
- 120
- 114
- 121
- 121
- 114
- 121
- 121
- 116
- 123
- 123
- 115
- 122
- 122
- 115
- 122
- 122
- 115
- 122
- 122
- 113
- 120
- 120
- 115
- 122
- 122
- 114
- 121
- 121
- 116
- 124
- 121
- 115
- 122
- 120
- 114
- 121
- 119
- 115
- 122
- 120
- 115
- 122
- 120
- 116
- 124
- 121
- 114
- 121
- 119
- 115
- 122
- 120
- 114
- 121
- 119
- 117
- 125
- 122
- 116
- 124
- 121
- 116
- 124
- 121
- 114
- 121
- 121
- 116
- 123
- 123
- 117
- 125
- 122
- 116
- 124
- 121
- 114
- 121
- 119
- 115
- 122
- 120
- 117
- 125
- 122
- 115
- 122
- 120
- 115
- 122
- 120
- 116
- 124
- 121
- 115
- 122
- 120
- 115
- 122
- 120
- 116
- 125
- 122
- 115
- 124
- 121
- 116
- 125
- '...'
---
A message was lost!!!
	total count change:2
	total count: 2---
A message was lost!!!
	total count change:3
	total count: 5---
header:
... and so on

MartyG-RealSense, 2025-02-21:
The depth is being configured to your specification but not the color.

If you have built the ros2-development wrapper in place of the the wrapper that you were using before and your custom color profile is still not being provided on your D405, then I recommend waiting for a response from the ROS wrapper development team to the question that you posted on the PR.

gitbothubredbot, 2025-02-21:
I was wondering if you could comment on the fact that why the data being transferred is with such high latency, or infact stale at some points where its not being transmitted? I get that at 30fps its high frequency data, so is it a drawback and a fundamental issue of the ROS DDS or is there somewhat a workaround how i can figure out this PUB - SUB over the Raspberry Pi to my workstation? (I tried using the ZeroMQ Socket also tried sending over UDP but again very low quality streams with latency > 1000 ms )

gitbothubredbot, 2025-02-21:
@MartyG-RealSense I managed to run the node at lower parameters by reinstanciating the container with the developement pacakge thank you for the tip! 
akkodis@akkodis:~/realsense_docker$ docker run -it --rm \
  --privileged \
  --net=host \
  -v /dev/bus/usb:/dev/bus/usb \
  -e ROS_DOMAIN_ID=30 \
  realsense2-humble-rsusb
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-21-12-48-34-204686-akkodis-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [65]
[realsense2_camera_node-1] [INFO] [1740142114.421206038] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1740142114.421362112] [camera.camera]: Built with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1740142114.421389557] [camera.camera]: Running with LibRealSense v2.56.0
[realsense2_camera_node-1] [INFO] [1740142114.683622229] [camera.camera]: Device with serial number 218622273877 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1740142114.683743118] [camera.camera]: Device with physical ID 5-1-2 was found.
[realsense2_camera_node-1] [INFO] [1740142114.683766970] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1740142114.684152007] [camera.camera]: Device with port number 5-1 was found.
[realsense2_camera_node-1] [INFO] [1740142114.684181544] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1740142114.684356915] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1740142114.684689860] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1740142114.684722879] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1740142114.684743897] [camera.camera]: Device Serial No: 218622273877
[realsense2_camera_node-1] [INFO] [1740142114.684762675] [camera.camera]: Device physical port: 5-1-2
[realsense2_camera_node-1] [INFO] [1740142114.684784694] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1740142114.684806138] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1740142114.684824657] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1740142117.193598401] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1740142117.195051496] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1740142117.196175109] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1740142117.210588502] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740142117.393353623] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740142117.427308899] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 424, Height: 240, FPS: 5
[realsense2_camera_node-1] [INFO] [1740142117.427438344] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 424, Height: 240, FPS: 5
[realsense2_camera_node-1]  21/02 12:48:37,430 WARNING [281473322117344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  21/02 12:48:37,443 WARNING [281473322117344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1740142117.443186257] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  21/02 13:07:04,254 WARNING [281473322117344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

i do see the stream on rqt, but again the stream gets stale in the middle and i have a frozen picture and when i echo the topic i have this issue : 

```
header:
  stamp:
    sec: 1740143395
    nanosec: 52171631
  frame_id: camera_color_optical_frame
height: 240
width: 424
encoding: rgb8
is_bigendian: 0
step: 1272
data:
- 103
- 110
- 114
- 104
- 111
- 115
- 102
- 109
- 113
- 104
- 111
- 115
- 103
- 111
- 112
- 105
- 113
- 114
- 104
- 112
- 113
- 102
- 110
- 111
- 106
- 114
- 113
- 105
- 113
- 112
- 106
- 114
- 113
- 107
- 115
- 114
- 106
- 114
- 113
- 105
- 113
- 112
- 107
- 115
- 114
- 106
- 114
- 113
- 105
- 113
- 112
- 106
- 114
- 113
- 106
- 114
- 113
- 104
- 112
- 111
- 104
- 112
- 111
- 105
- 113
- 112
- 106
- 114
- 113
- 105
- 113
- 112
- 106
- 114
- 113
- 106
- 114
- 113
- 106
- 114
- 113
- 105
- 113
- 112
- 104
- 112
- 109
- 106
- 114
- 111
- 105
- 113
- 110
- 105
- 113
- 110
- 105
- 113
- 112
- 105
- 113
- 112
- 106
- 114
- 113
- 106
- 114
- 113
- 105
- 113
- 112
- 104
- 112
- 111
- 106
- 114
- 113
- 106
- 114
- 113
- 106
- 114
- 113
- 105
- 113
- 112
- 106
- 114
- '...'
---
A message was lost!!!
	total count change:1
	total count: 423---
A message was lost!!!
	total count change:9
	total count: 432---
A message was lost!!!
	total count change:15
	total count: 447---
A message was lost!!!
	total count change:15
	total count: 462---
A message was lost!!!
	total count change:11
	total count: 473---
header:
  stamp:
    sec: 1740143405
    nanosec: 451990967
  frame_id: camera_color_optical_frame
height: 240
width: 424
encoding: rgb8
is_bigendian: 0
step: 1272
data:
- 106
- 112
- 112
- 106
- 112
- 112
- 105
- 111
- 111
- 103
- 109
- 109
- 105
- 113
- 114
- 106
- 114
- 115
- 103
- 114
- 114
- 103
- 114
- 114
```

can you comment that why at such low resolutions i am unable to stream flawless data over to the workstation? 

```
akkodis@akkodis:~$ free
               total        used        free      shared  buff/cache   available
Mem:         8128200      891996     3142376       19320     4313120     7236204
Swap:        1048572           0     1048572
```

the Pi doesnt seem to be overloaded either..

MartyG-RealSense, 2025-02-21:
There have been a couple of past cases where RealSense ROS users were getting low performance when using DDS.  A user at https://github.com/IntelRealSense/realsense-ros/issues/1703 found that switching their DDS to **Cyclone** and setting more memory for buffering helped to provide a good FPS.

gitbothubredbot, 2025-02-21:
Much much better! Will try now will a little higher resolution and see if its still working fine! 
Thank you @MartyG-RealSense

MartyG-RealSense, 2025-02-22:
A RealSense ROS user who had a performance drop when subscribing to multiple topics commented at https://github.com/IntelRealSense/realsense-ros/issues/2250#issuecomment-1038075712 that they found it was due to their librealsense SDK and ROS wrapper versions being mismatched.

Your setup also has a version mismatch.

[realsense2_camera_node-1] [INFO] [1740142114.421206038] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1740142114.421362112] [camera.camera]: Built with LibRealSense v2.56.0

I'm not sure why the log is identifying that you have librealsense SDK version **2.56.0**, as there is not a librealsense version with that number, just 2.56.1, 2.56.2 or 2.56.3.

If you built the **ros2-development** branch of the ROS wrapper and its version number in the launch log is 4.55.1 then the correct SDK version to use with it would be librealsense **2.55.1**.

gitbothubredbot, 2025-02-23:
```
akkodis@akkodis:~/realsense_docker$ docker run -it --rm \
  --privileged \
  --net=host \
  -v /dev/bus/usb:/dev/bus/usb \
  -e ROS_DOMAIN_ID=30 \
  -e RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  realsense2-humble-rsusb
root@akkodis:~/ros2_ws# ros2 launch realsense2_camera rs_launch.py \
  depth_module.depth_profile:=640x480x15 \
  depth_module.color_profile:=640x480x15 \
  pointcloud.enable:=true
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-23-16-20-42-985473-akkodis-82
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [83]
[realsense2_camera_node-1] [INFO] [1740327643.187674276] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1740327643.187812406] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740327643.187835183] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740327643.448236549] [camera.camera]: Device with serial number 218622273877 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1740327643.448465772] [camera.camera]: Device with physical ID 5-1-2 was found.
[realsense2_camera_node-1] [INFO] [1740327643.448528272] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1740327643.448861458] [camera.camera]: Device with port number 5-1 was found.
[realsense2_camera_node-1] [INFO] [1740327643.448902421] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1740327643.449073625] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1740327643.449593533] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1740327643.449993571] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1740327643.450054293] [camera.camera]: Device Serial No: 218622273877
[realsense2_camera_node-1] [INFO] [1740327643.450100330] [camera.camera]: Device physical port: 5-1-2
[realsense2_camera_node-1] [INFO] [1740327643.450132219] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1740327643.450165182] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1740327643.450192868] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1740327645.944367361] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1740327645.945756308] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1740327645.957971643] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740327646.140583804] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740327646.202834335] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1740327646.202981206] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1]  23/02 16:20:46,203 WARNING [281472919464160] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1740327646.204820376] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  23/02 16:20:46,214 WARNING [281472919464160] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  23/02 16:21:02,170 ERROR [281472718137568] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132
[realsense2_camera_node-1]  23/02 16:21:13,507 ERROR [281472548268256] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[realsense2_camera_node-1]  23/02 16:21:24,819 ERROR [281472718137568] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 132

```
I tried matching the SDK with the Node like you said, but now I seem to have this new watchdog errors everytime i try to view the stream in rqt or rviz.

MartyG-RealSense, 2025-02-24:
If you change the librealsense SDK version then the ROS wrapper has to be built again afterwards.  Did you build 4.55.1 again after installing 2.55.1 please?

gitbothubredbot, 2025-02-24:
The realsense2_camera package in the ros2-development branch is looking for librealsense version 2.56.0 (or higher), but my Dockerfile installs v2.55.1. CMake rejects v2.55.1 because it doesn’t meet the minimum version requirement.

MartyG-RealSense, 2025-02-24:
So the last known good configuration, in which you were able to obtain the pointcloud at 424x240x5, was librealsense 2.56.0 and RealSense ROS v4.55.1 ?

If your Dockerfile installs librealsense from dkms packages then it would make sense why you can only get 2.55.1, as the 2.56 beta versions of librealsense do not have packages, so 2.55.1 would be the most recent version that has packages.

closed 2025-03-15 by MartyG-RealSense