realsenseai/realsense-ros #3307: No message published in topic once enabled IMU

AMLIFY, 2025-02-25:
Camera Model: D455
SDK Version: 2.55.1
Platform: Jetson Orin Nano, JetPack 6.1
OS Version: 22.04.5 LTS
Kernel: 5.15.148-tegra
I was introducing IMU to my project recently. However, once I switched on the "enable_gyro" and "enable_accel" in the launch file, all topics of the camera could receive no message. And I checked in realsense-viewer. It showed "No Frame Received" once I switched on the motion module. The same error didn't occurred on my colleague's board with my camera.

MartyG-RealSense, 2025-02-25:
Hi @AMLIFY  RealSense cameras equipped with an IMU can experience problems when used with JetPack 6 that range from IMU data not being accessible to the camera not being detected.  The effect that Jetson users experience varies.

https://github.com/IntelRealSense/librealsense/issues/13784 has more information about this issue.  The fix listed in that discussion that works most often - and also worked for the user in that particular case - is to build the librealsense SDK from source code with the **libuvc backend** installation method and then downgrade the camera's firmware driver version.

AMLIFY, 2025-02-26:
Hi @MartyG-RealSense I downgraded the firmware back to 5.13.0.50, and the IMU could work now. Thanks for your useful advice!

closed 2025-02-26 by AMLIFY

MartyG-RealSense, 2025-02-26:
You are very welcome, @AMLIFY - I'm pleased to hear that a firmware downgrade worked for you too.  Thanks very much for the update!