realsenseai/realsense-ros #3308: Can't lower depth_fps

AldoPenaGa, 2025-02-27:
I can't lower the depth_fps running on my D455, at first I decided to use a custom launchfile for the camera and a LiDAR that I have connected (which I don't include here), I added the following to the launchfile: "<launch>
  
  <group ns="camera">
    
    <arg name="serial_no"           default=""/>
    <arg name="usb_port_id"         default=""/>
    <arg name="device_type"         default=""/>
    <arg name="json_file_path"      default=""/>
    <arg name="camera"              default="camera"/>
    <arg name="tf_prefix"           default="$(arg camera)"/>
    <arg name="external_manager"    default="false"/>
    <arg name="manager"             default="realsense2_camera_manager"/>
    <arg name="output"              default="screen"/>
    <arg name="respawn"             default="false"/>

    
    <arg name="enable_depth"        default="true"/>
    <arg name="enable_color"        default="true"/>
    <arg name="align_depth"         default="false"/>
    <arg name="publish_tf"          default="true"/>
    <arg name="tf_publish_rate"     default="0"/>

    <arg name="depth_width" default="1280"/>
    <arg name="depth_height" default="720"/>
    <arg name="depth_fps"   default="10"/>
    <arg name="color_width" default="1280"/>
    <arg name="color_height" default="720"/>
    <arg name="color_fps"   default="10"/>

    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      
      <arg name="tf_prefix"         value="$(arg tf_prefix)"/>
      <arg name="external_manager"  value="$(arg external_manager)"/>
      <arg name="manager"           value="$(arg manager)"/>
      <arg name="output"            value="$(arg output)"/>
      <arg name="respawn"           value="$(arg respawn)"/>
      <arg name="serial_no"         value="$(arg serial_no)"/>
      <arg name="usb_port_id"       value="$(arg usb_port_id)"/>
      <arg name="device_type"       value="$(arg device_type)"/>
      <arg name="json_file_path"    value="$(arg json_file_path)"/>

      
      <arg name="depth_width"       value="$(arg depth_width)"/>
      <arg name="depth_height"      value="$(arg depth_height)"/>
      <arg name="enable_depth"      value="$(arg enable_depth)"/>
      <arg name="color_width"       value="$(arg color_width)"/>
      <arg name="color_height"      value="$(arg color_height)"/>
      <arg name="enable_color"      value="$(arg enable_color)"/>

      <arg name="align_depth"       value="$(arg align_depth)"/>
      <arg name="publish_tf"        value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"   value="$(arg tf_publish_rate)"/>
    </include>
  </group>

Still, the fps didn't lower to 10. I tried making the same change on /opt/ros/noetic/share/realsense2_camera/launch/rs_camera.launch, but it didn't lower the fps either.

MartyG-RealSense, 2025-02-27:
Hi @AldoPenaGa  The FPS should be set to 5 or 15 on your D455 instead of 10.

AldoPenaGa, 2025-02-27:
I changed them to 15, and still: [ INFO] [1740679602.419557141]: Done Setting Dynamic reconfig parameters.
[ INFO] [1740679602.420662178]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1740679602.422020510]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1740679602.422049898]: setupPublishers...
[ INFO] [1740679602.423753886]: Expected frequency for depth = 30.00000
[ INFO] [1740679602.425940450]: Expected frequency for color = 30.00000

MartyG-RealSense, 2025-02-27:
It looks as though the launch is only paying attention to the FPS value that you configured and not the stream width and height values.  The RealSense ROS1 wrapper will reject a custom configuration and apply the default stream profiles instead unless all three parameters are provided for each stream type.

Let's try defining it with a roslaunch instruction, as the values set in the launch instruction will override the values defined in the launch file.

`roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15`

AldoPenaGa, 2025-02-28:
If launch rs_camera.launch with that command, it actually goes to 15 fps, but I need it to work in this custom launchfile since I intend to launch this ROS node along another node.

AldoPenaGa, 2025-02-28:
I solved it by including the launchfile in the joint launchfile, it is a quicker fix instead of creating this new part for the custom launchfile.
Thank you.

MartyG-RealSense, 2025-02-28:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!

closed 2025-03-08 by MartyG-RealSense