realsenseai/realsense-ros #3309: No pointCloud2 ros2 topic available

Debactor, 2025-03-02:
May I have your assist?
**System Description**
Jetson orin nano（arm64) with Jetpack 5.1.2
Ubuntu 20.04.6 LTS
ros2 foxy
Intel RealSense D435
LibRealSense v2.55.1
RealSense ROS v4.55.1
Device USB type: 3.2
Device FW version: 5.16.0.1

**Problem**
When I run the following command, 
`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`
it show:
```
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2025-03-02-10-41-11-319731-robot-46336
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [46339]
[realsense2_camera_node-1] [INFO] [1740883272.088601290] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1740883272.088871006] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740883272.088949421] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1740883272.141047956] [camera.camera]: Device with serial number 814412070436 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1740883272.141211284] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1740883272.141256957] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1740883272.141692977] [camera.camera]: Device with port number 2-1.1 was found.
[realsense2_camera_node-1] [INFO] [1740883272.141737914] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1740883272.142304487] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1740883272.142774114] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1740883272.142892760] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1740883272.142927135] [camera.camera]: Device Serial No: 814412070436
[realsense2_camera_node-1] [INFO] [1740883272.142950884] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1740883272.143000877] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1740883272.143020337] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1740883272.143038805] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1740883272.311830391] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1740883272.314164825] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1740883272.321593555] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1740883272.337402366] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1740883272.361345836] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740883272.361787105] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1740883272.375027516] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1740883272.390403444] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1740883272.416820863] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1740883272.426752860] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1740883272.430901852] [camera.camera]: RealSense Node Is Up!
```
But when I used Rviz2 to visualize the topic /camera/camera/depth/color/points, the promblem emerged:
`[realsense2_camera_node-1] [WARN] [1740883294.503523871] [camera.camera]: No stream match for pointcloud chosen texture Process - Color`
And I can get neither the PointCloud2 of /camera/camera/depth/color/points in rviz, nor the /camera/camera/color/image_raw. I wonder why this happened and how to solve it?

MartyG-RealSense, 2025-03-02:
Hi @Debactor  The warning `No stream match for pointcloud chosen texture Process - Color` indicates that color frames are missing, perhaps because some of them are being dropped.

Is the pointcloud displayed in RViz if you use the command below to texture the pointcloud with the infrared image instead of tge RGB color image?

`ros2 launch realsense2_camera rs_launch.py enable_color:=false infra1_enable:=true pointcloud.enable:=true pointcloud.stream_filter:=1`

I note that you are using ROS2 Foxy.  Support for Foxy was removed in RealSense ROS wrapper 4.54.1 onwards.  4.55.1 is meant to be used with ROS2 Humble or Iron.

Debactor, 2025-03-02:
> Hi [@Debactor](https://github.com/Debactor) The warning `No stream match for pointcloud chosen texture Process - Color` indicates that color frames are missing, perhaps because some of them are [...]

Hi @MartyG-RealSense 
Thank you for your answer, but I still got the same warning  `No stream match for pointcloud chosen texture Process - Color` when I used the command 
`ros2 launch realsense2_camera rs_launch.py enable_color:=false infra1_enable:=true pointcloud.enable:=true pointcloud.stream_filter:=1`

Even if I changed the version of SDK, firmware and realsense-ros, the same warning emerged. Now my system description is:
    LibRealSense v2.51.1
    RealSense ROS v4.51.1
    Device FW version: 05.13.00.50

In Rviz2, the pointcloud is still not displayed, and there is no choice in the  "Color Transformer"  .  And when I use the command` ros2 topic echo /camera/depth/color/points` ， nothing is displayed.

MartyG-RealSense, 2025-03-02:
There is not a clear reason why your pointcloud topic is empty of data even though it is published and the pointcloud is not being displayed in RViz,

Let's next try setting both depth and color to the same resolution in 4.51.1 using the launch instruction below.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 rgb_camera.profile:=1280x720x3 pointcloud.enable:=true`

Debactor, 2025-03-03:
If I use this command 
`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 rgb_camera.profile:=1280x720x30 pointcloud.enable:=true`
there will be an  some warnings,and surely the pointcloud is not displayed:
```
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2025-03-03-09-23-41-944298-robot-5157
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5159]
[realsense2_camera_node-1] [INFO] [1740965022.598831732] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1740965022.599096777] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1740965022.599200158] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1740965022.666957969] [camera.camera]: Device with serial number 814412070436 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1740965022.667134996] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1740965022.667180477] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1740965022.667618612] [camera.camera]: Device with port number 2-1.1 was found.
[realsense2_camera_node-1] [INFO] [1740965022.667669311] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1740965022.677858640] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1740965022.679075267] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1740965022.679137967] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1740965022.679170198] [camera.camera]: Device Serial No: 814412070436
[realsense2_camera_node-1] [INFO] [1740965022.679197883] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1740965022.679225409] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1740965022.679252838] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1740965022.679282188] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1740965022.814740006] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1740965022.829611262] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1740965022.893584445] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1740965022.980243532] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1740965022.989514983] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1740965023.095208458] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1740965023.097422085] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1740965023.105660212] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1740965023.143464857] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1740965023.147680068] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1740965023.163837154] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1740965024.993899375] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.74097e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1740965024.994070066] [camera.camera]: Hardware Notification:Depth stream start failure,1.74097e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1740965043.891943724] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

```
It seems that the depth stream is not working.

MartyG-RealSense, 2025-03-03:
Let's next try a very low stream configuration of 640x480 and 6 FPS for both depth and color.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x6 rgb_camera.profile:=640x480x6 pointcloud.enable:=true`

If performance improves when using a low stream configuration and becomes worse when using a high configuration such as 1280x720 at 30 FPS, it could suggest that the Orin Nano is providing the camera with insufficient power.

Intel strongly recommend using the instructions at the link below to enable the **barrel jack** connector for extra power on Nano type boards.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

MartyG-RealSense, 2025-03-03:
My research of your question suggests that the Orin Nano cannot access maximum-power nvp mode MAXN because it needs JetPack to be updated from 5.1.2 to JetPack 6.  As you are using Ubuntu 20.04 though it would not be practical to do this, as it is my understanding that 20.04 supports JetPack 5 but 22.04 or newer would be needed in order to use JP6.

https://www.cytron.io/tutorial/upgrade-your-jetson-orin-nano-to-the-super-powered-version

Does the pointcloud work normally in RViz at 640x480x30 even if Hardware Error warnings are generating in the log?  If the pointcloud is displayed normally at 30 FPS despite the warnings then it is likely to be safe to ignore the warnings.

Debactor, 2025-03-04:
> 1. Does the pointcloud work normally in RViz at 640x480x30 even if Hardware Error warnings are generating in the log? If the pointcloud is displayed normally at 30 FPS despite the warnings then it is likely to be safe to ignore the warnings.
 
No, there is no pointcloud displayed in rviz. Even if I set to 640x480x10, nothing is displayed, either. 
What is interesting is that when I set the FPS to lower than 6(such as 640x480x3), no pointcloud is displayed.  And when I use the command `ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x3 rgb_camera.profile:=640x480x3 pointcloud.enable:=true`, there are errors and warnings as followed:
```
[realsense2_camera_node-1] [ERROR] [1741053971.176036942] [camera.camera]: Given value, 640x480x3 is invalid. Set ROS param back to: 848x480x30
[realsense2_camera_node-1] [WARN] [1741053971.191888075] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [ERROR] [1741053971.239813642] [camera.camera]: Given value, 640x480x3 is invalid. Set ROS param back to: 640x480x30
[realsense2_camera_node-1] [INFO] [1741053971.304765921] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1741053971.312631374] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1741053971.398573747] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1741053971.399450791] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1741053971.403959178] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1741053971.423923874] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1741053971.427745302] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1741053971.465966368] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1741053975.271586123] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1]  04/03 10:06:57,120 WARNING [281471923235232] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
```
I only succeed when I set to 6FPS.

> 2. My research of your question suggests that the Orin Nano cannot access maximum-power nvp mode MAXN because it needs JetPack to be updated from 5.1.2 to JetPack 6. As you are using Ubuntu 20.04 though it would not be practical to do this, as it is my understanding that 20.04 supports JetPack 5 but 22.04 or newer would be needed in order to use JP6.

As far as I know, the maximum-power nvp mode means that the frequency of the CPU, GPU will be set higher. Is it that when the power consumed by modules such as the CPU and GPU is higher, the power allocated to peripheral devices such as USB should be less rather than more, given that the electric power provided by the external power supply is constant? This makes me confused.

Do I have no other options except to update the Ubuntu system to version 22.04 and update JetPack to version 6.2?

MartyG-RealSense, 2025-03-04:
6 FPS is the minimum FPS supported on D435.  If the FPS is set to less than 6, such as 3, then the configuration will be rejected and the default stream profile at 30 FPS will be applied to the launch instead.  This is shown in the log.

`Given value, 640x480x3 is invalid.  Set ROS param back to: 640x480x30`

Even when a Jetson board is powered by a mains electricity adapter, USB devices should still be drawing the same amount of power from the Jetson's power supply.  This method of supplying power to a USB device is less stable than using a mains-plug powered USB hub so that cameras attached to the hub are powered independently from the Jetson by the mains electricity and so are typically more stable and are not drawing upon the Jetson's own power reserves.

If you do not plan to use MAXN mode then upgrading to JetPack 6 would likely not make a difference to your pointcloud problem, since the main reason for a JP6 upgrade when using librealsense would be (a) if you needed to use Ubuntu 22.04, and (b) if you had problems with accessing a RealSense IMU (which the D435 model does not have).

Your idea of writing your own node may have merit.  Another ROS2 user at https://github.com/IntelRealSense/librealsense/issues/13671#issuecomment-2695053394 yesterday solved their pointcloud problem by writing their own simple ROS2 wrapper.

Debactor, 2025-03-04:
Thanks for your patience! 

>  This method of supplying power to a USB device is less stable than using a mains-plug powered USB hub so that cameras attached to the hub are powered independently from the Jetson by the mains electricity and so are typically more stable and are not drawing upon the Jetson's own power reserves.

Could you please help me how to deploy the "mains-plug powered USB hub"? I have no idea about it.

> Your idea of writing your own node may have merit. Another ROS2 user at https://github.com/IntelRealSense/librealsense/issues/13671#issuecomment-2695053394 yesterday solved their pointcloud problem by writing their own simple ROS2 wrapper.
Let me try this!

MartyG-RealSense, 2025-03-04:
Mains powered USB3 hubs can be purchased from stores such as Amazon by searching them for the term **powered usb3 hub**

Powered hubs have a lead that goes from the hub to an electricity mains plug socket.  They also have a USB cable that goes from the hub to a USB port on the computer.  

The USB cable supplied with the hub comes in USB-A (full size connector) or USB-C (micro-sized connector), so if you do not have a micro-sized USB-C port on your computer then you need to make sure you choose a hub with a full-size USB-A connector.  You can tell which size of connector it has from images in store listings.

**USB-A**

[image]

**USB-C**

[image]

You could insert a USB-A hub into the port that the camera is currently using and then attach the camera to the hub.

closed 2025-03-22 by MartyG-RealSense