realsenseai/realsense-ros #3312: Base Frame ID parameter missing in launch file

FarStryke21, 2025-03-04:
Hey! 

I am looking to link my robot with the RealSense camera (a D435i) which is mounted on the tool. My URDF defines the mounting of what would be the camera_link of the sensor. 

The documentation for the repository states the availability of a base_frame_id which can be set during launch. The `rs_launch.py` file seems to be missing this parameter. Apart from this, everything works well!

I have noticed the parameter defaults to `camera_link`. However, the ability to define it at runtime would be a great help as we would not need a redundant transform defining a fixed joint between the link created in the URDF and the camera_link.

Is there a way to expose this parameter to the launch file? I am looking for launching something like this:
`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true base_frame_id:=tool_rs_mount_1`

FarStryke21, 2025-03-05:
Hi @MartyG-RealSense !

That is exactly what I am looking for. Adding this exposes the `base_frame_id `parameter in the launch file. I spent some time tracing how the links are created and named. If I were to add this parameter to the launch configuration, wouldn't it be changed to `camera_tool_rs_mount_1` on execution? My understanding is that the RealSense package enforces a strict naming convention for each camera. 

Now, I could get around and change the link name in my URDF to `camera_tool_rs_mount_1`, solving the issue. But I was hoping to disable the `camera_` which would be prepended to the link names? Simply passing empty strings to the naming arguments in the launch file would still result in the `_` being prepended.

MartyG-RealSense, 2025-03-05:
You can define a custom name and namespace for each camera.  Instructions for doing so can be found in the section of the ROS2 wrapper instructions linked to below.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#camera-name-and-camera-namespace

If the name / namespace parameters are not used then the name and namespace are set to 'camera' by default.

Nir-Az, 2025-03-11:
PR #3313  merged

closed 2025-03-11 by Nir-Az