realsenseai/realsense-ros #3314: Not able to set widhtxheighxfps using depth profile for realsense d435i

Akumar201, 2025-03-10:
I am trying to set depth profile on my realsense d435i, the issue is when I am changing it to any other value it is not able to talk to the hardware 
```

[realsense2_camera_node-1] [WARN] [1741581289.129918624] [camera.camera_top]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74158e+12,Warn,Frames Timeout
[realsense2_camera_node-1]  09/03 21:34:54,134 WARNING [134003281626688] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1741581294.134894479] [camera.camera_top]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74158e+12,Warn,Frames Timeout
[realsense2_camera_node-1]  09/03 21:34:59,139 WARNING [134003281626688] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1741581299.139939820] [camera.camera_top]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74158e+12,Warn,Frames Timeout
```
This is my launch file 

```

def launch_setup(context, *args, **kwargs):
    """Dynamically set the RealSense serial number and resolution."""
    camera_name = LaunchConfiguration('camera_name').perform(context)  # Get the camera name from the argument
    serial_no = load_serial_number(camera_name)  # Read the serial number from the YAML file
    depth_profile = LaunchConfiguration('depth_profile').perform(context)
    color_profile = LaunchConfiguration('color_profile').perform(context)
    return [
        launch_ros.actions.Node(
            package='realsense2_camera',
            executable='realsense2_camera_node',
            name=camera_name,
            parameters=[{
                'serial_no': serial_no,
                'enable_depth': True,
                'enable_color': True,
                'align_depth': True,
                'enable_pointcloud': True,
                'depth_module.depth_profile': depth_profile,  
                'rgb_camera.color_profile': color_profile   
            }],
            output='screen'
        )
    ]

def generate_launch_description():
    return launch.LaunchDescription([
        # Camera selection
        DeclareLaunchArgument(
            'camera_name',
            default_value='camera_top',
            description='Camera name to load from config file (e.g., camera_top, camera_wrist)'
        ),
        DeclareLaunchArgument(
            'depth_profile',
            default_value='1280x720x6',  
            description='Depth module profile (widthxheightxFPS)'
        ),
        # Color Profile
        DeclareLaunchArgument(
            'color_profile',
            default_value='848x480x60',  
            description='Color camera profile (widthxheightxFPS)'
        ),
        # Start node setup
        OpaqueFunction(function=launch_setup)
    ])

```

The camera is working at USB 3.0 

```
/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    |__ Port 2: Dev 3, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 5, Class=Human Interface Device, Driver=usbhid, 5000M

```

MartyG-RealSense, 2025-03-10:
Hi @Akumar201  How does the launch behave if you set the configuration with a **ros2 launch** instruction instead of launching the node with a yaml file?

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=848x480x60 pointcloud.enable:=true align_depth.enable:=true`

Akumar201, 2025-03-10:
@MartyG-RealSense , this command that you suggested worked. 

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=848x480x60 pointcloud.enable:=true align_depth.enable:=true`

However as soon as I select 1280x720x30 for both I am getting the same error 
```

[realsense2_camera_node-1] [WARN] [1741621628.592530683] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.74162e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1741621628.592606207] [camera.camera]: Hardware Notification:Depth stream start failure,1.74162e+12,Error,Hardware Error
[realsense2_camera_node-1]  10/03 08:47:12,598 WARNING [140624913237568] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1741621632.598662842] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74162e+12,Warn,Frames Timeout
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```
for my user case I want both to be 1280x720 , I think I am fine with the low fps. Thank You

MartyG-RealSense, 2025-03-10:
You are very welcome!  D435i supports 6, 15 and 30 FPS for 1280x720, with 60 FPS available for 848x480 resolution.

Akumar201, 2025-03-10:
Okay , Thank you for your prompt replie. So what would be the best fps if I want 1280x720 from both camera rgb and depth as 1280x720 resolution.

MartyG-RealSense, 2025-03-11:
I would recommend testing 15 FPS first and then testing 30 if 15 works well.

Akumar201, 2025-03-11:
@MartyG-RealSense , thank you for your suggestion. I am closing this issue for now.

closed 2025-03-11 by Akumar201