realsenseai/realsense-ros #3315: RGB resolution and resulting change in pixel density

Likhith-Sugganahalli, 2025-03-10:
Hi,

I am using the d435i for a visual application with the camera set to resolution of `848x480` and 60 fps.I had picked this mainly for the higher fps, but right now, my requirements are such that the lower pixel density is hurting my algorithms more than the increased fps is helping.

When i switch to `1280x720`, it is much better, but now there is an increased FoV, which is a problem.Is there a way that i can use the higher resolution and still have the same FoV as the lower `848x480`.

One way would be that i crop the edges myself, but then i want to ensure that the FoV is exact, can you help me out with this please

MartyG-RealSense, 2025-03-10:
Hi @Likhith-Sugganahalli  Whilst it is possible to program cropping of an image to restrict the horizontal / vertical planes, it is complicated to do so.  https://github.com/IntelRealSense/librealsense/issues/2769#issuecomment-442035864 suggests a method for cropping a 3D pointcloud with Python code, whilst https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157 has a method for cropping XY with C++ code.

If you use depth to color alignment with the **align_to** instruction instead of cropping the image then on a D435i the outer edges of the aligned image should be automatically cropped off, because the RGB color sensor on the D435i model has a smaller field of view than the depth sensor and so it cuts off the areas of the depth image that the RGB sensor cannot see.

Likhith-Sugganahalli, 2025-03-10:
@MartyG-RealSense 

Appreciate the quick response,

Got it, for this work, i am actually not using the depth data in any way, so i am only concerned with the FoV change of the RGB sensor with change in resolution.The use case is such that the distance to scene is not varying significantly, so i will try just cropping the XY of the rgb image.

thanks for your help.

MartyG-RealSense, 2025-03-10:
You are very welcome.  Please do let me know if I can assist with anything else.  Good luck!

closed 2025-03-10 by Likhith-Sugganahalli